espressif/arduino-esp32 #2817: Esp32-cam face recognition works with 1.0.1, not with 1.0.2

djsamperi, 2019-05-24:
### Hardware:
Board:							ESP32-CAM module
Core Installation version:			1.0.2
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    yes
Upload Speed:				       921600
Computer OS:                                        Windows 10

### Description:
ESP32-CAM face recognition (and face enrolling) does not work with core version 1.0.2, while
it works fine with core version 1.0.1. With version 1.0.2 there is no green recognition rectangle, and there is no information about face recognition in the console log. The face recognition buttons do not do anything when core version 1.0.2 is installed, in my testing.

### Sketch: 
I'm using the ESP32 / CameraWebServer sketch that comes with the core libs for ESP32.

### Debug Messages:
There are no error messages, but there is no face recognition.

fl3tching101, 2019-06-15:
Seconded. I was running my ESP32-CAM with V1.0.2 and couldn't get facial recognition/enrolling working, saw this, switched to V1.0.1 and it works. Something must have gone wrong when porting over versions. I am using the ESP32-CAM DIY More model (AI Thinker knock off).

the-ica, 2019-07-13:
Same here ! how is it possible to debug ?

mwargan, 2019-07-14:
And another +1 here running TTGO cam with SSD1306, with BME280 position

rramacciani, 2019-08-18:
+1 Esp32 cam

closed 2019-10-14 by me-no-dev