espressif/arduino-esp32 #2818: documentation: please provide a readme.txt to all ino Examples + fully detailed comments in either source code

dsyleixa, 2019-05-24:
hello,
I appreciate the multiple Arduino examples in https://github.com/espressif/arduino-esp32/tree/master/libraries and those which are installed to the Arduino libraries folder, but it's hard to see what either example actually does by which requirements and by which functionalities.
Looking into the ino examples itself it's also not clearly visible and understandable what  the single code lines mean, what they do in detail and which arguments passed to ither function does or means.
e.g. in the WiFi folder there are 

	WiFiAccessPoint 	add WiFiAccessPoint example (#1833) 	8 months ago
	WiFiBlueToothSwitch 	Major IDF and Arduino Update 	2 years ago
	WiFiClient 	initial import 	3 years ago
	WiFiClientBasic 	Clarified comments and debug messages in WiFiClient Example (#2389) 	3 months ago
	WiFiClientEnterprise 	Updated table of tested locations of Eduroam network (#2157) 	6 months ago
	WiFiClientEvents 	WiFi: improve WiFiEvent list in WiFiClientEvents example (#1917) 	6 months ago
	WiFiIPv6 	update windows toolchain and platform.txt 	2 years ago
	WiFiMulti 	initial import 	3 years ago
	WiFiScan 	Remove non-working WiFi examples and fix some that require changes 	3 years ago
	WiFiSmartConfig 	Add smartConfig support (#136) 	2 years ago
	WiFiTelnetToSerial 	Fix TelnetToSerial sketch 	11 months ago
	WiFiUDPClient 	Added a basic WiFi UDP client example (#114) 	2 years ago

but as a beginner one is overwhelmed and overburdended by all those functions not knowing which is what for and by which to start perhaps apart from just WiFiClientBasic)  and where then to proceeed for a deeper understanding. didactically meaningful building on each other.

So I would like to request to provide
a) a readme.txt to all ino Examples for a detailed description of this special example, and
b) fully detailed comments in either source code line.

CelliesProjects, 2019-05-24:
I gues you will need some C++ knowledge before you can continue your adventure.
If you lack knowledge to understand these examples study more.
Part of the fun is 'getting it' and testing the examples to see what they encompass.

Basically you are begging.

dsyleixa, 2019-05-24:
I do understand C(++), but not C(++) is crucial but the syntax of the library functions and how they have to be called. For that purpose a better lib documentation is needed.
The pure names of the ino files actually are not so much meaningful to beginners (e.g. WifiServer, WebServer, UDP or SPIFFS) if one has no idea yet what WifiServer, WebServer, UDP, or SPIFFS eventually is about at all.

CelliesProjects, 2019-05-24:
If one has no idea, the internet can be of great help.

dsyleixa, 2019-05-24:
@CelliesProjects:
it's about working on a API code  documentation  for Arduino users which is suitable to be integrated to the esp32 repo examples.
For general basic Arduino users there are examples  also in Arduino documentation and Arduino Playground, too, but your esp32 API functions are not covered by those.

Other way round your statement makes utterly no sense, because then one wouldn't need a repo or a doumentaion at all, and for end user questions you might always answer snootily "Google is you friend".

So please stop arguing and let people answer and contribute who are willing to help.

atanisoft, 2019-05-24:
This would be a great opportunity to contribute some additional documentation to the project.

Feel free to submit a PR with documentation added to the examples and it can be reviewed/discussed.

dsyleixa, 2019-05-24:
the problem is actually:
I don't understand them yet by myself :-/

atanisoft, 2019-05-24:
Most of the API are pretty generic and similar/same as other Arduino environments. But the esp32 specific HAL API is a bit sparse on documentation in a few areas and requires a bit of delving into the code and referring to esp-idf docs often.

A PR is a pull-request, it is how you can submit code changes to this repository.

I'd suggest picking one area to start in and review what docs are available and improve on those to start with. Learn what the code does and how it works along the way. Once that area is fleshed out a bit more, move to another, etc.

dsyleixa, 2019-05-24:
which is the first tutorial code, which the 2nd, which the 3rd,....?

atanisoft, 2019-05-24:
> which is the first tutorial code, which the 2nd, which the 3rd,....?

Whichever you feel like starting with

dsyleixa, 2019-05-24:
I don't know where to start at best and then where to proceed,, because I don't understand which code is what for, that is what I am finally talking about here.

closed 2019-08-14 by stale[bot]