espressif/arduino-esp32 #2826: Feature Request: lwip with NAT & IP Forwarding features

mantatcheung, 2019-05-26:
I'd like to implement a WiFi Repeater with ESP32.  Can the standard Arduino libraries support NAT and IP Forwarding?  I see that using ESP32 IDF can perform such with modification of the IDF lwip component - https://github.com/jonask1337/esp-lwip.  If Arduino does not have this feature at this moment, can you include it in your future releases?

atanisoft, 2019-05-26:
This functionality comes from esp-idf, if it is not available there it will need to first be implemented there. Please open an issue against esp-idf for tracking the functionality on that side.

kotakomputer, 2019-05-27:
+1 this NAT Feature must be available, that's why WIFI_AP_STA required. Currently ESP32 has WIFI_AP_STA mode but not enable to make Client access internet.

closed 2019-08-14 by stale[bot]

athusrv, 2020-08-11:
+1 for NAT/NAPT feature... is this planned to be implemented soon?