espressif/arduino-esp32 #2827: stl::unordered_map insert() causes abort() when out-of-memory

zoliweiss, 2019-05-26:
### Hardware:

Board: ESP32 DoIt module - Chip is ESP32D0WDQ6 (revision 1)
Core Installation version: 1.0.2-rc1
IDE name: Arduino IDE 1.8.5
Flash Frequency: 80Mhz
PSRAM enabled: IDK
Upload Speed: 921600
Computer OS: Win 7

### Description:

When I insert too much entries into unordered_map container, the ESP32 suddenly reboots. Should be throwing bad_alloc exception. I tried catching (...) also. Exception decoder can't decode anything. I am not sure if this is an ESP32/GCC/... problem.

### The code:
```
  std::unordered_map<int, std::string> l_map;

  for(int i = 0; i < 100000; i++)
  {
    printf("%d\n", i);

    try
    {
      auto l_result = l_map.insert({i, std::to_string(i)});

      std::cout << l_result.second << std::endl;
    }
    catch(const std::bad_alloc&)
    {
      printf("OOOOOPS\n");
    }
  }
```

### Debug Messages:
```
1
5378
1
5379
1
5380
1
5381
abort() was called at PC 0x400f594f on core 1

Backtrace: 0x4008b384:0x3ffb1e40 0x4008b5b5:0x3ffb1e60 0x400f594f:0x3ffb1e80 0x400f5996:0x3ffb1ea0 0x400f5b71:0x3ffb1ec0 0x400e6140:0x3ffb1ee0 0x400d18aa:0x3ffb1f00 0x400d1986:0x3ffb1f30 0x40100eab:0x3ffb1fb0 0x4008765d:0x3ffb1fd0

Rebooting...
```

lbernstone, 2019-05-26:
This would need to be fixed upstream in any case.  Open an issue at https://github.com/espressif/esp-idf/issues

igrr, 2019-05-27:
Is it possible to increase CONFIG_CXX_EXCEPTIONS_EMG_POOL_SIZE to a non-zero value, so that exception object can be created when there is an out-of-memory condition?

closed 2019-08-14 by stale[bot]