espressif/arduino-esp32 #2830: 8 MB PSRAM himem?

jcw, 2019-05-27:
Make your question, not a Statement, inclusive. Include all pertinent information:

I'm trying to make the [himem banking demo](https://github.com/espressif/esp-idf/tree/master/examples/system/himem) work. Tried with platformio and the arduino ide, see [this forum discussion](https://community.platformio.org/t/esp32-8-mb-psram-himem/7966).

Latest IDE, github head <https://github.com/espressif/arduino-esp32.git>, latest MacOS, target is based on WROVER (TTGO T7 v1.3).

The test code I have is:

```
extern "C" {
#include <esp_spiram.h>
#include <esp_himem.h>
}

void setup() {
  esp_spiram_init();
  printf("spiram size %u\n", esp_spiram_get_size());
  printf("himem free %u\n", esp_himem_get_free_size());
  printf("himem phys %u\n", esp_himem_get_phys_size());
  printf("himem reserved %u\n", esp_himem_reserved_area_size());
}

void loop() {}
```

The output I see is:

```
entry 0x400806a8
E (121) spiram: SPI RAM not initialized
E (121) esp_himem: Cannot allocate memory for meta info. Not initializing!
spiram size 8388608
himem free 0
himem phys 4456448
himem reserved 262144
```

nevercast, 2019-05-28:
```
E (121) spiram: SPI RAM not initialized
E (121) esp_himem: Cannot allocate memory for meta info. Not initializing!
```

These are errors. You need to work out why your spiram isn't working

nevercast, 2019-05-28:
Are you able to compile arduino-esp32 and test the customized version? Using arduino as a component and changing the esp-idf. If 
`spiram: SPI RAM not initialized` doesn't help you figure out what is wrong you could try add more logging here:

https://github.com/espressif/esp-idf/blob/master/components/esp32/esp_himem.c#L159

I would check which of `s_ram_descriptor,  s_range_descriptor` fail to allocate

nevercast, 2019-05-28:
Futher, the example you linked requires changing the sdkconfig. https://github.com/espressif/esp-idf/blob/master/examples/system/himem/sdkconfig.defaults

Last I checked, Platform.IO does not allow you to change the sdkconfig with the Arduino environment. This may have changed.

nevercast, 2019-05-28:
I have tried the example myself on my own inhouse WROVER board and it works.

```
rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6228
load:0x40078000,len:10180
ho 0 tail 12 room 4
load:0x40080400,len:6660
entry 0x40080764
I (30) boot: ESP-IDF v3.2-18-g977854975 2nd stage bootloader
I (30) boot: compile time 15:08:34
I (33) boot: Enabling RNG early entropy source...
I (36) boot: SPI Speed      : 40MHz
I (40) boot: SPI Mode       : DIO
I (44) boot: SPI Flash Size : 4MB
I (48) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (67) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (74) boot:  2 app0             OTA app          00 10 00010000 00140000
I (82) boot:  3 app1             OTA app          00 11 00150000 00140000
I (89) boot:  4 spiffs           Unknown data     01 82 00290000 00170000
I (97) boot: End of partition table
I (101) boot: No factory image, trying OTA 0
I (106) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1d7d8 (120792) map
I (157) esp_image: segment 1: paddr=0x0002d800 vaddr=0x3ffb0000 size=0x020c0 (  8384) load
I (160) esp_image: segment 2: paddr=0x0002f8c8 vaddr=0x40080000 size=0x00400 (  1024) load
I (164) esp_image: segment 3: paddr=0x0002fcd0 vaddr=0x40080400 size=0x00340 (   832) load
I (173) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x1e048 (122952) map
I (225) esp_image: segment 5: paddr=0x0004e068 vaddr=0x40080740 size=0x0eb84 ( 60292) load
I (250) esp_image: segment 6: paddr=0x0005cbf4 vaddr=0x50000000 size=0x0000c (    12) load
I (259) boot: Loaded app from partition at offset 0x10000
I (259) boot: Disabling RNG early entropy source...
I (263) spiram: Found 64MBit SPI RAM device
I (266) spiram: SPI RAM mode: flash 40m sram 40m
I (272) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (279) cpu_start: Pro cpu up.
I (283) cpu_start: Starting app cpu, entry point is 0x40081214
I (0) cpu_start: App cpu up.
I (1110) spiram: SPI SRAM memory test OK
I (1110) heap_init: Initializing. RAM available for dynamic allocation:
I (1111) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1117) heap_init: At 3FFB3688 len 0002C978 (178 KiB): DRAM
I (1123) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1129) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1136) heap_init: At 4008F2C4 len 00010D3C (67 KiB): IRAM
I (1142) cpu_start: Pro cpu start user code
I (1147) spiram: Adding pool of 3840K of external SPI memory to heap allocator
I (161) esp_himem: Initialized. Using last 8 32KB address blocks for bank switching on 4352 KB of physical memory.
I (163) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (172) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (272) spiram: Found 64MBit SPI RAM device
I (272) spiram: SPI RAM mode: flash 40m sram 40m
I (282) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
spiram size 8388608
himem free 4456448
himem phys 4456448
himem reserved 262144
```

nevercast, 2019-05-28:
I suspect you have not enabled spiram when building the firmware. Please detail how you're building the firmware and what config you may have changed.

jcw, 2019-05-28:
Many thanks for these comments - I will investigate further.

PS. How do you change ESP32's logging verbosity in the Arduino IDE or in PlatformIO? Also, how do I change the sdkconfig.h in the Arduino IDE? My apologies, I'm not familiar at all with ESP32 in the IDE.

jcw, 2019-05-28:
Fromm what I can tell, all the required settings are appropriate in the `arduino-esp32` setup:

https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/config/sdkconfig.h

jcw, 2019-05-28:
> I suspect you have not enabled spiram when building the firmware. Please detail how you're building the firmware and what config you may have changed.

In PlatformIO, I have the following:

```
$ cat platformio.ini
[env:esp32]
build_flags =
    -DWROVER
    -DBOARD_HAS_PSRAM
    -mfix-esp32-psram-cache-issue
platform = espressif32
board = esp-wrover-kit
framework = arduino
upload_port = /dev/cu.SLAB_USBtoUART
monitor_port = /dev/cu.SLAB_USBtoUART
monitor_speed = 115200
```

I've tried rebuilding in PlatformIO with these changes, but it made no difference:

```
/*#define CONFIG_SPIRAM_USE_CAPS_ALLOC 1*/
#define CONFIG_SPIRAM_USE_MALLOC 1
```

(in `~/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/config/sdkconfig.h`)

In Arduino IDE, i cloned `esspressig/arduino-esp32` from github, into the `hardware/espressif/esp32/` directory.

So my questions to you are: did you do your test in the Arduino IDE? And how do you get the verbose logging showing all the "I" messages at boot-up?

nevercast, 2019-05-28:
I am not using the Arduino IDE, I am using Arduino as an ESPIDF component. The Arduino IDE uses prebuilt libraries and does not compile the IDF, only your application code. Platform IO when I used it some months ago, did not work correctly in Arduino mode, only in ESP32 mode (The Arduino sdkconfig could not be modified, this was a big)

It seems you are using the espressif pio environment so you should be able to modify the config but in the end I stopped using Platform.IO for plain Arduino on IDF Component. This has been the most reliable

nevercast, 2019-05-28:
Note, sdkconfig and sdkconfig.h are not the same. Sdkconfig is used to generate the header file. Modifying the header file may not work as expected

jcw, 2019-05-28:
Aha, clear, thanks. Trying espidf in pio now. I can't switch away from pio (it's far too convenient). If himem can't be made to work under pio, then I'll have to stick with 4 MB SPI RAM. Not the end of the world, but it'd perhaps be good to leave this issue open, to await future enhancements.

Let's also ping @ivankravets for completeness ...

nevercast, 2019-05-28:
Are you using the paid features of PIO? If not, I can help you setup VSCode to work with Arduino as a component.

jcw, 2019-05-28:
Ok, for reference, here's what I get with `framework = espidf` under PlatformIO. First off, I had to change a few headers, the himem example does not match what pio has - these three changes fix it:

```
#include "esp_spiram.h"
#include "rom/cache.h"
#include "sdkconfig.h"  // <-- not changed
#include "esp_himem.h"
```

Then, linking fails:

```
Linking .pioenvs/esp32/firmware.elf
.pioenvs/esp32/libesp32.a(esp_himem.o):(.literal.esp_himem_get_phys_size+0x4): undefined reference to `esp_spiram_get_size'
.pioenvs/esp32/libesp32.a(esp_himem.o):(.literal.esp_himem_unmap+0x1c): undefined reference to `esp_spiram_writeback_cache'
.pioenvs/esp32/libesp32.a(esp_himem.o): In function `esp_himem_get_phys_size':
esp_himem.c:(.text.esp_himem_get_phys_size+0x3): undefined reference to `esp_spiram_get_size'
.pioenvs/esp32/libesp32.a(esp_himem.o): In function `esp_himem_unmap':
esp_himem.c:(.text.esp_himem_unmap+0x88): undefined reference to `esp_spiram_writeback_cache'
collect2: error: ld returned 1 exit status
*** [.pioenvs/esp32/firmware.elf] Error 1
```

Which is odd, since it does seem to be present:

```
$ grep -rl esp_spiram_get_size .
./.pioenvs/esp32/esp32/esp_himem.o
./.pioenvs/esp32/libesp32.a
```

jcw, 2019-05-28:
No, free pio. Using command line, not vscode/atom gui.

ivankravets, 2019-05-28:
Maybe this PR will help https://github.com/platformio/platform-espressif32/pull/142?

jcw, 2019-05-28:
Hm, espidf framework generates `src/sdkconfig.h` as part of the build, but no mention of SPIRAM defines whatsoever. Maybe the PIO frameworks are still lagging - here's what I see as latest updates:

* Updating framework-arduinoespressif32    @ 2.10002.190416 [Up-to-date]
* Updating framework-espidf                @ 3.30200.190418 [Up-to-date]

ivankravets, 2019-05-28:
As I remember we just copy a default `sdkconfig.h`. You can modify it to your requirements.

jcw, 2019-05-28:
Baby steps - still in framework espidf under pio - I added the following to `src/sdkconfig.h`:

```
#define CONFIG_ESP32_SPIRAM_SUPPORT 1
#define CONFIG_SPIRAM_BOOT_INIT 1
#define CONFIG_SPIRAM_IGNORE_NOTFOUND 0
/*
#define CONFIG_SPIRAM_USE_MALLOC 1
*/
#define CONFIG_SPIRAM_TYPE_AUTO 1
#define CONFIG_SPIRAM_SIZE -1
#define CONFIG_SPIRAM_SPEED_40M 1
#define CONFIG_SPIRAM_MEMTEST 1
#define CONFIG_SPIRAM_CACHE_WORKAROUND 1
#define CONFIG_SPIRAM_BANKSWITCH_ENABLE 1
#define CONFIG_SPIRAM_BANKSWITCH_RESERVE 4
```

Note: setting CONFIG_SPIRAM_USE_MALLOC breaks the re-build in several places.

The number of linker errors increases. For some reason, `spiram.c` is not being compiled in.

ivankravets, 2019-05-28:
Please file an issue here https://github.com/platformio/platform-espressif32/issues

TridentTD, 2019-06-25:
I have just created new repository for Arduino-ESP32 PSRAM Himem supported

Try at
https://github.com/TridentTD/arduino_esp32_himem

closed 2019-09-07 by stale[bot]

arkhipenko, 2021-07-15:
https://github.com/arkhipenko/ESP32Himem

Better late than never...