espressif/arduino-esp32 #2831: is there a std::mutex available for std::thread?

dsyleixa, 2019-05-28:
hello,
is there a std::mutex available for std::thread?
ref.:
http://www.cplusplus.com/reference/mutex/mutex/
https://de.cppreference.com/w/cpp/thread/mutex

atanisoft, 2019-05-28:
There likely is since it is in the c++ spec. Since the std::thread spec is based around pthread it is highly likely that std::mutex is based on pthread mutex objects as well.

On the esp32 it is advised to use FreeRTOS Mutex/Semaphore objects instead as that is more "native" and doesn't go through wrapper objects.

dsyleixa, 2019-05-28:
I was asking because when I use the code above I get the error
```
exit status 1
'mutex' in namespace 'std' does not name a type
```

I would appreciate nonetheless, if std::thread was fully supported by the std:: features  for C++11 compatibility

atanisoft, 2019-05-28:
From the above code sample you are missing the header that defines that template class. It would appear that adding:
```
#include <mutex>
```
might fix it.

dsyleixa, 2019-05-28:
oh yes, your right, I didn't expect that that would need an extra include!
Thanks a lot! 8-)

closed 2019-05-28 by dsyleixa