espressif/arduino-esp32 #2832: Backward compatibility

mytrain, 2019-05-28:
### Hardware:
Board:							TTGO with OLED 
Core Installation version:			git dowloaded yesterday
IDE name:						Arduino IDE
Flash Frequency:					80MHz
PSRAM enabled:                                     No
Upload Speed:				       115200
Computer OS:                                        Windows 

### Description:
SPPIFS, EEPROM and NVS are not compatible with older version (my previous version of my IDE was a more than one year old version with SDK v3.0).

Let me provide some contextual information. I developing  a  network of connected object connecting to a master node. When connecting ESP32 is automatically updated with a version of the firmware stored by the master node. This design avoids interface issues when the connected object goes from one master node to another one that is not configured with the same version.
As a consequence a connected object can be updated with a newer or older version depending on the master node it is connecting to. Every thing was fine as far as I did not update my IDE (hardware/espressif/esp32/...). 

With the newest version  I found the following compatibility issues:
1) EEPROM : located on nvs partition instead of specific partition. This makes retreival of old data impossible. However I can handle this point by keeping an old style default.csv and the old stype EEPROM.h/cpp code.

2) NVS : Someting changed. When updating to a newer version it seems that forward compatibility is well managed as my ESP32 restarts correctly. However in case of updating with an older version (3.0) I've got the following Exception
> assertion "false && "item should have been present in cache"" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_item_hash_list.cpp", line 85, function: void nvs::HashList::erase(size_t)
abort() was called at PC 0x400f0323 on core 0

3) SPIFFS : Introduction of METADATA broke the SPIFFS structure. I need to format in both cases (update with newer or older version). I can handle it by detection of a failure to read files but it requires extra time for downloading all SPIFFS files from the master node.
Is there any chance that SDKConfig option can remain unchanged in the future or that Arduino IDE introduce a way for recompliling SDK when changing option out of the box and without installing by myself the espressif IDF.

OK, this is not an conventional issue, and I'm not expecting a strait answer, but I wanted to open the debat on keeping this wonderfull piece of design (EPS32 on Arduino IDE) compatible with the past and the future.

Thanks

lbernstone, 2019-05-28:
1) I'll write a conversion method for EEPROM when I get some free time.
2&3) nvs and SPIFFS are handled by esp-idf.  They do intend for upgrades to both of these components to be seamless, particularly nvs which is pretty critical to proper operation of the device.  If you find that upgrades are causing problems, then please post full issues (with error logs and minimal sketches), so we can determine where the cause of the problem is and post it [upstream](https://github.com/espressif/esp-idf/issues) if that is the source of the problem.

lbernstone, 2019-05-29:
https://github.com/espressif/arduino-esp32/pull/2841

rljonesau, 2019-06-01:
I just updated to 1.0.3-rc1 last night.
I found my code would lock up if I tried to send a html file from SPIFFS to a connecting browser.
In WiFiClient, where it reads chunks of file to send, the file read size would always be 0 bytes, despite asking for 1360.

So after lots of debug prints etc getting deeper into cryptic file system code, I give up and come to the Arduino ESP32 repo here, find this message, re-format SPIFFS, and all is good again!

So yeah, there is some sort of backwards compatibility that is preventing file reads from progressing...

closed 2019-08-14 by stale[bot]