espressif/arduino-esp32 #2834: Idle hook callback

themindfactory, 2019-05-28:
I have created a hook to the idle task with [https://docs.espressif.com/projects/esp-idf/en/v3.0.7/api-reference/system/hooks.html#_CPPv231esp_register_freertos_idle_hook22esp_freertos_idle_cb_t]

It says to have the callback return true if you want the hook to be called once per tick... I assumed we are getting 1000ticks/second as portTICK_PERIOD_MS equates to 1.

However in a 1 second span the callback in my case is called 4000+ times.

```
bool idle() {
  static uint32_t cnt = 0;

  ++cnt;
  EVERYMS(1000) {
    Serial.println(cnt);
    cnt = 0;
  }

  return true; // true only call once per tick
}

esp_register_freertos_idle_hook(idle);
```

Currently using 1.0.0 of the arduino board support.

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.0
IDE name:						Arduino IDE
Flash Frequency:					80Mhz?
PSRAM enabled:                                     no
Upload Speed:					500000
Computer OS:                                        Windows 7

themindfactory, 2019-05-31:
Reason for the once per tick requirement is I was hoping to use it in a calculation to determine CPU usage... 1 call per tick is deterministic, otherwise it kind of a crap shot... (I think)

atanisoft, 2019-05-31:
Use esp_register_freertos_tick_hook_for_cpu instead. The idle hook will be called a lot more often, ie everytime the FreeRTOS idle tasks for a core runs.

You might also want to consider registering two hooks since there are two cores.

themindfactory, 2019-05-31:
If you look at the source code it will be called a lot, but I want it to call me once per tick, there are 1000 ticks per second, so if it calls me 900 times, in 1 second, the CPU load is 10%, as it was free 900/1000 times.... but for some reason its calling my idle function 1000+ times a second so this is confusing.

atanisoft, 2019-05-31:
You are hooking the IDLE callback and not the TICK callback. See my comment above which points you to the correct function that *WILL* be called once per tick...

themindfactory, 2019-05-31:
OK I will try that... it said in the docs that if your callback returned true it would only call it once per tick.... even if you used idle hook... 

https://github.com/pycom/pycom-esp-idf/blob/master/components/esp32/freertos_hooks.c#L51 

I will try the other function for ticks and see.... 

Thanks!

themindfactory, 2019-05-31:
If that works then I will do it for both CPU and I can track both % usages.

atanisoft, 2019-05-31:
> it said in the docs that if your callback returned true it would only call it once per tick.... even if you used idle hook...
> 
> https://github.com/pycom/pycom-esp-idf/blob/master/components/esp32/freertos_hooks.c#L51

The github link you provided is *NOT* ESP-IDF so it is not relevant to arduino-esp32 in any meaningful manner. However, the code looks similar to ESP-IDF for this piece and does not match what the ESP-IDF docs claim. Instead, if any of the idle hook functions return false it will prevent the ESP32 core from going into an IDLE/power-save state. For the IDLE hook support there is no means for only having it called once per tick. I'd suggest filing a documentation issue over on https://github.com/espressif/esp-idf so they can clarify / fix the ESP-IDF documentation for the IDLE callback.

themindfactory, 2019-05-31:
Yeah I see that now... I have now registers 2 callbacks one for each CPU and it telling me 1000 calls per second, so the CPU is at 0% load... now the question is am I incorrectly assuming the idle will only be called when the processor has time?? Is my logic flawed?

atanisoft, 2019-05-31:
Tick call back is called regardless of load. You would need to check the currently running tasks for each core to determine load. There are other options to determine load as well.

themindfactory, 2019-05-31:
So I was onto something using idle hook?

atanisoft, 2019-05-31:
not necessarily, the idle hook doesn't necessarily mean the cpu is idle.

themindfactory, 2019-05-31:
I am having much difficulty finding documentation that describes the exact operation of these functions.... idle task is lowest priority, so if other stuff with more priority is executing then it will get "left out", so I thought I would add a counter to the idle task, this counter printed out every second, I would assume as the processor got more busy with tasks like WIFI and serving pages.... I could see this counter descrease... but no... it seems to increase with load.... :-(

closed 2019-08-14 by stale[bot]