espressif/arduino-esp32 #2835: Using miniz

MarXoftNl, 2019-05-28:
My developing friends,

I request data from an api-server. Quite a lot... several kB's. So... Accept: gzip,deflate. Server wants to get rid of data asap, I'm not the only one requesting data.
So now I got this zipped data and I would really like to unzip it.
Is there any tutorial or example on how to use miniz, or zlib or whatever unzip-stuff there is for Arduino/ESP32.

Thank you in advance.

closed 2019-08-14 by stale[bot]

luisfear, 2021-11-20:
> My developing friends,
>
> I request data from an api-server. Quite a lot... several kB's. So... Accept: gzip,deflate. Server wants to get rid of data asap, I'm not the only one requesting data. So now [...]

Did you find a way to do it?

chegewara, 2021-11-20:
https://github.com/tobozo/ESP32-targz