espressif/arduino-esp32 #2839: Issue with ArduinoHttpClient and WiFi/WiFiClientSecure

brentru, 2019-05-29:
Getting a compile error with libraries involving ArduinoHTTPClient and the ESP32WiFi/WiFiClientSecure. These libraries, incl. Adafruit IO Arduino, had a similar issue with the ESP8266 Arduno BSP 2.5.0 before changes to WiFiClient were reverted in 2.5.1

----------------------------- Remove above -----------------------------

### Hardware:
Board:							Adafruit Huzzah ESP32
Core Installation version:			1.0.2
IDE name:						Arduino IDE
Flash Frequency:					80MHz

### Description:
Describe your problem here

### Sketch:  
```
Example in https://github.com/adafruit/Adafruit_IO_Arduino/tree/master/examples/adafruitio_00_publish
```

### Debug Messages:
```
Users/brentrubell/Documents/Arduino/libraries/Adafruit_IO_Arduino/src/wifi/AdafruitIO_ESP32.cpp: In constructor 'AdafruitIO_ESP32::AdafruitIO_ESP32(const char*, const char*, const char*, const char*)':
/Users/brentrubell/Documents/Arduino/libraries/Adafruit_IO_Arduino/src/wifi/AdafruitIO_ESP32.cpp:22:53: error: invalid new-expression of abstract class type 'HttpClient'
   _http = new HttpClient(*_client, _host, _http_port)=0;
                                                     ^
In file included from /Users/brentrubell/Documents/Arduino/libraries/ArduinoHttpClient/src/ArduinoHttpClient.h:8:0,
                 from /Users/brentrubell/Documents/Arduino/libraries/Adafruit_IO_Arduino/src/AdafruitIO.h:23,
                 from /Users/brentrubell/Documents/Arduino/libraries/Adafruit_IO_Arduino/src/wifi/AdafruitIO_ESP32.h:18,
                 from /Users/brentrubell/Documents/Arduino/libraries/Adafruit_IO_Arduino/src/wifi/AdafruitIO_ESP32.cpp:14:
/Users/brentrubell/Documents/Arduino/libraries/ArduinoHttpClient/src/HttpClient.h:41:7: note:   because the following virtual functions are pure within 'HttpClient':
 class HttpClient : public Client
       ^
In file included from /Users/brentrubell/Library/Arduino15/packages/esp32/hardware/esp32/1.0.2/cores/esp32/Arduino.h:157:0,
                 from /Users/brentrubell/Documents/Arduino/libraries/Adafruit_IO_Arduino/src/wifi/AdafruitIO_ESP32.h:17,
                 from /Users/brentrubell/Documents/Arduino/libraries/Adafruit_IO_Arduino/src/wifi/AdafruitIO_ESP32.cpp:14:
/Users/brentrubell/Library/Arduino15/packages/esp32/hardware/esp32/1.0.2/cores/esp32/Client.h:31:17: note: 	virtual int Client::connect(IPAddress, uint16_t, int)
     virtual int connect(IPAddress ip, uint16_t port, int timeout) =0;
                 ^
/Users/brentrubell/Library/Arduino15/packages/esp32/hardware/esp32/1.0.2/cores/esp32/Client.h:32:17: note: 	virtual int Client::connect(const char*, uint16_t, int)
     virtual int connect(const char *host, uint16_t port, int timeout) =0;
                 ^
Multiple libraries were found for "WiFi.h"
 Used: /Users/brentrubell/Library/Arduino15/packages/esp32/hardware/esp32/1.0.2/libraries/WiFi
 Not used: /Users/brentrubell/Documents/Arduino/libraries/WiFiNINA
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
Using library Adafruit_IO_Arduino at version 3.2.0 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_IO_Arduino 
Using library Adafruit_MQTT_Library at version 1.0.1 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_MQTT_Library 
Using library ArduinoHttpClient at version 0.4.0 in folder: /Users/brentrubell/Documents/Arduino/libraries/ArduinoHttpClient 
Using library WiFi at version 1.0 in folder: /Users/brentrubell/Library/Arduino15/packages/esp32/hardware/esp32/1.0.2/libraries/WiFi 
Using library WiFiClientSecure at version 1.0 in folder: /Users/brentrubell/Library/Arduino15/packages/esp32/hardware/esp32/1.0.2/libraries/WiFiClientSecure 
```

SRGDamia1, 2019-05-30:
It's this:  https://github.com/espressif/arduino-esp32/issues/2755

brentru, 2019-06-10:
Closing.  #2755 should be addressed by devs, thanks @SRGDamia1

closed 2019-06-10 by brentru