espressif/arduino-esp32 #2846: EEPROM, and WiFiClient issues with updated library .

tommakov, 2019-05-31:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git?
IDE name:						?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

lbernstone, 2019-05-31:
EEPROM was just me not knowing the spec.
https://github.com/espressif/arduino-esp32/blob/master/libraries/EEPROM/src/EEPROM.cpp#L114
-       memset(key_data, 0, size);
+       memset(key_data, 0xFF, size);

tommakov, 2019-06-02:
I changed the one line of code mentioned above and it did not fix the issue.  It still reads zeros.

lbernstone, 2019-06-02:
You'll need to wipe the EEPROM (or the device).  It only fills the empty space on init.

closed 2019-08-15 by stale[bot]