espressif/arduino-esp32 #2847: TouchRead crashes randomly

exocode, 2019-05-31:
### Hardware:
Board:							ESP32 DevKit V1 
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git?
IDE name:						PlatformIO, version 3.6.7
Upload Speed:					115200
Computer OS:                                        Mac OSX

### Description:
TouchRead example crashes randomly.

What could cause the issue?

### Sketch:  (leave the backquotes for [code formatting]

platformio.ini
```ini
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
monitor_speed = 115200
```

main.cpp
```cpp

#include "Arduino.h"

//ESP32 Touch Interrupt Test
// teo2017012.01
// Just test touch pin - Touch2 is T2 which is on GPIO 2 - Touch3 is T3 which is on GPIO 15.
// teo20170112.02
// Add interrupt handler

int threshold = 40;
int pin1 = T0;
int pin2 = T3;

void gotTouch()
{
    Serial.println("Touched1\n");
}

void gotTouch1()
{
    Serial.println("Touched2\n");
}
void setup()
{
    Serial.begin(115200);
    delay(1000); // give me time to bring up serial monitor
    printf("\n ESP32 Touch Interrupt Test\n");
    touchAttachInterrupt(pin1, gotTouch, threshold);
    touchAttachInterrupt(pin2, gotTouch1, threshold);
}

void loop()
{

    Serial.println(touchRead(pin1)); // get value using  T2

    Serial.println(touchRead(pin2)); // get value using T3

    Serial.println("");
    delay(1000);
}

```

### Debug Messages:
```

Touched1

Touched2

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40086252  PS      : 0x00060034  A0      : 0x80086a5c  A1      : 0x3ffbe140  
A2      : 0x3ffbec2c  A3      : 0x0000cdcd  A4      : 0xb33fffff  A5      : 0x00000001  
A6      : 0x00060021  A7      : 0x0000abab  A8      : 0x0000cdcd  A9      : 0x3ffbe140  
A10     : 0x00000003  A11     : 0x00060023  A12     : 0x00060021  A13     : 0x00000001  
A14     : 0x00060520  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x400e95fa  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40086252

Backtrace: 0x40086252:0x3ffbe140 0x40086a59:0x3ffbe170 0x40087a57:0x3ffbe190 0x4008772d:0x3ffbe1b0 0x400816c2:0x3ffbe1c0 0x400e95f7:0x3ffbbff0 0x400d545f:0x3ffbc010 0x40086a26:0x3ffbc030 0x40087969:0x3ffbc050

Core 1 register dump:
PC      : 0x40087f8e  PS      : 0x00060e34  A0      : 0x80086db7  A1      : 0x3ffbe680  
A2      : 0x3ffb81f0  A3      : 0x3ffb8074  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00060e23  A7      : 0x00000000  A8      : 0x3ffb8074  A9      : 0x3ffb8074  
A10     : 0x00000018  A11     : 0x00000018  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060e21  A15     : 0x00000000  SAR     : 0x00000015  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd  

Backtrace: 0x40087f8e:0x3ffbe680 0x40086db4:0x3ffbe6a0 0x40085863:0x3ffbe6c0 0x400d17b5:0x3ffbe700 0x400d0ce9:0x3ffbe720 0x400d0eda:0x3ffbe740 0x400d0f01:0x3ffbe760 0x400d0bbd:0x3ffbe780 0x40080f03:0x3ffbe7a0 0x400816b9:0x3ffbe7c0 0x400d17c1:0x3ffb1ea0 0x400d0ce9:0x3ffb1ec0 0x400d0eda:0x3ffb1ee0 0x400d0f5d:0x3ffb1f00 0x400d0ff0:0x3ffb1f50 0x400d1004:0x3ffb1f70 0x400d0c36:0x3ffb1f90 0x400d1c99:0x3ffb1fb0 0x40087969:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6412
entry 0x400806a8

 ESP32 Touch Interrupt Test
76
70

76
70

76
70
```

lbernstone, 2019-05-31:
Many issues here:
1) GPIO2 & 15 are [strapping pins](https://github.com/espressif/arduino-esp32/#esp32dev-board-pinmap) and may give you false changes immediately after boot.
2) Interrupt callbacks should be tagged with IRAM_ATTR
3) Don't interact with hardware or non-iram variables in ISRs.  Change a flag variable and do the processing in your main loop.

exocode, 2019-06-01:
Thank you for your response. I am a noob in ESP32 and have to investigate what all these terms mean. I am only a bit curious that the examples provided in this repo need such modifications.

lbernstone, 2019-06-01:
The examples are simplistic and just demonstrate the features.  TouchInterrupt.ino uses a flag variable as I recommended.

exocode, 2019-06-01:
Sorry, not able to see Point 2.) IRAM_ATTR in the example here: 

https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Touch/TouchInterrupt/TouchInterrupt.ino

lbernstone, 2019-06-01:
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/GPIO/GPIOInterrupt/GPIOInterrupt.ino#L12

closed 2019-08-14 by stale[bot]