espressif/arduino-esp32 #2852: can't use OTA until/unless you update from esp-idf

rayozzie, 2019-06-03:
Thank you for your great work on this project.

I am using native esp ota on Arduino.  It compiles just fine on my mac, which happens to have esp-idf installed.  However, without esp-idf (or when using platform.io), there are many compile errors.  The root cause appears to be that arduino-esp32 incorporates an older version of esp-idf.  (I'm not quite sure of how Arduino is finding the "correct" version when esp-idf is installed; perhaps search path.)

Here's but one of many examples of the root cause of the compile errors:

https://github.com/espressif/esp-idf/blob/master/components/bootloader_support/include/esp_flash_partitions.h

https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/bootloader_support/esp_flash_partitions.h

This has increased in priority because since adding OTA support this is now impacting downstream developers of my app.  And so, if there's information about when the new esp-idf will be incorporated, I'd very much appreciate knowing.

Thank you again for your time and efforts.

robouden, 2019-06-03:
Error on Arduino is:

dfu.cpp:35:5: error: 'esp_ota_img_states_t' was not declared in this scope
     esp_ota_img_states_t ota_state;
     ^
dfu.cpp:36:47: error: 'ota_state' was not declared in this scope
     if (esp_ota_get_state_partition(running, &ota_state) == ESP_OK) {
                                               ^
dfu.cpp:36:56: error: 'esp_ota_get_state_partition' was not declared in this scope
     if (esp_ota_get_state_partition(running, &ota_state) == ESP_OK) {
                                                        ^
dfu.cpp:37:26: error: 'ESP_OTA_IMG_PENDING_VERIFY' was not declared in this scope
         if (ota_state == ESP_OTA_IMG_PENDING_VERIFY) {
                          ^
dfu.cpp:42:56: error: 'esp_ota_mark_app_valid_cancel_rollback' was not declared in this scope
                 esp_ota_mark_app_valid_cancel_rollback();
                                                        ^
dfu.cpp:45:62: error: 'esp_ota_mark_app_invalid_rollback_and_reboot' was not declared in this scope
                 esp_ota_mark_app_invalid_rollback_and_reboot();

I hope this helps.

atanisoft, 2019-06-03:
Arduino-esp32 is pinned to the last stable release of esp-idf, v3.2, and includes a copy of it which is used for compilation in all environments.

OTA works fine with the v3.2 APIs, you may just need to adjust your code for the available functions. You can refer to the Update library available with arduino-esp32 as a reference for how to implement OTA.

rayozzie, 2019-06-03:
Ok, today I will investigate whether I can successfully back-port to the stable release.  

FYI, my code is a derivative of the “native” example, because no HTTP is involved. The firmware is dribbled over the course of weeks on low power cellular, and then updated offline.

atanisoft, 2019-06-03:
> FYI, my code is a derivative of the “native” example, because no HTTP is involved. The firmware is dribbled over the course of weeks on low power cellular, and then updated offline.

The Update code can be done from offline, refer to this sample: https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/examples/SD_Update/SD_Update.ino

This uses the SD as storage and source for the updated firmware. You can substitute other storage options in the same manner.

robouden, 2019-06-03:
atanisoft,

Thanks for the suggestion. 
Let's hope Ray can fix the issue.

regards
rob

PaulEverblix, 2019-06-07:
Hello !

I'm having the same issue as you. Working with the Arduino Core for ESP32 on PlatformIO, I have compiling errors.
When creating a new project, it compiles with no problem.
However, when I enable the rollback feature (implemented in the 1.8 release of the core), I get the same errors as you.
To enable it, I alter the .platformio\packages\framework-arduinoespressif32\tools\sdk\include\config\sdkconfig.h file, adding #define CONFIG_APP_ROLLBACK_ENABLE
It appears that the files in .platformio\packages\framework-arduinoespressif32\tools\sdk\include\bootloader_support are not the latest and do not include the mandatory classes and enums.

OTA works fine in 1.7, but do lack the rollback feature, that I need.
Have you found a way to avoid the problem and get your code to compile ?
Thanks a lot.

rayozzie, 2019-06-07:
I'm closing the issue because I see clearly now that I will just need to wait until arduino-esp32 incorporates a stable release later than v3.2.  I have added conditional compilation of the problematic code so as to work around the problem.  Thank you.

closed 2019-06-07 by rayozzie