espressif/arduino-esp32 #2853: pAdvertising->stop(); Don't work.

m0923678421, 2019-06-03:
### Hardware:                                         ESP32-S/ESP-WROOM-32
Board:							ESP32 Dev Module
Core Installation version:			1.0.1
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                             yes
Upload Speed:					921600
Computer OS:                                        Windows 7

### Description:
I use the BLE_iBeacon example and disable deep sleep at that sketch.
But even the code run over "pAdvertising->stop();" this line.
I can still scan this iBeacon device with smartphone.
I also try btStop(); , btStart(); and reStart(); function.
But when the code run over btStop(); again.
ESP32 will reboot itself.
Help me please.

How can I stop iBeacon and restart in sketch without reboot?

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

/*
   Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp
   Ported to Arduino ESP32 by pcbreflux
*/

 
/*
   Create a BLE server that will send periodic iBeacon frames.
   The design of creating the BLE server is:
   1. Create a BLE Server
   2. Create advertising data
   3. Start advertising.
   4. wait
   5. Stop advertising.
   6. deep sleep
   
*/
#include "sys/time.h"

#include "BLEDevice.h"
#include "BLEUtils.h"
#include "BLEBeacon.h"
#include "esp_sleep.h"

#define GPIO_DEEP_SLEEP_DURATION     10  // sleep x seconds and then wake up
RTC_DATA_ATTR static time_t last;        // remember last boot in RTC Memory
RTC_DATA_ATTR static uint32_t bootcount; // remember number of boots in RTC Memory

#ifdef __cplusplus
extern "C" {
#endif

uint8_t temprature_sens_read();
//uint8_t g_phyFuns;

#ifdef __cplusplus
}
#endif

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/
BLEAdvertising *pAdvertising;
struct timeval now;

#define BEACON_UUID           "8ec76ea3-6668-48da-9866-75be8bc86f4d" // UUID 1 128-Bit (may use linux tool uuidgen or random numbers via https://www.uuidgenerator.net/)

void setBeacon() {

  BLEBeacon oBeacon = BLEBeacon();
  oBeacon.setManufacturerId(0x4C00); // fake Apple 0x004C LSB (ENDIAN_CHANGE_U16!)
  oBeacon.setProximityUUID(BLEUUID(BEACON_UUID));
  oBeacon.setMajor((bootcount & 0xFFFF0000) >> 16);
  oBeacon.setMinor(bootcount&0xFFFF);
  BLEAdvertisementData oAdvertisementData = BLEAdvertisementData();
  BLEAdvertisementData oScanResponseData = BLEAdvertisementData();
  
  oAdvertisementData.setFlags(0x04); // BR_EDR_NOT_SUPPORTED 0x04
  
  std::string strServiceData = "";
  
  strServiceData += (char)26;     // Len
  strServiceData += (char)0xFF;   // Type
  strServiceData += oBeacon.getData(); 
  oAdvertisementData.addData(strServiceData);
  
  pAdvertising->setAdvertisementData(oAdvertisementData);
  pAdvertising->setScanResponseData(oScanResponseData);

}

void setup() {

    
  Serial.begin(115200);
  gettimeofday(&now, NULL);

  Serial.printf("start ESP32 %d\n",bootcount++);

  Serial.printf("deep sleep (%lds since last reset, %lds since last boot)\n",now.tv_sec,now.tv_sec-last);

  last = now.tv_sec;
  
  // Create the BLE Device
  BLEDevice::init("");

  // Create the BLE Server
  // BLEServer *pServer = BLEDevice::createServer(); // <-- no longer required to instantiate BLEServer, less flash and ram usage

  pAdvertising = BLEDevice::getAdvertising();
  
  setBeacon();
   // Start advertising
  pAdvertising->start();
  Serial.println("Advertizing started...");
  delay(100);
  pAdvertising->stop();
  /*
  Serial.printf("enter deep sleep\n");
  esp_deep_sleep(1000000LL * GPIO_DEEP_SLEEP_DURATION);
  Serial.printf("in deep sleep\n");
  */
}

void loop() {
}
```

### Debug Messages:
```
[W][esp32-hal-psram.c:30] psramInit(): PSRAM init failed!
start ESP32 0
deep sleep (0s since last reset, 0s since last boot)
[I][BLEDevice.cpp:561] getAdvertising(): create advertising
[D][BLEDevice.cpp:563] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:152] setAdvertisementData(): >> setAdvertisementData
[D][BLEAdvertising.cpp:160] setAdvertisementData(): << setAdvertisementData
[D][BLEDevice.cpp:563] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:169] setScanResponseData(): >> setScanResponseData
[D][BLEAdvertising.cpp:479] handleGAPEvent(): handleGAPEvent [event no: 4]
[E][BLEAdvertising.cpp:174] setScanResponseData(): esp_ble_gap_config_scan_rsp_data_raw: 258 Unknown ESP_ERR error
[D][BLEAdvertising.cpp:177] setScanResponseData(): << setScanResponseData
[D][BLEAdvertising.cpp:186] start(): >> start: customAdvData: 1, customScanResponseData: 1
[D][BLEAdvertising.cpp:205] start(): - no services advertised
[D][BLEAdvertising.cpp:246] start(): << start
[D][BLEDevice.cpp:563] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:479] handleGAPEvent(): handleGAPEvent [event no: 6]
[D][BLEAdvertising.cpp:256] stop(): >> stop
[D][BLEAdvertising.cpp:262] stop(): << stop
[D][BLEDevice.cpp:563] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:479] handleGAPEvent(): handleGAPEvent [event no: 17]
[I][BLEAdvertising.cpp:495] handleGAPEvent(): STOP advertising
[D][BLEAdvertising.cpp:186] start(): >> start: customAdvData: 1, customScanResponseData: 1
[D][BLEAdvertising.cpp:205] start(): - no services advertised
[D][BLEAdvertising.cpp:246] start(): << start
[D][BLEDevice.cpp:563] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:479] handleGAPEvent(): handleGAPEvent [event no: 6]
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10088
load:0x40080400,len:6380
entry 0x400806a4
[W][esp32-hal-psram.c:30] psramInit(): PSRAM init failed!
start ESP32 0
deep sleep (0s since last reset, 0s since last boot)
[I][BLEDevice.cpp:561] getAdvertising(): create advertising
[D][BLEDevice.cpp:563] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:152] setAdvertisementData(): >> setAdvertisementData
[D][BLEAdvertising.cpp:160] setAdvertisementData(): << setAdvertisementData
[D][BLEDevice.cpp:563] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:169] setScanResponseData(): >> setScanResponseData
[D][BLEAdvertising.cpp:479] handleGAPEvent(): handleGAPEvent [event no: 4]
[E][BLEAdvertising.cpp:174] setScanResponseData(): esp_ble_gap_config_scan_rsp_data_raw: 258 Unknown ESP_ERR error
[D][BLEAdvertising.cpp:177] setScanResponseData(): << setScanResponseData
[D][BLEAdvertising.cpp:186] start(): >> start: customAdvData: 1, customScanResponseData: 1
[D][BLEAdvertising.cpp:205] start(): - no services advertised
[D][BLEAdvertising.cpp:246] start(): << start
[D][BLEDevice.cpp:563] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:479] handleGAPEvent(): handleGAPEvent [event no: 6]
[D][BLEAdvertising.cpp:256] stop(): >> stop
[D][BLEAdvertising.cpp:262] stop(): << stop
[D][BLEDevice.cpp:563] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:479] handleGAPEvent(): handleGAPEvent [event no: 17]
[I][BLEAdvertising.cpp:495] handleGAPEvent(): STOP advertising
[D][BLEAdvertising.cpp:186] start(): >> start: customAdvData: 1, customScanResponseData: 1
[D][BLEAdvertising.cpp:205] start(): - no services advertised
[D][BLEAdvertising.cpp:246] start(): << start
[D][BLEDevice.cpp:563] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:479] handleGAPEvent(): handleGAPEvent [event no: 6]

```

m0923678421, 2019-06-03:
### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
/*
   Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp
   Ported to Arduino ESP32 by pcbreflux
*/

/*
   Create a BLE server that will send periodic iBeacon frames.
   The design of creating the BLE server is:
   1. Create a BLE Server
   2. Create advertising data
   3. Start advertising.
   4. wait
   5. Stop advertising.
   6. deep sleep

*/
#include "sys/time.h"

#include "BLEDevice.h"
#include "BLEUtils.h"
#include "BLEBeacon.h"
#include "esp_sleep.h"

#define GPIO_DEEP_SLEEP_DURATION     10  // sleep x seconds and then wake up
RTC_DATA_ATTR static time_t last;        // remember last boot in RTC Memory
RTC_DATA_ATTR static uint32_t bootcount; // remember number of boots in RTC Memory

#ifdef __cplusplus
extern "C" {
#endif

uint8_t temprature_sens_read();
//uint8_t g_phyFuns;

#ifdef __cplusplus
}
#endif

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/
BLEAdvertising *pAdvertising;
struct timeval now;

#define BEACON_UUID           "8ec76ea3-6668-48da-9866-75be8bc86f4d" // UUID 1 128-Bit (may use linux tool uuidgen or random numbers via https://www.uuidgenerator.net/)

unsigned long lastTime = 0;
void setBeacon() {

  BLEBeacon oBeacon = BLEBeacon();
  oBeacon.setManufacturerId(0x4C00); // fake Apple 0x004C LSB (ENDIAN_CHANGE_U16!)
  oBeacon.setProximityUUID(BLEUUID(BEACON_UUID));
  oBeacon.setMajor((bootcount & 0xFFFF0000) >> 16);
  oBeacon.setMinor(bootcount & 0xFFFF);
  BLEAdvertisementData oAdvertisementData = BLEAdvertisementData();
  BLEAdvertisementData oScanResponseData = BLEAdvertisementData();

  oAdvertisementData.setFlags(0x04); // BR_EDR_NOT_SUPPORTED 0x04

  std::string strServiceData = "";

  strServiceData += (char)26;     // Len
  strServiceData += (char)0xFF;   // Type
  strServiceData += oBeacon.getData();
  oAdvertisementData.addData(strServiceData);

  pAdvertising->setAdvertisementData(oAdvertisementData);
  pAdvertising->setScanResponseData(oScanResponseData);

[... 175 lines cut ...]
[D][BLEDevice.cpp:563] getAdvertising(): get advertising
[E][BLEAdvertising.cpp:174] setScanResponseData(): esp_ble_gap_config_scan_rsp_data_raw: 258 Unknown ESP_ERR error
[D][BLEAdvertising.cpp:479] handleGAPEvent(): handleGAPEvent [event no: 4]
[D][BLEAdvertising.cpp:177] setScanResponseData(): << setScanResponseData
[D][BLEAdvertising.cpp:186] start(): >> start: customAdvData: 1, customScanResponseData: 1
[D][BLEAdvertising.cpp:205] start(): - no services advertised
[D][BLEAdvertising.cpp:246] start(): << start
[D][BLEDevice.cpp:563] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:479] handleGAPEvent(): handleGAPEvent [event no: 6]
[D][BLEAdvertising.cpp:256] stop(): >> stop
[D][BLEAdvertising.cpp:262] stop(): << stop
[D][BLEAdvertising.cpp:152] setAdvertisementData(): >> setAdvertisementData
[D][BLEAdvertising.cpp:160] setAdvertisementData(): << setAdvertisementData
[D][BLEAdvertising.cpp:169] setScanResponseData(): >> setScanResponseData
[D][BLEDevice.cpp:563] getAdvertising(): get advertising
[E][BLEAdvertising.cpp:174] setScanResponseData(): esp_ble_gap_config_scan_rsp_data_raw: 258 Unknown ESP_ERR error
[D][BLEAdvertising.cpp:479] handleGAPEvent(): handleGAPEvent [event no: 17]
[D][BLEAdvertising.cpp:177] setScanResponseData(): << setScanResponseData
[I][BLEAdvertising.cpp:495] handleGAPEvent(): STOP advertising
[D][BLEAdvertising.cpp:186] start(): >> start: customAdvData: 1, customScanResponseData: 1
[D][BLEAdvertising.cpp:186] start(): >> start: customAdvData: 1, customScanResponseData: 1
[D][BLEAdvertising.cpp:205] start(): - no services advertised
[D][BLEAdvertising.cpp:246] start(): << start

```

taichunmin, 2019-07-08:
I have same issue. I find a link about that. Can we fix that?
https://github.com/nkolban/esp32-snippets/issues/797

taichunmin, 2019-09-06:
Please fix this bug.

beegee-tokyo, 2019-09-06:
Fixed already with #3034

closed 2019-11-19 by stale[bot]