espressif/arduino-esp32 #2854: AP-STA Mode, Static IP as STA, DHCP as AP

UnmeiArcana, 2019-06-04:
### Hardware:
Board:							Wroom, Heltec, ttgo,... ugh... We have truckloads of varied ESP32 devices here being involved in this issue.
Core Installation version:			Not sure, but all of our Espressif and Arduino related software are up to date
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Multiple computers on work, only Windows versions though.

### Description:
So we have many different ESP32 boards here in AP-STA mode. As soon as you grab a static IP as STA, your device loses it's ability to assign IP to any devices connected to it's AP via DHCP.

If you try doing WiFi.config(0,0,0) thing, it allows connected STA devices to have IPs assigned to them but this time the assigning device immediately loses it's static IP on other network. 

Is there a workaround I'm missing? And no there are no external routers involved in this to help with static IP assigning. This is a network composed of an army of ESP32 devices.

me-no-dev, 2019-06-05:
Could you please give a minimum example that we can have a go at? There are some fixes to be added soon, want to make sure that this is covered :)

UnmeiArcana, 2019-06-06:
I was already planning to write an example that can reproduce the issue but still couldn't find an opportunity. I'll get to it whenever I can.

UnmeiArcana, 2019-06-08:
@me-no-dev 
There we go, due to complication of issue this was as minimum as I could get. Also sorry for delay as I was quite busy the past few days.

Also the devices I used to reproduce the error with this example were Heltec WiFi LoRa 32 as DEVICE1 and WROOM-32 as DEVICE2 and DEVICE3 if it helps, but it doesn't really matter which devices are involved.

```
#include <WiFi.h>

//Only enable the device you are compiling for, to easily configure devices
#define DEVICE1 //Server Device: where it connects to as STA doesn't matter. Will act as a place for Device2 to connect to with static IP
//#define DEVICE2 //Middle Device: will simulate an internal device of network who should grab a static IP from our Server
//#define DEVICE3 //End Point Device: will simulate an outsider without a static ip connecting to Middle device and asking DHCP for an IP address

//Submask won't change in any way, we will also use same password for all SSIDs to avoid confusion
  IPAddress Submask = IPAddress(255,255,255,0);
  String InternalPASS = "TestPassword";

//We will define which IP to assign to a device and their SSIDs differently
//We will also tell which SSID they are supposed to connect to

#ifdef DEVICE1
  IPAddress DeviceIP = IPAddress(192,168,4,1);
  String DeviceSSID = "ESP32DEVICE1";
  String TargetDeviceSSID = ""; //Doesn't Really matter, just feed it any SSID to connect to if you want, can stay empty
  String TargetDevicePASS = ""; //Give it a password too if you want it to connect somewhere
#endif

#ifdef DEVICE2
  IPAddress DeviceIP = IPAddress(192,168,4,2);
  IPAddress TargetGatewayIP = IPAddress(192,168,4,1);
  String DeviceSSID = "ESP32DEVICE2";
  String TargetDeviceSSID = "ESP32DEVICE1";
#endif

#ifdef DEVICE3
  IPAddress DeviceIP = IPAddress(192,168,4,3);
  String DeviceSSID = "ESP32DEVICE3";
  String TargetDeviceSSID = "ESP32DEVICE2";
#endif

void setup() {
  Serial.begin(115200);
  WiFi.mode(WIFI_AP_STA);
  WiFi.disconnect(true,true);

  WiFi.softAPConfig(DeviceIP, DeviceIP, Submask); //IP,Gateway,SubnetMask
  WiFi.softAP(DeviceSSID.c_str(),InternalPASS.c_str(),1,0,4); //SSID,PASS,Channel,HideSSID,MaxConnections

//Device2 will ask for a static ip from Device1, at this point Device2 should lose it's ability to assign ip address as AP using DHCP
#ifdef DEVICE2
  if (!WiFi.config(DeviceIP, TargetGatewayIP, Submask)) { //IP,Gateway,SubnetMask
    Serial.println("Wifi - STA Failed to configure. Restarting...");
    delay(100);
    ESP.restart();
  }
#endif

//DEVICE1 may have another SSID so allowing it to use different password configured above
#ifdef DEVICE1 
  WiFi.begin(TargetDeviceSSID.c_str(), TargetDevicePASS.c_str());
#else
  WiFi.begin(TargetDeviceSSID.c_str(), InternalPASS.c_str());
#endif
}

void loop() {
#ifdef DEVICE1
  Serial.println("Me just a happy device running as server.");
#else
  Serial.println(WiFi.localIP());
  if (WiFi.localIP()[0] == 0){
    Serial.println("Me an unhappy device");
  }else{
    Serial.println("Me a happy device");
  }
#endif
  
  Serial.println();
  delay(5000);
}
```

UnmeiArcana, 2019-08-08:
I don't know why such a bot is added to clear non solved issues.
I am still patiently waiting for any feedback or fixes on this.

closed 2019-10-21 by stale[bot]

Sventecc, 2025-03-15:
Hello,
i have the same problem with my ESP32C6.
Has anyone been able to solve this?