espressif/arduino-esp32 #2855: avr/io.h not found

chrisaclark, 2019-06-04:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do. to complie a simple sketch that includes FreeRTOS
Describe your system( Hardware, computer, O/S, core version, environment) ESP32 WROOM
Describe what is failing Compilation
Show the shortest possible code that will duplicate the erro
r#include <Arduino_FreeRTOS.h>
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
Show the EXACT error message(it doesn't work is not enough)
Arduino: 1.8.7 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\chris\AppData\Local\Arduino15\packages -hardware C:\Users\chris\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\chris\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\chris\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10807 -build-path C:\Users\chris\AppData\Local\Temp\arduino_build_150115 -warnings=default -build-cache C:\Users\chris\AppData\Local\Temp\arduino_cache_886809 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\chris\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\chris\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\chris\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\chris\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\chris\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\chris\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -verbose C:\Users\chris\Documents\Arduino\sketch_jun04b\sketch_jun04b.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\chris\AppData\Local\Arduino15\packages -hardware C:\Users\chris\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\chris\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\chris\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10807 -build-path C:\Users\chris\AppData\Local\Temp\arduino_build_150115 -warnings=default -build-cache C:\Users\chris\AppData\Local\Temp\arduino_cache_886809 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\chris\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\chris\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\chris\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\chris\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\chris\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\chris\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -verbose C:\Users\chris\Documents\Arduino\sketch_jun04b\sketch_jun04b.ino
Using board 'esp32' from platform in folder: C:\Users\chris\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2
Using core 'esp32' from platform in folder: C:\Users\chris\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2
Detecting libraries used...
"C:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/config" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/app_trace" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/app_update" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/asio" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/bootloader_support" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/bt" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/coap" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/console" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/driver" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp-tls" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp32" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_event" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_http_client" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_http_server" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_https_ota" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_ringbuf" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/ethernet" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/expat" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/fatfs" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/freemodbus" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/freertos" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/heap" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/idf_test" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/jsmn" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/json" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/libsodium" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/log" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/lwip" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/mbedtls" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/mdns" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/micro-ecc" "-IC:\\Users\\chris\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/in [... cut ...]
If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module
Core Installation/update date:			?11/jul/2017?
IDE name:							Arduino: 1.8.7 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

### Description:
It seems to try to pickup from the AVR libraries

### Sketch:  See above

lbernstone, 2019-06-04:
FreeRTOS is compiled into the arduino-esp32 libraries.  If needed, include the headers in the [tree](https://github.com/espressif/arduino-esp32/tree/master/tools/sdk/include/freertos/freertos), although the main functions should already be available from [Arduino.h](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Arduino.h#L32-L34)

chrisaclark, 2019-06-04:
Many thanks for the speedy reply.  Apolgies but I am not quite sure what you mean by including the headers or the reference to arduino.h.  I have tried it again below but still the same error.  

It is referenced by FreeRTOSVariant.h with a 2019 copyright 

#include <Arduino.h>
#include <FreeRTOS.h>
#include <FreeRTOSConfig.h>
#include <Arduino_FreeRTOS.h>
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}

lbernstone, 2019-06-04:
Just remove the `#include <Arduino_FreeRTOS.h>` from your code.  FreeRTOS is already available in arduino-esp32.

chrisaclark, 2019-06-04:
Thanks goes on further now when I put the whole sketch in.

closed 2019-08-17 by stale[bot]