espressif/arduino-esp32 #2856: SYSTEM_EVENT_STA_GOT_IP shots twice when setting Static IP.

luisubieda, 2019-06-04:
----------------------------- Remove above -----------------------------

### Hardware:
Board:							?node32?
Core Installation version:			?1.0.2?
IDE name:						?Arduino IDE?
Flash Frequency:					?80MHz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10?

### Description:
Guys, 
I'd like to share the issue I'm having looking for some advice/help you could provide: 

The project I'm working on consists of a ESP32 mostly on Deep-Sleep with periodical wake-up intervals. 
I'm using the WiFiClientEvents example as starting point.

The scheme I'm implementing right now is the following: 
Boot (configure everything and start WiFi) -> Do nothing but waiting for "SYSTEM_EVENT_STA_GOT_IP" -> within that event, perform all the application-related actions and disconnect -> wait for "SYSTEM_EVENT_STA_DISCONNECTED" and within that event, go to deep-sleep and wake-up after N seconds.

The thing is that when I assign a Static IP for AP connection, the "SYSTEM_EVENT_STA_GOT_IP" gets triggered twice every time. The problem with that is that the actions I'm trying to perform during wake-up time, are repeated, and that messes up everything. Is this caused by any WiFi misconfiguration I could fix? 

Some information that could help troubleshooting: 
- The double-event triggering has nothing to do with Deep-Sleep, since I tested the system without going to sleep periodically and the double-event persists.
- It only happens with Static-IP, since going with DHCP works normally.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

/*

*What it does:*
- Turns on, Connects to AP.
- Sleeps for 5 seconds and gets back on.

*Sources:*
- Espressif's WiFi Client Events example: https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClientEvents/WiFiClientEvents.ino
- Espressif's Deepsleep w. TimerWakeUp example: https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/DeepSleep/TimerWakeUp/TimerWakeUp.ino

*Instructions to run:*
- Unzip in Arduino's root-folder.
- Open Arduino IDE
- Set WiFi credentials properly
- Set correct USB Port and Board set to "Node32s".
- Flash it (we had to press BOOT button for it to do it correctly).
- Open Serial terminal set to 115200 bps and evaluate its performance.
*/

/*
* WiFi Events Table

0  SYSTEM_EVENT_WIFI_READY               < ESP32 WiFi ready
1  SYSTEM_EVENT_SCAN_DONE                < ESP32 finish scanning AP
2  SYSTEM_EVENT_STA_START                < ESP32 station start
3  SYSTEM_EVENT_STA_STOP                 < ESP32 station stop
4  SYSTEM_EVENT_STA_CONNECTED            < ESP32 station connected to AP
5  SYSTEM_EVENT_STA_DISCONNECTED         < ESP32 station disconnected from AP
6  SYSTEM_EVENT_STA_AUTHMODE_CHANGE      < the auth mode of AP connected by ESP32 station changed
7  SYSTEM_EVENT_STA_GOT_IP               < ESP32 station got IP from connected AP
8  SYSTEM_EVENT_STA_LOST_IP              < ESP32 station lost IP and the IP is reset to 0
9  SYSTEM_EVENT_STA_WPS_ER_SUCCESS       < ESP32 station wps succeeds in enrollee mode
10 SYSTEM_EVENT_STA_WPS_ER_FAILED        < ESP32 station wps fails in enrollee mode
11 SYSTEM_EVENT_STA_WPS_ER_TIMEOUT       < ESP32 station wps timeout in enrollee mode
12 SYSTEM_EVENT_STA_WPS_ER_PIN           < ESP32 station wps pin code in enrollee mode
13 SYSTEM_EVENT_AP_START                 < ESP32 soft-AP start
14 SYSTEM_EVENT_AP_STOP                  < ESP32 soft-AP stop
15 SYSTEM_EVENT_AP_STACONNECTED          < a station connected to ESP32 soft-AP
16 SYSTEM_EVENT_AP_STADISCONNECTED       < a station disconnected from ESP32 soft-AP
17 SYSTEM_EVENT_AP_STAIPASSIGNED         < ESP32 soft-AP assign an IP to a connected station
[... 158 lines cut ...]
load:0x40080400,len:5868
entry 0x4008069c

[WiFi-event] event: 0 -> WiFi interface ready
[WiFi-event] event: 2 -> WiFi client started
[WiFi-event] event: 7 -> Obtained IP address
Channel: 1
IP Addr: 172.20.10.2
Gateway: 172.20.10.1
Subnet Mask: 255.255.255.240
DNS: 8.8.4.4
BSSID: 76:33:CB:8E:FC:82
Disconnecting from WiFi...
[WiFi-event] event: 7 -> Obtained IP address
Channel: 1
IP Addr: 172.20.10.2
Gateway: 172.20.10.1
Subnet Mask: 255.255.255.240
DNS: 8.8.4.4
BSSID: 
Disconnecting from WiFi...
[WiFi-event] event: 5 -> Disconnected from WiFi access point
Going to sleep... Coming back in 5 Seconds
Wake duration: 964 ms 
```

lbernstone, 2019-06-04:
I think you are probably seeing an autoreconnect attempt.  If you want to stay offline, set `WiFi.setAutoReconnect(false)`.  Otherwise, turn logging level to verbose so you can see the full transaction with the AP.

luisubieda, 2019-06-04:
@lbernstone Thanks for your reply. Did try the ``WiFi.setAutoReconnect(false)`` but the issue didn't stop. Here I'm posting the Serial Output for the Sketch runnin with **Core Debug Level to Verbose**.

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c

[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[WiFi-event] event: 2 -> WiFi client started
[WiFi-event] event: 0 -> WiFi interface ready
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:379] _eventCallback(): STA IP: 172.20.10.2, MASK: 255.255.255.240, GW: 172.20.10.1
Disconnecting from WiFi...
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 7 - STA_GOT_IP
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 8 - ASSOC_LEAVE
Going to sleep... Coming back in 5 Seconds
Wake duration: 970 ms

Also, here's the output as well **without sending the ESP32 to deep-sleep**: 

Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c

[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[WiFi-event] event: 2 -> WiFi client started
[WiFi-event] event: 0 -> WiFi interface ready
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:379] _eventCallback(): STA IP: 172.20.10.2, MASK: 255.255.255.240, GW: 172.20.10.1
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:379] _eventCallback(): STA IP: 172.20.10.2, MASK: 255.255.255.240, GW: 172.20.10.1

pglen, 2019-06-25:
I had a similar occurrence in 'C', the client connection reported twice. The error was that the event loop got initialized twice. I suspect something similar going on here with the onevent / config combo.

closed 2019-09-07 by stale[bot]

tokra, 2020-12-19:
i am experiencing this issue too. anyone solved it ?

rhamnett, 2021-02-15:
I get this problem as well when using a static IP

pglen, 2021-02-16:
The newer IDFs do not have this error (3+)

rhamnett, 2021-02-17:
I needed to use 
```
SYSTEM_EVENT_AP_STAIPASSIGNED
```

pglen, 2021-02-17:
Glad to hear it!

juliangomp, 2021-02-25:
Same problem here, I tried this:

> I needed to use
> 
> ```
> SYSTEM_EVENT_AP_STAIPASSIGNED
> ```

but the problem persists.

rhamnett, 2021-02-25:
SYSTEM_EVENT_AP_STAIPASSIGNED replaces SYSTEM_EVENT_STA_GOT_IP you should only get one message @juliangomp

radzimir, 2024-02-17:
I can reproduce SYSTEM_EVENT_STA_GOT_IP by restarting my AP, which simulates weak, flaky signal.