espressif/arduino-esp32 #2858: How can I remap TX2 and RX2 to different pins ?

GeorgeFlorian, 2019-06-05:
OS: Linux Mint 19.1
Board: ESP32 Wrover-B DevKitV4

Hello !

I have a ESP32 Wrover-B DevKitV4 that has GPIO 16 and 17, meaning TX2 and RX2, NOT CONNECTED. But I need to use them with a GPS Module.

I've just learned that I can remap the GPIOs. But I have no idea how.

I feel like I should mention that I've never used ESP-IDF. I used Arduino IDE, but I've since moved to PlatformIO IDE.

YAFIYET, 2019-06-08:
I have an experiance with the ESP32 wich is not same others .

**Step 1**: Find the concerned  HardwareSerial.cpp file in your PC.
The path was like that for my case --> C:\Users\XXXXX\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1\cores\esp32 
**Step 2**: Modify pin numbers as you need (please find right pin numbers from your board details) in this file:
    #define RX2 16 -->25
    #define TX2 17 -->27
**Step 3** : Save and exit from cpp file and restart your arduino ide.
I hope it will solve your problem.

luc-github, 2019-06-08:
why not just define them in begin ?
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.h#L58
` void begin(unsigned long baud, uint32_t config=SERIAL_8N1, int8_t rxPin=-1, int8_t txPin=-1, bool invert=false, unsigned long timeout_ms = 20000UL);`

YAFIYET, 2019-06-08:
Hello
These ports are hardware serial and its coded Hardwareserial.cpp file.
You mean softwareserial. You can do that (attention port selection) if you select it.
The question is why Hardwareserial we are choosing? Its more performance than software serial.
BR
yafiyet@gmail.com<mailto:yafiyet@gmail.com>

Windows 10 için Posta<https://go.microsoft.com/fwlink/?LinkId=550986> ile gönderildi

________________________________
Gönderen: Luc <notifications@github.com>
Gönderildi: Saturday, June 8, 2019 2:58:02 PM
Kime: espressif/arduino-esp32
Bilgi: yafiyet; Comment
Konu: Re: [espressif/arduino-esp32] How can I remap TX2 and RX2 to different pins ? (#2858)

why not just define them in begin ?
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.h#L58
void begin(unsigned long baud, uint32_t config=SERIAL_8N1, int8_t rxPin=-1, int8_t txPin=-1, bool invert=false, unsigned long timeout_ms = 20000UL);

luc-github, 2019-06-08:
Did you check the link ? It is HardwareSerial.cpp

stickbreaker, 2019-06-08:
@YAFIYET HardwareSerial had Default pin assignment, But, you can override these defaults.  When you initially configure the `Serial()` object, just specify the pins you want to use.  The `Serial.begin()` function is defined as:
```
void begin(unsigned long baud, uint32_t config=SERIAL_8N1, int8_t rxPin=-1, int8_t txPin=-1, bool invert=false, unsigned long timeout_ms = 20000UL);
```
As you can see `begin()` can accept 6 parameters, 5 of them have default values, so you only have to specify baud rate if you want to use the defaults.

to configure `Serial()` to use pin 16 as Rx and 17 as  tx
```
Serial.begin(115200,SERIAL_8N1,16,17);
```

GeorgeFlorian, 2019-06-13:
> @YAFIYET HardwareSerial had Default pin assignment, But, you can override these defaults. When you initially configure the `Serial()` object, just specify the pins you want to use. The [...]

That's how I actually ended up doing.
Thank you !

closed 2019-06-13 by GeorgeFlorian