espressif/arduino-esp32 #2859: CORRUPT HEAP when attempting Face Detection using CameraWebServer

AlanFuller, 2019-06-05:
###Hardware
Board: TTGO T-Camera ESP32 WROVER & PSRAM Camera Module ESP32-WROVER-B OV2640 Camera Module 0.96 OLED
Core Installation version:	1.02
IDE name: Arduino IDE
Flash Frequency: 80Mhz
PSRAM enabled: yes
Upload Speed:	921600
Computer OS: macOS Mojave Version 10.14.4

### Description:
I am using the ESP example code for CameraWebServer. Added the pin definitions in 'camera_pins.h' for the camera for this board and set up my own WiFi credentials.
The code flashes OK and the webpage for control works OK. I select the 'V-flip' option.
The video streams At about 20 fps successfully. When I select the 'Face Detection' option after a few seconds I get a message as below:

MJPG: 10313B 220ms (4.5fps), AVG: 280ms (3.6fps), 137+61+0+0=198 0
CORRUPT HEAP: multi_heap.c:165 detected at 0x3ffdfa14
abort() was called at PC 0x40090f2b on core 1

Backtrace: 0x40091b44:0x3ffdc890 0x40091d75:0x3ffdc8b0 0x40090f2b:0x3ffdc8d0 0x4009124f:0x3ffdc8f0 0x400917c2:0x3ffdc910 0x4008840a:0x3ffdc930 0x400899b5:0x3ffdc950 0x4000bec7:0x3ffdc970 0x400ea159:0x3ffdc990 0x400d3052:0x3ffdc9b0 0x4012ba4d:0x3ffdcb60 0x4012aca1:0x3ffdcb90 0x4012ad7d:0x3ffdcc20 0x4012b188:0x3ffdcc40 0x4012a2be:0x3ffdcc60 0x4012a310:0x3ffdcca0 0x4008dae1:0x3ffdccc0

Rebooting...

### Sketch:  The only changes to the sketch are to include by own WiFi credentials.

The pin definitionI added to 'camera_pins.h' is as follows:

#elif defined(CAMERA_MODEL_TTGO)
#define PWDN_GPIO_NUM    -1
#define RESET_GPIO_NUM   -1
#define XCLK_GPIO_NUM    32
#define SIOD_GPIO_NUM    13
#define SIOC_GPIO_NUM    12

#define Y9_GPIO_NUM      39
#define Y8_GPIO_NUM      36
#define Y7_GPIO_NUM      23
#define Y6_GPIO_NUM      18
#define Y5_GPIO_NUM      15
#define Y4_GPIO_NUM       4
#define Y3_GPIO_NUM      14
#define Y2_GPIO_NUM       5
#define VSYNC_GPIO_NUM   27
#define HREF_GPIO_NUM    25
#define PCLK_GPIO_NUM    19

dougalcampbell, 2019-06-11:
I'm getting the same thing with 1.0.3-rc1.

AlanFuller, 2019-08-06:
Update 6 August 2019:
After upgrading to Core Installation version: 1.0.3-rc1 this problem was resolved.

closed 2019-08-06 by AlanFuller