espressif/arduino-esp32 #2861: Pls add 'make menuconfig' in arduino option.

ArTyGill, 2019-06-06:
I want set only one ble client.

lbernstone, 2019-06-06:
Arduino is configured to be easy to use for hobbyists with little programming experience.  Part of this is compiling the underlying libraries with some defaults, which may not be appropriate for more experienced users.  If you want more control over the settings, you can use [arduino as an esp-idf component](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md) or [platform.io](platformio.org), which will give you more control.

chegewara, 2019-06-06:
> I want set only one ble client.

You dont need sdkconfig to do this. Just discard any connection attempts when you are connected.

closed 2019-08-19 by stale[bot]