espressif/arduino-esp32 #2866: BLE Library Compilation Error with Arduino as ESP-IDF component

coolbreeze413, 2019-06-06:
### Hardware:
Board:							ANY

### Description:
Compilation Error in BLE_Advertising.cpp at lines 265 and 274:

```
log_v(">> setPrivateAddress")
log_v("<< setPrivateAddress")
```

https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLEAdvertising.cpp#L265

https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLEAdvertising.cpp#L274

Both are missing semicolons!

Introduced as part of commit:
https://github.com/espressif/arduino-esp32/commit/bea7bd1852bae767864dbb582ea4d4123bc968c1

atanisoft, 2019-08-02:
@coolbreeze413 please consider closing this issue as this has been fixed via PR #2876

coolbreeze413, 2019-08-02:
closing the issue, as fix merged.
apologize for delay, missed the merge notification.

closed 2019-08-02 by coolbreeze413