espressif/arduino-esp32 #2868: WiFi.waitForConnectResult() returns WL_NO_SHIELD

mirozmrzli, 2019-06-06:
I have just updated esp-idf and arduino. I'm using arduino as a component in the project. This is the relevant code:

		char ssid[32], pwd[32];
		settings.readSsidPwd(ssid, sizeof(ssid), pwd, sizeof(pwd));
		ESP_LOGI(TAG, "connecting to %s/%s", ssid, pwd);
		WiFi.begin(ssid, pwd);
		int connRes = WiFi.waitForConnectResult();
		if (connRes == WL_CONNECTED)
		{
			// all is gud!
			ESP_LOGI(TAG, "we're connected");
			configTime(0, 0, "pool.ntp.org", "time.nist.gov");	// lol!
			IPAddress local_adr = WiFi.localIP();
			ESP_LOGI(TAG, "local ip address: %d.%d.%d.%d", local_adr[0],local_adr[1],local_adr[2],local_adr[3]);
		}
		else
		{
			ESP_LOGE(TAG, "we're not connected. going to provisioning mode (error %d)", connRes);
			provision(settings);
		}
	}

and this is the log:

I (1282) APP: connecting to xxx/xxxxxx
I (1292) wifi: wifi driver task: 3ffd054c, prio:23, stack:3584, core=0
I (1292) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1292) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1312) wifi: wifi firmware version: 17f94ee
I (1312) wifi: config NVS flash: enabled
I (1312) wifi: config nano formating: disabled
I (1312) wifi: Init dynamic tx buffer num: 32
I (1322) wifi: Init data frame dynamic rx buffer num: 32
I (1322) wifi: Init management frame dynamic rx buffer num: 32
I (1332) wifi: Init management short buffer num: 32
I (1332) wifi: Init static rx buffer size: 1600
I (1342) wifi: Init static rx buffer num: 10
I (1342) wifi: Init dynamic rx buffer num: 32
I (1432) phy: phy_version: 4100, 2a5dd04, Jan 23 2019, 21:00:07, 0, 0
I (1432) wifi: mode : sta (b4:e6:2d:dc:f3:85)
I (1562) wifi: new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (1562) wifi: state: init -> auth (b0)
I (1562) wifi: state: auth -> assoc (0)
I (1572) wifi: state: assoc -> run (10)
I (1592) wifi: connected with xxxx, channel 11, bssid = e8:2c:6d:03:f4:bd
I (1592) wifi: pm start, type: 1

I (2282) tcpip_adapter: sta ip: 192.168.1.104, mask: 255.255.255.0, gw: 192.168.1.100
E (11442) APP: we're not connected. going to provisioning mode (error 255)
I (11442) wifi: state: run -> init (0)
I (11442) wifi: pm stop, total sleep time: 9025757 us / 9848099 us

I (11442) wifi: new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (11452) wifi: mode : softAP (b4:e6:2d:dc:f3:86)
I (11462) wifi: Total power save buffer number: 16
I (11462) wifi: Init max length of beacon: 752/752
I (11462) wifi: Init max length of beacon: 752/752

mirozmrzli, 2019-06-07:
It seems that the last compatible esp-idf version this framework works with is a commit back in end of March 2019. Is there an official document/guideline/post/anything that talks about arduino component and esp-idf compatibility?

mirozmrzli, 2019-06-07:
Didn't mean to close an issue, duh.

VijitSingh97, 2019-06-23:
Any luck, I am getting wifi.status() == WL_NO_SHIELD, and cannot get a connect message even after my device gets an IP.

atanisoft, 2019-06-23:
Arduino is only compatible with esp IDF v3.2 at this time.

The only reason I can find for WL_NO_SHIELD is if there is an init failure of the WiFi library (ie begin fails).

closed 2019-09-05 by stale[bot]

YordanYanakiev, 2021-04-18:
This issue is very problematic. Still waiting for solution.
It's already 4 years now and peoples keep reporting it.

antfobe, 2021-05-25:
For anybody still having this error, apparently it can be caused by something wrong in the flash memory. From nvs_flash_init(), mentioned in [https://github.com/espressif/arduino-esp32/issues/761](https://github.com/espressif/arduino-esp32/issues/761), I was getting ESP_ERR_NVS_NEW_VERSION_FOUND (0x1110): NVS partition contains data in new format and cannot be recognized by this version of code.

A full chip erase `esptool.py --chip esp32 -p /dev/ttyUSB0 erase_flash` fixed my problem.

mlord, 2021-11-15:
The state transitions from WiFi.status() include some unexpected values mid-connect.  This confuses WiFi.waitForConnectResult() which expects them to be linear, when they are not always so.

I used this loop to wait for a connection, with a 10-second timeout:

#define time_after(a,b)  ((long)(b) - (long)(a) < 0)  // From Linux kernel
  WiFi.begin (ssid, passwd);
  unsigned long timeout = millis() + 10000;
  while (!time_after(millis(), timeout)) {
    int stat = WiFi.status();
    Serial.printf(" %d", WiFi.status());
    if (WiFi.status() == WL_CONNECTED) 
      break;
    delay(50);
  }
  Serial.println();
  if ( WiFi.status() != WL_CONNECTED )
  {
     ....
  }

The resulting output can be very surprising.  Most often it looks like this:

 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 3
Which means WL_DISCONNECTED(6) ... WL_CONNECTED(3).

But sometimes it gives this:
6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 0 3
Similar to above, but with an unexpected WL_IDLE_STATUS(0) before WL_CONNECTED(3).

And sometimes it even does this:
6 6 6 6 6 6 6 6 6 6 6 6 6 4 4 4 4 4 4 4 4 4 4 4 4 4 4 3
Which means WL_DISCONNECTED(6) ... WL_CONNECT_FAILED(4) ... WL_CONNECTED(3).

So.. don't trust the state transistions.. Instead just loop and wait for WL_CONNECTED or a timeout.

mlord, 2021-11-15:
On further testing, with a faster "sampling rate", it appears that the status ALWAYS goes to WL_IDLE_STATUS(0) just before it changes to WL_CONNECTED(3).  So these sequences are both very common for me here:
6 ... 0 --> 3
6 ... 4 ... 0 --> 3