espressif/arduino-esp32 #2869: Can CameraWebServer & SoftAP run together on same ESP32-Camera?

CameraWebServer, 2019-06-07:
Is it possible to run CameraWebServer and SoftAP on same ESP32-CAMERA?
Want to connect to the SoftAP from SmartPhone - using Chrome ...
Adding this to CameraWebServer in Arduino IDE - does not work.

#include <WiFi.h>
#include <WiFiClient.h>
#include <WiFiAP.h>
...

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println();

  WiFi.softAP(ssid, password);
  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(myIP);
  ...

closed 2019-08-20 by stale[bot]

podericof, 2019-10-12:
Hi there, any progress here ? thank you!

me-no-dev, 2019-10-14:
there is little to no info in the issue. what happens when you start AP, what happens when you try to connect? Are there any messages in the console? Did you set debug level to something other than None in the menu?

fdoljack, 2019-11-16:
hi,
This post appears to relate to my question.  I have built camerawebserver and it is running on two devices.  Then I noticed that the both the station and softAP modes are active and the webserver can be connected to either through my wifi LAN or be connecting to ESP-Camera and then connecting to 192.168.4.0.  I really don't want this because any neighbor can then connect to my cameras!!!

fdoljack, 2019-11-16:
BTW, after opening the browser I can stream by addressing http://192.168.4.0/stream.

fdoljack, 2019-11-16:
sorry for the correction,  but I can stream by sending http://192.168.4.1:81/stream