espressif/arduino-esp32 #2870: Bootloading problem with UART1

Embel-Technologies, 2019-06-07:
### Hardware:
Board:							ESP32 Dev Module(Self developed)
IDE name:						Arduino IDE and esptool.py
Flash Frequency:					40Mhz
PSRAM enabled:                                     yes
Upload Speed:					115200
Computer OS:                                        Windows 10, Ubuntu

Description:
Hi everyone,

I am working on a hardware designing on ESP32 chip, I have attached all the required component for it as per datasheets and some development modules. but i have done some changes on that. I am using UART1 for a debugging purpose instead of UART0  and I haven't use auto-boot-loading circuit on that but I have connected GPIO0 and CHIP-PU for Boot and Reset pin.

The problem is I am unable access UART1 or UART2 for debugging or for boot-loading. Is there any problem with UART1 for debugging purpose? or if we can do programming through it so please help me to solve it. Is there any need to define or giving the address of UART1 to any file so Please tell me where i need to change. I have done many things to find a solutions,
Thank you in advance.

stickbreaker, 2019-06-07:
The boot loader is ROM code that uses Uart0 on pins 1 and 3.  The boot loader is hard coded, you cannot change it.  You can use OTA to change the code stored in flash, but you have to initially use the boot loader to load your OTA code.  After you get your initial OTA code into flash, you can reconfigure debug to use whatever serial device you choose. 

Chuck.

Embel-Technologies, 2019-06-08:
Thank you.. but i want to code on flash through UART1. Can we do that on UART1.

stickbreaker, 2019-06-08:
You will have to first upload your own code(using the ROM bootload on UART0 Pin 1,3) that supports UART1, Your code would have  to always have to be in control and update FLASH with your code.  There is no way to initially program the ESP32 with using bootloader, and it is not possible to overwrite the builtin bootloader.

Embel-Technologies, 2019-06-08:
I have connected CP2102 to UART1 for debugging. Is it possible to?

atanisoft, 2019-06-08:
No it is not possible, uart0 is required for flashing as it is hardwired into the hardware and bootloader software.

atanisoft, 2019-08-02:
@Embel-Technologies please consider closing this issue if your issue has been resolved.

negativekelvin, 2019-08-04:
Actually these answers are not correct.  You can connect to the ROM bootloader on UART1.  The problem is the default pins are on gpio 9&10 and most modules/boards have a flash chip connected to these pins.  But if you have a pico kit you can try it because gpio 9&10 are not used for flash.

atanisoft, 2019-08-04:
@negativekelvin it is only possible with a custom IDF compile to change the bootloader. In the arduino-esp32 env it is precompiled and it is not possible to alter it for the average end user.

negativekelvin, 2019-08-04:
@atanisoft ROM bootloader can't be changed by anyone

> rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/**UART1**/SDIO_REI_REO_V2))

atanisoft, 2019-08-04:
@negativekelvin true but the user space bootloader that it calls defaults to uart0 and without recompiling IDF it can't be changed.

Also virtually every commercially available esp32 board is wired as defaults (uart0)

negativekelvin, 2019-08-04:
Flashing via UART1 has nothing to do with user space bootloader.  Debugging, yes you may have to recompile or just reconfigure it in your setup() function but no reason to give incomplete/incorrect answers.  Do you think @Embel-Technologies would rather rework his hardware or recompile his bootloader?

atanisoft, 2019-08-04:
It would depend on the debugging approach, it could be as simple as calling Serial1.begin(...); Serial1.setDebugOutput(true);

That should redirect printf and similar logging to uart1.

As for the rom bootloader, I haven't looked at the early esp32 boards but I believe some of the first gen did not support anything other than uart0 for the upload in the bootloader. It is more common with the output you provide above now though.

closed 2019-10-17 by stale[bot]

LucaH360, 2023-01-14:
#include "esp_camera.h"
#include <WiFi.h>

//
// WARNING!!! PSRAM IC required for UXGA resolution and high JPEG quality
//            Ensure ESP32 Wrover Module or other board with PSRAM is selected
//            Partial images will be transmitted if image exceeds buffer size
//

// Select camera model
#define CAMERA_MODEL_WROVER_KIT  // Has PSRAM
//#define CAMERA_MODEL_ESP_EYE // Has PSRAM
//#define CAMERA_MODEL_M5STACK_PSRAM // Has PSRAM
//#define CAMERA_MODEL_M5STACK_V2_PSRAM // M5Camera version B Has PSRAM
//#define CAMERA_MODEL_M5STACK_WIDE // Has PSRAM
//#define CAMERA_MODEL_M5STACK_ESP32CAM // No PSRAM
//#define CAMERA_MODEL_AI_THINKER // Has PSRAM
//#define CAMERA_MODEL_TTGO_T_JOURNAL // No PSRAM

#include "camera_pins.h"

const char* ssid = "SDNET1";
const char* password = "943394";

void startCameraServer();

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println();

  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG;

  // if PSRAM IC present, init with UXGA resolution and higher JPEG quality
  //                      for larger pre-allocated frame buffer.
  if (psramFound()) {
    config.frame_size = FRAMESIZE_UXGA;
    config.jpeg_quality = 10;
    config.fb_count = 2;
  } else {
    config.frame_size = FRAMESIZE_SVGA;
    config.jpeg_quality = 12;
    config.fb_count = 1;
  }

#if defined(CAMERA_MODEL_ESP_EYE)
  pinMode(13, INPUT_PULLUP);
  pinMode(14, INPUT_PULLUP);
#endif

  // camera init
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    return;
  }

  sensor_t* s = esp_camera_sensor_get();
  // initial sensors are flipped vertically and colors are a bit saturated
  if (s->id.PID == OV3660_PID) {
    s->set_vflip(s, 1);        // flip it back
    s->set_brightness(s, 1);   // up the brightness just a bit
    s->set_saturation(s, -2);  // lower the saturation
  }
  // drop down frame size for higher initial frame rate
  s->set_framesize(s, FRAMESIZE_QVGA);

#if defined(CAMERA_MODEL_M5STACK_WIDE) || defined(CAMERA_MODEL_M5STACK_ESP32CAM)
  s->set_vflip(s, 1);
  s->set_hmirror(s, 1);
#endif

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");

  startCameraServer();

  Serial.print("Camera Ready! Use 'http://");
  Serial.print(WiFi.localIP());
  Serial.println("' to connect");
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(10000);
}

ce code je j'ai besoin d'utiliser et beaucoup tros gros je peux faire quoi pour le mettre moins 

Croquis trop gros; vois https://support.arduino.cc/hc/en-us/articles/360013825179 pour des conseils de réduction.
Le croquis utilise 2594774 octets (197%) de l'espace de stockage de programmes. Le maximum est de 1310720 octets.
Les variables globales utilisent 56248 octets (17%) de mémoire dynamique, ce qui laisse 271432 octets pour les variables locales. Le maximum est de 327680 octets.
text section exceeds available space in board

Compilation error: text section exceeds available space in board

LucaH360, 2023-01-14:
merci beaucoup beaucoup beaucoup beaucoup d'avevnce

lbernstone, 2023-01-14:
@LucaH360 Use Tools->Partition Scheme->Huge APP