espressif/arduino-esp32 #2874: Reliability of internal watchdog in ESP32-WROOM

DeltaVetal26, 2019-06-08:
Hello, 

I`m creating a controller based on ESP32-DevKit V4.

The ESP32 has an internal watchdog timer.
The question is, is this watchdog reliable, or should external devices be used to protect against freezing?
Could there be cases when the internal WatchDog does not reset the board in the event of a failure?

My controller will be in an inaccessible place, so it is necessary to make it as stable as possible.

lbernstone, 2019-06-08:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/wdts.html
The watchdog is a timer that prevents processes from monopolizing the cpu.  Triggering the watchdog timer indicates an error in your code. It really is not intended as a production failsafe.  There are many fatal errors which will not trigger the watchdog.

DeltaVetal26, 2019-06-09:
@lbernstone 

Thanks for the answer!

And what will happen when using external watchdog timer and OTA update? Suppose ESP downloads firmware from a server and at this time does not send pulses to an external timer. Or at this time the ESP does not respond to a reset?

lbernstone, 2019-06-09:
OTA is just a standard, single threaded process.  Run your keepalive on core0 (system), and OTA should have no effect.  When done correctly, the system will survive an interrupted OTA anyhow.

closed 2019-08-18 by DeltaVetal26