espressif/arduino-esp32 #2875: setCACert() works when provided a string constant, but not a string from `.c_str()`

mdszy, 2019-06-08:
My code looks like the following:

```cpp
  if(!SPIFFS.begin(true)) {
    Serial.println("Error mounting SPIFFS.");
  }

  File file = SPIFFS.open("/root.cer");

  if(!file) {
    Serial.println("Error opening the file.");
  }

  Serial.println("CA Root certificate: ");

  String ca_cert = file.readString();

  Serial.println(ca_cert);
  espClient.setCACert(ca_cert.c_str());

  file.close();
```
This code doesn't work. I cannot connect to the secure server. However, if I replace `espClient.setCACert(ca_cert.c_str());` with `espClient.setCACert(ROOTCERT);` where `ROOTCERT` is defined as such:

```cpp
#define ROOTCERT "-----BEGIN CERTIFICATE-----\n" \
"MIIDSjCCAjKgAwIBAgIQRK+wgNajJ7qJMDmGLvhAazANBgkqhkiG9w0BAQUFADA/\n" \
"MSQwIgYDVQQKExtEaWdpdGFsIFNpZ25hdHVyZSBUcnVzdCBDby4xFzAVBgNVBAMT\n" \
"DkRTVCBSb290IENBIFgzMB4XDTAwMDkzMDIxMTIxOVoXDTIxMDkzMDE0MDExNVow\n" \
"PzEkMCIGA1UEChMbRGlnaXRhbCBTaWduYXR1cmUgVHJ1c3QgQ28uMRcwFQYDVQQD\n" \
"Ew5EU1QgUm9vdCBDQSBYMzCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoCggEB\n" \
"AN+v6ZdQCINXtMxiZfaQguzH0yxrMMpb7NnDfcdAwRgUi+DoM3ZJKuM/IUmTrE4O\n" \
"rz5Iy2Xu/NMhD2XSKtkyj4zl93ewEnu1lcCJo6m67XMuegwGMoOifooUMM0RoOEq\n" \
"OLl5CjH9UL2AZd+3UWODyOKIYepLYYHsUmu5ouJLGiifSKOeDNoJjj4XLh7dIN9b\n" \
"xiqKqy69cK3FCxolkHRyxXtqqzTWMIn/5WgTe1QLyNau7Fqckh49ZLOMxt+/yUFw\n" \
"7BZy1SbsOFU5Q9D8/RhcQPGX69Wam40dutolucbY38EVAjqr2m7xPi71XAicPNaD\n" \
"aeQQmxkqtilX4+U9m5/wAl0CAwEAAaNCMEAwDwYDVR0TAQH/BAUwAwEB/zAOBgNV\n" \
"HQ8BAf8EBAMCAQYwHQYDVR0OBBYEFMSnsaR7LHH62+FLkHX/xBVghYkQMA0GCSqG\n" \
"SIb3DQEBBQUAA4IBAQCjGiybFwBcqR7uKGY3Or+Dxz9LwwmglSBd49lZRNI+DT69\n" \
"ikugdB/OEIKcdBodfpga3csTS7MgROSR6cz8faXbauX+5v3gTt23ADq1cEmv8uXr\n" \
"AvHRAosZy5Q6XkjEGB5YGV8eAlrwDPGxrancWYaLbumR9YbK+rlmM6pZW87ipxZz\n" \
"R8srzJmwN0jP41ZL9c8PDHIyh8bwRLtTcm1D9SZImlJnt1ir/md2cXjbDaJWFBM5\n" \
"JDGFoqgCWjBH4d1QB7wCCZAA62RjYJsWvIjJEubSfZGL+T0yjWW06XyxV3bqxbYo\n" \
"Ob8VZRzI9neWagqNdwvYkQsEjgfbKbYK7p2CNTUQ\n" \
"-----END CERTIFICATE-----\n"
```

The code works just fine. The `ROOTCERT` string is taken directly from the certificate file, so they must be identical.

The certificate file was downloaded and exported using Windows's certificate exporter. I've tried converting line endings to no avail.

However, if I change the code to be like this:

```cpp
String constString = ROOTCERT;
espClient.setCACert(constString.c_str());
```

it also doesn't work. So it seems to be a problem with something in `.c_str()`

I also added this code:

```cpp
 if(strcmp(constString.c_str(), ROOTCERT))
     Serial.println("Constant and converted string are equal.");
   else
     Serial.println("Constant and converted string are different.");
```

and I get "Constant and converted string are different."

What gives? I'm completely stumped here.

mdszy, 2019-06-09:
Okay, I fixed it. I really hope this comes in handy for anyone else with this issue so they don't have to spend hours poking around and trying random things until it works.

`.c_str()` points to the internal buffer in a `String` object, which apparently is incompatible with the `setCACert()` function.

Adding the following code to copy the string into a new buffer fixes it:

```cpp
char *dest;  
dest = (char *)malloc(sizeof(char) * (ca_cert.length()+1));
strcpy(dest, ca_cert.c_str());
  
espClient.setCACert(dest);
```

closed 2019-06-09 by mdszy

joeltbennett, 2022-12-11:
It has helped a lot, thank you - I'm using it to download CA certs from a server (and store with SPIFFS) before initiating secure connections, that way I don't have to update my MCU code any time the certificate updates.

winnergeorge, 2023-07-20:
I was also stuck with the same issue for a couple of days, then I came across your post and IT WORKS!!!!. Thanks a lot

ismaelit, 2023-12-22:
Really helped. Thanks for sharing!