espressif/arduino-esp32 #2886: ESP-32-can-not-support-its-wifi-maximum-range-when-a-mqtt-server-is-also-running-simultenously

MAINAKMONDAL98, 2019-06-11:
I noticed that my esp 32 is disconnected from mqtt server when the hotspot is far away from esp 32.
If I switched off my hotspot then the esp 32 is disconnected from internet and stops all the linked features.
Then when I turned on the hotspot,it again connected and started all the server client relationship.
If I am providing my hotspot from long distance to esp it is coonected with wifi but failed to coonect with mqtt server.
client.state() ==-2 or -3.
If I take esp closer to hotspot it reconnected and client.state()==0
It again rebuild the pubsubclient relationship
I attached four pictures of all states in this repository.
https://github.com/MAINAKMONDAL98/ESP-32-can-not-support-its-wifi-maximum-range-when-a-mqtt-server-is-also-running-simultenously
https://github.com/MAINAKMONDAL98/ESP-32-can-not-support-its-wifi-maximum-range-when-a-mqtt-server-is-also-running-simultenously/blob/master/HOTSPOT%20FROM%20LONG%20RANGE%20AND%20MQTT%20DISCONNECTED.png
https://github.com/MAINAKMONDAL98/ESP-32-can-not-support-its-wifi-maximum-range-when-a-mqtt-server-is-also-running-simultenously/blob/master/HOTSPOT%20FROM%20SMALL%20RANGE%20THEN%20MQTT%20CONNECTED.png

MAINAKMONDAL98, 2019-06-11:
https://github.com/MAINAKMONDAL98/ESP-32-can-not-support-its-wifi-maximum-range-when-a-mqtt-server-is-also-running-simultenously

atanisoft, 2019-06-11:
Pictures without code is not meaningful. Very likely it has nothing to do with MQTT but instead WiFi signal strength being weak when the hotspot is far away.

closed 2019-08-25 by stale[bot]