espressif/arduino-esp32 #2887: Implement PHP cURL request into ESP32 HTTPClient

danish-anwar-butt, 2019-06-12:
Hi,
I'm new to GitHub forum so it may seem irrelevant but I hope I will get help here.
I'm uploading a file to Azure blob storage using REST API request. The file uploaded successfully using PHP curl request. But when I transfer the code into ESP32 HTTP client I'm getting an Authentication error. I also verify the HMAC key generated by the ESP32 Arduino code in PHP code.
I'm thinking that I'm missing something in adding header.
Here is my PHP code :

`
       $header = array (
        "x-ms-blob-type: BlockBlob",
        "x-ms-date: " . $date,
        "x-ms-version: " . $version,
        "Authorization: SharedKey mikeiot:" . $signature_str,
        "Content-Type: text/plain; charset=UTF-8",
        "Content-Length: " . strlen($fdata),
        );

    $url =  "https://acount_name.blob.core.windows.net/container/test.txt";
    $ch = curl_init ();
    curl_setopt($ch, CURLOPT_SSL_VERIFYPEER, false);
    curl_setopt ( $ch, CURLOPT_CUSTOMREQUEST, 'PUT' );
    curl_setopt ( $ch, CURLOPT_URL, $url );
    curl_setopt ( $ch, CURLOPT_RETURNTRANSFER, true );
    curl_setopt ( $ch, CURLOPT_HTTPHEADER, $header);
    curl_setopt($ch, CURLOPT_POSTFIELDS, $fdata);
    curl_setopt ( $ch, CURLOPT_HEADER, True );
    echo($ch);
    curl_exec ( $ch );
`

and here is the corresponding Arduino IDE code:
`
String sig = get_hmac();
  Serial.println(sig);

  if (WiFi.status() == WL_CONNECTED) { //Check WiFi connection status
    HTTPClient http;
    http.begin("https://account_name.blob.core.windows.net/container/test.txt");  //Specify destination for HTTP request
    
    http.addHeader("x-ms-blob-type", "BlockBlob");
    http.addHeader("x-ms-date", Date);
    http.addHeader("x-ms-version", "2018-03-28");
    http.addHeader("Authorization", "SharedKey account_name:"+sig);
    http.addHeader("Content-Type", "text/plain; charset=UTF-8");
    http.addHeader("Content-Length",  "20");
                 
    int httpResponseCode = http.PUT(d);   
    if (httpResponseCode > 0) {
      String response = http.getString(); 
      Serial.println(httpResponseCode);   
      Serial.println(response);        
    } else {
      Serial.print("Error on sending POST: ");
      Serial.println(httpResponseCode);
    }
    http.end();  //Free resources
  } else {
    Serial.println("Error in WiFi connection");
  }`

The error I'm getting is : 

> <AuthenticationErrorDetail>The MAC signature found in the HTTP request 
> 'sVrSFpplX+Gkcb4R/iSDf6eqUP4srkZuVD68/pAeYwE=' is not the same as any computed 
> signature.....

Although the HMAC signature created from same String works in PHP
here is the reference page of the API:
https://docs.microsoft.com/en-us/rest/api/storageservices/put-blob
Thanks

boarchuz, 2019-06-12:
Looks like the http request is working fine so this has nothing to do with ESP32/Arduino.

While we’re here though, you seem to still have placeholder “account_name” in the Authorization header. Might want to check “Date” contains the actual date too.

danish-anwar-butt, 2019-06-13:
@boarchuz Thanks for your response.
Yes, Date is actual date getting from NTP server. Also, HTTP request working fine as I mentioned I'm a bit confused about adding the header and signature is correct. If I send the request with the same signature in PHP I get a correct response. So I think I'm a haivng mess with header

danish-anwar-butt, 2019-06-13:
I manage to fix the problem by removing the line

 `http.addHeader("Content-Length",  "20");`

The header parameter is added by the PUT request.

closed 2019-06-13 by danish-anwar-butt