espressif/arduino-esp32 #2889: Arduino install Esp32 and Esp8266 Boards Manager problem

electrocoder, 2019-06-12:
Hi maker.
I want to esp8266 and esp32 board install Arduino IDE. Problem is only esp8266 or esp32 in view.  Only the last one is displayed.

Thanks.

### Hardware:
Board:							ESP32 and ESP8266
IDE name:							Arduino IDE 1.8.9
Computer OS:                                            Ubuntu 18.04.2 LTS 64 bit

lbernstone, 2019-06-12:
It is definitely possible to have both.  Can you run the following from a terminal prompt:
`
ls ~/.arduino15/packages/esp8266/hardware/esp8266/*/boards.txt
ls ~/.arduino15/packages/esp32/hardware/esp32/*/boards.txt
`

electrocoder, 2019-06-13:
e@e:~$ ls ~/.arduino15/packages/esp8266/hardware/esp8266/*/boards.txt
/home/e/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/boards.txt
e@e:~$ ls ~/.arduino15/packages/esp32/hardware/esp32/*/boards.txt
/home/e/.arduino15/packages/esp32/hardware/esp32/1.0.2/boards.txt
e@e:~$

atanisoft, 2019-06-13:
How did you install the boards? Did you use the board manager option? If so, did you list both esp32 and esp8266 json files with a comma separator?

electrocoder, 2019-06-13:
I didn't separate json files with commas. I'm trying now.

electrocoder, 2019-06-13:
Solved. Thank you.

closed 2019-06-13 by electrocoder