espressif/arduino-esp32 #2890: SmartConfig Resets Esp32

EmilioSalomone, 2019-06-13:
### Hardware:
Board:							ESP32 Wroom32
Core Installation version:			1.0.2
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                         Windows 10
### Description:

Hi to everyone,

i am making a device that has to be configured with SmartConfig after the startup ,

it has multiple tasks, when the one with smartconfig starts there is another one running on the other core of the ESP32, the weird problem is that in a determined spot of my house, when the smartConfig task starts , my esp32 begins to resets itself.
If i place my device into another place , it works like a charm.

i link the error shown in serial monitor with verbose debug on : 

"10:18:20.946 -> .Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
10:18:21.254 -> Core 0 register dump:
10:18:21.254 -> PC      : 0x400e7627  PS      : 0x00060e30  A0      : 0x8015ae2c  A1      : 0x3ffb5820  
10:18:21.289 -> A2      : 0x3ffcee80  A3      : 0x00000001  A4      : 0x3ffbf89c  A5      : 0x3ffc1b60  
10:18:21.289 -> A6      : 0x3ffc1b58  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffb5810  
10:18:21.289 -> A10     : 0x3ffcee80  A11     : 0x3ffbee22  A12     : 0x00000006  A13     : 0x3ffcee86  
10:18:21.289 -> A14     : 0x0000ffff  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
10:18:21.322 -> EXCVADDR: 0x0000002c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
10:18:21.322 -> 

10:18:21.322 -> Backtrace:0x400e7627:0x3ffb5820 0x4015ae29:0x3ffb58700x4015b098:0x3ffb58c0 0x4015b0af:0x3ffb5900 0x4015dd00:0x3ffb5920 0x4015de2a:0x3ffb5940 0x40090250:0x3ffb5960 0x40088ff9:0x3ffb59a0

"

Someone had the same issue?

Thank you so much for help.

atanisoft, 2019-06-13:
Please decode the stack trace using the backtrace decoder for the Arduino ide.

EmilioSalomone, 2019-06-13:
This is what Backtrace decoder prints out:

0x40088ff9: pxPortInitialiseStack at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 240

atanisoft, 2019-06-13:
Please share the full backtrace and not just the first line.

Also code to reproduce this would be useful.

EmilioSalo, 2019-06-17:
Backtrace:0x400e7627:0x3ffb5820 0x4015ae29:0x3ffb58700x4015b098:0x3ffb58c0 0x4015b0af:0x3ffb5900 0x4015dd00:0x3ffb5920 0x4015de2a:0x3ffb5940 0x40090250:0x3ffb5960 0x40088ff9:0x3ffb59a0

This is the full backtrace.

`void smartConfigStart(void) {
  if (digitalRead(buttonPin) == LOW) {
    unsigned long startTime = millis ();
    unsigned long stopTime = 50000;
    if (WiFi.status() != WL_CONNECTED) {
      delay(1000);
      WiFi.mode(WIFI_AP_STA);
      WiFi.beginSmartConfig();
      //Wait for SmartConfig packet from mobile
      Serial.println("Waiting for SmartConfig.");
      while ((millis () - startTime) < stopTime) {
        if (!WiFi.smartConfigDone()) {
          delay(500);
          Serial.print(".");
        }
        else {
          Serial.println("");
          Serial.println("SmartConfig received.");
          delay(2000);
          break;
        }
      }
    }
    //Wait for WiFi to connect to AP
    Serial.println("Waiting for WiFi");
    for (int i = 0; i < 10; i++) {
      if (WiFi.status() != WL_CONNECTED) {
        delay(2000);
        i++;
        connessioneSmartC = false;
        Serial.println("RiprovoSmart");
      }
    }
    if (WiFi.status() == WL_CONNECTED) {
      connessioneSmartC  = true;
      delay(4000);
      ledcWrite(channel, dutyCycle);
      delay(100);
      ledcWrite(channel, 0);
      delay(100);
      ledcWrite(channel, dutyCycle);
      delay(100);
      ledcWrite(channel, 0);
      digitalWrite(buzzerPin, LOW);
      Serial.println("WiFi connesso");
      Serial.println("Indirizzo IP: ");
      Serial.println(WiFi.localIP());
      Serial.println( WiFi.SSID() );
      strcpy(jsonssid, WiFi.SSID().c_str());
      Serial.println( WiFi.psk() );
      strcpy(jsonpwd, WiFi.psk().c_str());
      saveWifiJson();
      delay(4000);
      xTaskCreatePinnedToCore(
        codeForTask6,
        "serverLEDOn",
        8192,
        NULL,
        2,
        &Task6,
        0);

    }

    if (connessioneSmartC == false) {
      WiFi.stopSmartConfig();
      WiFi.disconnect(true);                     // disconnects STA Mode
      delay(1000);
      const char* ssid = jsonssid;
      const char* password = jsonpwd;
      Serial.println(ssid);
      Serial.println(password);
      Serial.print("Connessione a ");
      Serial.println(ssid);
      WiFi.persistent(false);
      WiFi.mode(WIFI_OFF);
      WiFi.mode(WIFI_STA);
      WiFi.begin(ssid, password);

      Serial.println( WiFi.SSID() );
      Serial.println( WiFi.psk() );
      for (int i = 0; i < 10; i++) {
        delay(1000);
        if (WiFi.status() != WL_CONNECTED) {
          i++;
          Serial.println("Riprovo");
        }
      }
      if (WiFi.status() == WL_CONNECTED) {
        Serial.println("WiFi connesso");
        Serial.println("Indirizzo IP: ");
        Serial.println(WiFi.localIP());
        xTaskCreatePinnedToCore(
          codeForTask6,
          "serverLEDOn",
          8192,
          NULL,
          2,
          &Task6,
          0);
      }
      else {
        delay(1000);
        xTaskCreatePinnedToCore(
          codeForTask3,
          "APMode",
          2048,
          NULL,
          2,
          &Task3,
          0);
      }
    }
  }
  else {
    wifibutton = false;
  }
  delay(2000);
  vTaskDelete( NULL );
}`

this is the vode of the "SmartConfig Task"

atanisoft, 2019-06-17:
Can you post the full decoded backtrace?

EmilioSalo, 2019-06-17:
Decoding stack results
0x40088ff9: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

This is the only message that the ESP Exception Decoder Returns, am i doing something wrong?

atanisoft, 2019-06-17:
Are you pasting the full output you pasted above? It looks like you are only pasting the last line of it

EmilioSalo, 2019-06-17:
[image]

This is what the exception decoder returns, i don't know why it decodes only the last line

atanisoft, 2019-06-17:
That is very odd, looks like a bug in the decoder. You might try pasting it as multiple lines with the 0x...:0x... pieces individually listed one per line

EmilioSalo, 2019-06-17:
[image]
unluckly it returns the same message, did you tried if your decoder gives back the same message? i have tried also to uninstall and install it back but nothing changes

atanisoft, 2019-06-17:
Since I don't have the code for your project I wouldn't be able to decode the exception.

It is very odd though.

EmilioSalo, 2019-06-17:
It's all very strange with this problem! 
Can you suggest me another decoder? If exists another one..

atanisoft, 2019-06-17:
I'm not aware of one that works inside the Arduino ide other than this one

EmilioSalo, 2019-06-18:
I was finally able to decode the backtrace with addr2line on command line, and this is what it returns:

```
0x400e7627: KISS_Find_channel_direct at ??:?
0x4015ae29: sta_rx_csa at ??:?
0x4015b098: sta_input at ??:?
0x4015b0af: sta_input at ??:?
0x4015dd00: cnx_clear_blacklist at ??:?
0x4015de2a: cnx_add_rc at ??:?
0x40090250: ppTxPkt at ??:?
0x40088ff9: xPortInIsrContext at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c:355 (discriminator 1)
```
there is a "known issue"? 
thank you so much!

atanisoft, 2019-06-18:
I'm not certain what might be causing this sort of crash. But it looks like possibly a bug in SmartConfig which is managed by esp-idf, so you might need to move this bug over to there for their input

EmilioSalo, 2019-07-05:
I finally decoded the backtrace correctly, this is what the decoder gives me back: 
`0x400ef157: mdns_parse_packet at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/mdns/mdns.c line 2890
`
`
0x40088ff9: vPortCPUReleaseMutex at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/portmux_impl.inc.h line 163
`
it seems like the mdns used by smartconfig crashes the entirely esp32, has someone faced this issue yet?

closed 2019-09-17 by stale[bot]

jhoughjr, 2019-11-20:
I'm having this exact same issue where as son as the ESP receives the smartconfig packet it reboots.

EmilioSalo, 2019-11-21:
Do you have the module soldered on a PCB made by yourself? in my case was a design issue, according to the design guideline the module has to be soldered in an area with no copper, no traces   and no planes of 1,5cm at right, left and upper side of the antenna, as shown in the picture:
[image]

jhoughjr, 2019-11-21:
I'll have to check, I'm using an M5StickC currently. Strangely after guru meditation erring and rebooting continuously, it rebooted and worked via SmartConfig. I still want to check this as I wan't a solution that is robust enough for consumers.  I don't make PCBs but a friend does. I'll make sure he sees this as it is good to know. Thanks for the info.