espressif/arduino-esp32 #2891: Bug in Print::printf()

glennramalho, 2019-06-13:
Hi,

I was using the Printf function in a StreamString and noticed that in some cases the end of the message when using a long statement got chopped off. Looking more carefully and using a debugger (I have been testing using a simulation of the ESP32) and single steppping through the code I noticed that the problem was seemed to be inside Print::printf().

Inside you have ``size_t Print::printf(const char *format, ...)
{
    char loc_buf[64];
    char * temp = loc_buf;
    va_list arg;
    va_list copy;
    **va_start(arg, format);**
    **va_copy(copy, arg);
    size_t len = vsnprintf(NULL, 0, format, arg);
    va_end(copy);**
    if(len >= sizeof(loc_buf)){
        temp = new char[len+1];
        if(temp == NULL) {
            return 0;
        }
    }
    **len = vsnprintf(temp, len+1, format, arg);**
    write((uint8_t*)temp, len);
    **va_end(arg);**
    if(len >= sizeof(loc_buf)){
        delete[] temp;
    }

But notice that you do the va_start and then a va_copy to create the copy of the arg argument list. Then you call vsprintf twice. I would think that the first one should be with "copy" as argument and the second time with "arg" as argument, but I see both cases using "arg" and none uses "copy". When I corrected my version here the problem went away,

So I would think we should change line 55 of Print.cpp to say "vsprintf(NULL, 0, format, copy);"

regards,

Glenn

everslick, 2019-06-16:
I think you are right. Care to provide a PR?

closed 2019-08-20 by me-no-dev

makerMcl, 2020-12-28:
Hello,
I have problem with ESP32-Arduino-integration: va_list for variadic argument does not get evaluated properly but points at some other memory area.
I am really sorry to place this here but has no idea who can help or get a hint to move forward. Code is in https://github.com/makerMcl/universalUi/blob/master/appendBuffer.h - at line 29 the output is some garbage (or guru error for the worst). Without explicit arguments to snprintf (no variadics) it works so I think problem is in va_start() catching wrong memory area? Is there a bug or did I something wrong?

PR #3060 (2019-08-20): Update Print::printf() to use the copied arg list
Fix #2891