espressif/arduino-esp32 #2893: Could not core dump to flash.

a-c-sreedhar-reddy, 2019-06-14:
### Hardware:
Board:							ESP32 Dev Module
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Ubuntu

### Description:
Wanted to store error log to flash. Got error undefined reference to esp_core_dump_to_flash()

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <rom/rtc.h>
#include <esp_core_dump.h>

void setup() {
    esp_core_dump_init();
}

void loop() {
}
```

### Debug Messages:
```

sketch/testprogram.ino.cpp.o:(.literal._Z5setupv+0x0): undefined reference to `esp_core_dump_init()'
sketch/testprogram.ino.cpp.o: In function `setup()':
/home/acreddy/Arduino/testprogram/testprogram.ino:5: undefined reference to `esp_core_dump_init()'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32 Dev Module. ```

lbernstone, 2019-06-14:
core dump is not compiled into the libraries (CONFIG_ESP32_ENABLE_COREDUMP).  You will need to build as a component to get it.

a-c-sreedhar-reddy, 2019-06-14:
@lbernstone  how can the reset logs be written to flash?

lbernstone, 2019-06-14:
You would have to write your own bootloader.

PRosenb, 2019-08-14:
@me-no-dev Any chance this could be added to `Arduino` without needing to build it as a component? The header file is already in the `Arduino` specific repo:
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/esp32/esp_core_dump.h

closed 2019-10-27 by stale[bot]

fightforlife, 2020-12-28:
Any news on this? 
For the ESP9266 there is another promising library https://github.com/krzychb/EspSaveCrash

sabado, 2021-06-11:
Why developers here don't want to enable this feature or even almost write a decent tutorial about it?

me-no-dev, 2021-06-17:
since this feature is not a runtime option in ESP-IDF, it is not possible to compile it for everyone. You can use the "Arduino as IDF component" route to enable that

sabado, 2021-07-21:
@a-c-sreedhar-reddy  @fightforlife 

Step 1. Compile arduino sdk using this script i made. It will allow you to enable core dump flash in menuconfig, then build and install the arduino-espressif sdk. 
https://github.com/sabado/esp32-easy-menuconfig

Step 2. Create a custom partition, this partition will reserve space in flash to store the crash

Step 3. In coments on esp_core_dump.h file, look how actually esp store data into flash when crash. Using that details you have to read the partition and send to a webserver. 
https://github.com/espressif/esp-idf/blob/master/components/espcoredump/include/esp_core_dump.h

That's all. Take time, but is possible.

MathieuDeprez, 2021-09-11:
Hi,
For those who don't want to compile their own Arduino sdk with Espressif tools, I have done it for you and I share my work on a Github repo.
This works fort Arduino framework v1.0.6 and PlatformIO.

https://github.com/MathieuDeprez/ESP32_CoreDump_Arduino_1.0.6

sabado, 2021-09-11:
@MathieuDeprez  i do the same months ago. is a simple script to run on linux, that gives you access to menuconfig and compiles everything. 

https://github.com/sabado/esp32-easy-menuconfig

MathieuDeprez, 2021-09-11:
@sabado Thank you for sharing that, I will try it tomorrow!
Do your script install IDF v3.3 with the good extensa ? Because you need it to use espcoredump.py in order to decode the coredump.