espressif/arduino-esp32 #2894: i2c with isr

cnc4less, 2019-06-16:
i connected  2 pcf8575  16 bit expander IO   onw on    esp32 pin 25 and the other on pin 33 , if i try to read the i2c from the isr it crashes and found the only way to do it is from the loop as u can see below, did any find a way to fix this problem and read the i2c bus from inside the isr function? the goal from this is to have many encoders with simple circuit. appreciate ur cooperation.

#include <Arduino.h>
#include <Wire.h>

int address_b = 0x21; //0100000 (7bit)
int address_e = 0x20; //0100000 (7bit)

uint16_t dataReceive_b; //16bit data received
uint16_t dataReceive_e; //16bit data received

uint16_t dataSend_b;  //16bit data sent
uint16_t dataSend_e;  //16bit data sent

String  myStr_b , myStr_e ;
uint8_t error;

uint8_t hi_b, lo_b;
uint8_t hi_e, lo_e;

//custom functions -----------------------------

//I2C/TWI success (transaction was successful).
static const uint8_t ku8TWISuccess                   = 0;
//I2C/TWI device not present (address sent, NACK received).
static const uint8_t ku8TWIDeviceNACK                = 2;
//I2C/TWI data not received (data sent, NACK received).
static const uint8_t ku8TWIDataNACK                  = 3;
//I2C/TWI other error.
static const uint8_t ku8TWIError                     = 4;

struct Button {
  const uint8_t PIN;
  uint32_t numberKeyPresses;
  bool pressed;
  
};

Button button1 = {25, 0, false};
Button button2 = {33, 0, false};

void IRAM_ATTR isr(void* arg) {
  Button* s = static_cast<Button*>(arg);
  s->numberKeyPresses += 1;
  s->pressed = true;

//readall();
}

void IRAM_ATTR isr() {
  button2.numberKeyPresses += 1;
  button2.pressed = true;
 // readall();
}

void setup() {
  dataSend_e = word(B11111111, B11111111);
  dataSend_b = word(B11111111, B11111111);
  
  
  Wire.begin(21, 22, 400000);
  write_e(dataSend_e);
   write_e(dataSend_b);
  Serial.begin(115200);
  pinMode(button1.PIN, INPUT_PULLUP);
  attachInterruptArg(button1.PIN, isr, &button1, FALLING);
  pinMode(button2.PIN, INPUT_PULLUP);
  attachInterrupt(button2.PIN, isr, FALLING);
}

void loop() {
  if (button1.pressed) {
    Serial.printf("Button 1 has been pressed %u times\n", button1.numberKeyPresses);
    button1.pressed = false;
    readall();
  }
  if (button2.pressed) {
    Serial.printf("Button 2 has been pressed %u times\n", button2.numberKeyPresses);
    button2.pressed = false;
  }
  static uint32_t lastMillis = 0;
  if (millis() - lastMillis > 10000) {
    lastMillis = millis();
    //  detachInterrupt(button1.PIN);
  }
// readall();
//  delay(100);
}

void readall ()
{
  Wire.beginTransmission(address_e);
  error = Wire.endTransmission();

  Wire.requestFrom(address_e, 2);

  if (error == ku8TWISuccess) { //okay!
    Wire.requestFrom(address_e, 2);
    if (Wire.available()) {
      lo_e = Wire.read();
      hi_e = Wire.read();
    }
  dataReceive_e = (word(hi_e, lo_e)); //joing bytes
myStr_e =  String(~dataReceive_e, BIN);
Serial.println( myStr_e);
}
}

void write_e(uint16_t dt) {
  Wire.beginTransmission(address_e);
  Wire.write(lowByte(dt));
  Wire.write(highByte(dt));
  error = Wire.endTransmission();
  if (error == ku8TWISuccess) { //okay!

  }
  else { //we have an error
    //do something here if you like
  }
    Wire.beginTransmission(address_b);
  Wire.write(lowByte(dt));
  Wire.write(highByte(dt));
  error = Wire.endTransmission();
  if (error == ku8TWISuccess) { //okay!

  }
  else { //we have an error
    //do something here if you like
  }
}

atanisoft, 2019-06-16:
An ISR must execute quickly and executing a blocking operation inside an ISR is not a supportable programming model.

Additionally, Wire is not tagged as IRAM_ATTR (nor should it be) so it is not accessible within an ISR context.

stickbreaker, 2019-06-16:
@cnc4less No, you cannot call `Wire()` from an Interrupt Context. To achieve what you describe, I would recommend you create a separate task that does the `Wire()` communications, This task would just be an infinite loop that hangs in a call to `xTaskNotifyWait()`.  Inside your ISR code you use [`xTaskNotifyFromISR()`](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/freertos.html#_CPPv418xTaskNotifyFromISR12TaskHandle_t8uint32_t13eNotifyActionP10BaseType_t) to "wakeup" the waiting task.  [FreeRTOS docs](http://www.FreeRTOS.org/RTOS-task-notifications.html)

`Wire()` and the 'I2C' subsystem is not designed to support multiple tasks concurrently using it.  So, All `Wire()` operations would need to be handled by that task. The `xTaskNotify()` functions can be used to pass control information through the notification.  I would probably use a bit value to control what I2C operation you want the task to do. [FreeRTOS Task Notification reference](http://www.freertos.org/RTOS-task-notifications.html) 
 >Task notifications can update the receiving task’s notification value in the following ways:
>
>* Set the receiving task’s notification value without overwriting a previous value
>* Overwrite the receiving task’s notification value
>* Set one or more bits in the receiving task’s notification value
>* Increment the receiving task’s notification value

Chuck.

closed 2019-06-16 by everslick

cnc4less, 2019-06-16:
tried the esp task still not what it should if you have more than one interrupt, the example below
will show 3 ways to present the interrupts , one from the isr handle,, task  and iar task notify, , if in the task or loop still it will just read it on time till i trigger the 2nd interrupt, how to fix this problem.

const byte interruptPin1 = 25;
const byte interruptPin2 = 33;

 
SemaphoreHandle_t syncSemaphore1;
SemaphoreHandle_t syncSemaphore2;

int nTasks=2;

 
void IRAM_ATTR handleInterrupt1() {
  xSemaphoreGiveFromISR(syncSemaphore1, NULL);
  Serial.println("Motion detected isr 1");

}

 void IRAM_ATTR handleInterrupt2() {
  xSemaphoreGiveFromISR(syncSemaphore2, NULL);
   Serial.println("Motion detected isr 2");

}

 void producerTask( void * parameter )
{
 
    xSemaphoreTake(syncSemaphore1, portMAX_DELAY);
    Serial.println("Motion detected 11");
   //   xQueueSend(queue, &i, portMAX_DELAY);
 
   //xSemaphoreGive(syncSemaphore1); 
 
     vTaskDelete( NULL );
 
}

void consumerTask( void * parameter)
{
    int element;
   
    xSemaphoreTake(syncSemaphore2, portMAX_DELAY);
    Serial.println("Motion detected 22");
 
      vTaskDelete( NULL );
 
}

void setup() {
 
  Serial.begin(115200);
 
   syncSemaphore1 = xSemaphoreCreateBinary();
   syncSemaphore2 = xSemaphoreCreateBinary();
 
  pinMode(interruptPin1, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(interruptPin1), handleInterrupt1, RISING);

   pinMode(interruptPin2, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(interruptPin2), handleInterrupt2, RISING);

  
  xTaskCreate(
                    producerTask,     // Task function. 
                    "Producer",       // String with name of task.
                    10000,            // Stack size in words.
                    NULL,             // Parameter passed as input of the task 
                    1,                // Priority of the task.
                    NULL);            // Task handle.
 
 
  xTaskCreate(
                    consumerTask,     // Task function. 
                    "Consumer",       // String with name of task. 
                    10000,            // Stack size in words. 
                    NULL,             // Parameter passed as input of the task
                    1,                // Priority of the task. 
                    NULL);            // Task handle. 
 
 

  Serial.println("Tasks launched and semaphore passed...");
}
 
void loop() {
 delay(1000);
    xSemaphoreTake(syncSemaphore1, portMAX_DELAY);
     Serial.println("Motion detected 1");
    

     xSemaphoreTake(syncSemaphore2, portMAX_DELAY);
     Serial.println("Motion detected 2");
   vTaskSuspend(NULL);
 
}

cnc4less, 2019-06-16:
modification : with infinite loop for;; in the tasks  now it works fine outside the loop.  just with isr triggers. 
is this approach acceptable or just a batch fix const byte interruptPin1 = 25;
const byte interruptPin2 = 33;

 
SemaphoreHandle_t syncSemaphore1;
SemaphoreHandle_t syncSemaphore2;

int nTasks=2;

 
void IRAM_ATTR handleInterrupt1() {
  xSemaphoreGiveFromISR(syncSemaphore1, NULL);
  Serial.println("Motion detected isr 1");

}

 void IRAM_ATTR handleInterrupt2() {
  xSemaphoreGiveFromISR(syncSemaphore2, NULL);
   Serial.println("Motion detected isr 2");

}

 void producerTask( void * parameter )
{
 for(;;) {
    xSemaphoreTake(syncSemaphore1, portMAX_DELAY);
    Serial.println("Motion detected 11");
   //   xQueueSend(queue, &i, portMAX_DELAY);
 
   //xSemaphoreGive(syncSemaphore1); 
 }
     vTaskDelete( NULL );
 
}

void consumerTask( void * parameter)
{
    int element;
   for(;;) {
    xSemaphoreTake(syncSemaphore2, portMAX_DELAY);
    Serial.println("Motion detected 22");
   }
      vTaskDelete( NULL );
   
}

void setup() {
 
  Serial.begin(115200);
 
   syncSemaphore1 = xSemaphoreCreateBinary();
   syncSemaphore2 = xSemaphoreCreateBinary();
 
  pinMode(interruptPin1, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(interruptPin1), handleInterrupt1, RISING);

   pinMode(interruptPin2, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(interruptPin2), handleInterrupt2, RISING);

  
  xTaskCreate(
                    producerTask,     // Task function. 
                    "Producer",       // String with name of task.
                    10000,            // Stack size in words.
                    NULL,             // Parameter passed as input of the task 
                    1,                // Priority of the task.
                    NULL);            // Task handle.
 
 
  xTaskCreate(
                    consumerTask,     // Task function. 
                    "Consumer",       // String with name of task. 
                    10000,            // Stack size in words. 
                    NULL,             // Parameter passed as input of the task
                    1,                // Priority of the task. 
                    NULL);            // Task handle. 
 
 

  Serial.println("Tasks launched and semaphore passed...");
}
 
void loop() {
 delay(1000);
   // xSemaphoreTake(syncSemaphore1, portMAX_DELAY);
   //  Serial.println("Motion detected 1");
    

   //  xSemaphoreTake(syncSemaphore2, portMAX_DELAY);
  //   Serial.println("Motion detected 2");
   vTaskSuspend(NULL);
 
}

cnc4less, 2019-06-16:
when triggering each interrupt now works fine ?.

13:48:01.043 -> Motion detected 22
13:48:01.320 -> Motion detected isr 2
13:48:01.320 -> Motion detected 22
13:48:01.320 -> Motion detected isr 2
13:48:01.320 -> Motion detected 22
13:53:43.799 -> Motion detected isr 1
13:53:43.799 -> Motion detected 11
13:53:43.834 -> Motion detected isr 1
13:53:43.834 -> Motion detected 11
13:53:44.389 -> Motion detected isr 1
13:53:44.389 -> Motion detected 11
13:53:44.436 -> Motion detected isr 1
13:53:44.436 -> Motion detected 11

atanisoft, 2019-06-16:
You shouldn't call Serial.print or any methods on it in any ISR.

stickbreaker, 2019-06-16:
@cnc4less Try this code:
```c++
/*@cnc4less
Author
cnc4less commented 29 minutes ago •

tried the esp task still not what it should if you have more than one interrupt, the example below
will show 3 ways to present the interrupts , one from the isr handle,, task and iar task notify, , if in the task or loop still it will just read it one time ONLY till i trigger the 2nd interrupt then rest the esp32 how to fix this problem.
*/
const byte interruptPin1 = 25;
const byte interruptPin2 = 33;

SemaphoreHandle_t syncSemaphore1;
SemaphoreHandle_t syncSemaphore2;

int nTasks=2;

void IRAM_ATTR handleInterrupt1() {
xSemaphoreGiveFromISR(syncSemaphore1, NULL);
// can't use Serial in interrupt
// Serial.println("Motion detected isr 1"); 

}

void IRAM_ATTR handleInterrupt2() {
xSemaphoreGiveFromISR(syncSemaphore2, NULL);
//Serial.println("Motion detected isr 2");

}

void producerTask( void * parameter )
{
/* Block for 1 second */
const TickType_t xDelay = 1000 / portTICK_PERIOD_MS;
while( true) {
  
	if(xSemaphoreTake(syncSemaphore1, xDelay)){
		Serial.printf("Motion detected 11, %d pending\n",uxSemaphoreGetCount(syncSemaphore1));
	} 
	else {
		Serial.printf("no synch1 in last second\n");
	}
}
 vTaskDelete( NULL );

}

void consumerTask( void * parameter)
{
/* Block for 1 second */
const TickType_t xDelay = 1000 / portTICK_PERIOD_MS;
while(true){
	if(xSemaphoreTake(syncSemaphore2, xDelay)){
		Serial.printf("Motion detected 22, %d pending\n",uxSemaphoreGetCount(syncSemaphore2));
	}
	else {
		Serial.println(" no motion in last Second ");
	}
}
vTaskDelete( NULL );

}

void setup() {

Serial.begin(115200);

syncSemaphore1 = xSemaphoreCreateCounting(1024,0); // Max 1024 before give will return fail, init at 0
syncSemaphore2 = xSemaphoreCreateCounting(1024,0);

pinMode(interruptPin1, INPUT_PULLUP);
attachInterrupt(digitalPinToInterrupt(interruptPin1), handleInterrupt1, RISING);

pinMode(interruptPin2, INPUT_PULLUP);
attachInterrupt(digitalPinToInterrupt(interruptPin2), handleInterrupt2, RISING);

xTaskCreate(
producerTask, // Task function.
"Producer", // String with name of task.
10000, // Stack size in words.
NULL, // Parameter passed as input of the task
1, // Priority of the task.
NULL); // Task handle.

xTaskCreate(
consumerTask, // Task function.
"Consumer", // String with name of task.
10000, // Stack size in words.
NULL, // Parameter passed as input of the task
1, // Priority of the task.
NULL); // Task handle.

Serial.println("Tasks launched and semaphore passed...");
}

void loop() {
delay(100);
unsigned long count =uxSemaphoreGetCount(syncSemaphore1);
if(count>0) {
	Serial.printf("Pending Motion 1 =%d\n",count);
}
count =uxSemaphoreGetCount(syncSemaphore2);
if(count>0) {
	Serial.printf("Pending Motion 2 =%d\n",count);
}

}
```
Chuck.

stickbreaker, 2019-06-16:
@cnc4less There is a consideration when using Counting Semaphores, the ESP32 implementation uses a Queue to implement the Semaphore, so when I "created" the Counting Semaphore with a max count of 1024, it actually created a 1024 element Queue, this wasted(imo) 1024 Words (32bits) of RAM memory.  It might be better to use a smaller max count, something more reasonable. 16, 32.  It all depends on your latency, and if you need to service each interrupt or can "miss" a few.

Chuck.

cnc4less, 2019-06-16:
@stickbreaker  thank you , very smooth operation  , it is working with better details ,  but why always returning  0 pending always  ,no matter how fast i trigger the interrupts still 0 pending, next week will write a function to read the i2c pins from the pcf8575 to read all encoders and see how all shall fir together.

15:34:20.271 ->  no motion in last Second 
15:34:21.146 -> no synch1 in last second
15:34:21.273 ->  no motion in last Second 
15:34:21.320 -> Motion detected 22, 0 pending
15:34:21.320 -> Motion detected 11, 0 pending
15:34:21.695 -> Motion detected 22, 0 pending
15:34:22.147 -> Motion detected 11, 0 pending
15:34:22.401 -> Motion detected 11, 0 pending

stickbreaker, 2019-06-17:
The 1024 was just incase the process/consumer tasks took longer, I don't know your hardware/code. So, i cya'd myself. You should code so that either a missed interrupt causes no problems or never miss an interrupt. Personally i choose the first. Handle over lapping interrupts.

Since you are never seeing overlapping reduce the max down 16.

The initial at 0 will cause the task to hang until it see the interrupt add semaphoregive's.  Change it to 10 and see the result.

Chuck.

stickbreaker, 2019-06-17:
Increase the length of debug text in the tasks, overflow the 128byte serial buffer. Manually pressing  a button could never happen faster than 10 times a second.  At 240MHz with two cores that is enough time to execute 48million processor instructions. I doubt my example compiles down to more than 500 instructions per interrupt.

cnc4less, 2019-06-17:
i see,  I will do some testing to find out  the max interrupt interval /sec capacity of this chip, for now  all working good , thank you

22:15:26.249 -> 0000000000000000
22:15:26.249 -> enc1   000
22:15:26.249 -> 00
22:15:26.249 -> -12
22:15:26.249 -> enc2   000
22:15:26.249 -> 00
22:15:26.249 -> 12
22:15:26.249 -> enc3   000
22:15:26.249 -> 00
22:15:26.249 -> 16
22:15:26.249 -> enc4   000
22:15:26.249 -> 00
22:15:26.249 -> 12