espressif/arduino-esp32 #2896: SSL/TLS handshake timeout with server preferring DHE cipher (with 4096 bit DH key)

chaosbiber, 2019-06-16:
Using *WiFiClientSecure* or *HTTPClient* I couldn't connect to **some** servers, especially running my customised nginx configuration.
Using wireshark and later enabling verbose output I found that the handshake wouldn't finish.

```
16:45:09.102 -> Starting connection to server...
16:45:09.102 -> [V][ssl_client.cpp:53] start_ssl_client(): Free internal heap before TLS 278304
16:45:09.102 -> [V][ssl_client.cpp:55] start_ssl_client(): Starting socket
[V][ssl_client.cpp:90] start_ssl_client(): Seeding the random number generator
16:45:09.139 -> [V][ssl_client.cpp:99] start_ssl_client(): Setting up the SSL/TLS structure...
16:45:09.139 -> [V][ssl_client.cpp:112] start_ssl_client(): Loading CA cert
16:45:09.172 -> [V][ssl_client.cpp:177] start_ssl_client(): Setting hostname for TLS session...
16:45:09.172 -> [V][ssl_client.cpp:192] start_ssl_client(): Performing the SSL/TLS handshake...
[E][ssl_client.cpp:33] handle_error(): SSL - The connection indicated an EOF
16:45:44.955 -> [E][ssl_client.cpp:35] handle_error(): MbedTLS message code: -29312
16:45:44.955 -> [E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -29312
16:45:44.991 -> [V][ssl_client.cpp:245] stop_ssl_socket(): Cleaning SSL connection.
16:45:44.991 -> Connection failed!
```

After changing the server's cipher suite to prefer ECDH it works:
```
16:43:56.575 -> Starting connection to server...
16:43:56.575 -> [V][ssl_client.cpp:53] start_ssl_client(): Free internal heap before TLS 278448
16:43:56.575 -> [V][ssl_client.cpp:55] start_ssl_client(): Starting socket
[V][ssl_client.cpp:90] start_ssl_client(): Seeding the random number generator
16:43:56.747 -> [V][ssl_client.cpp:99] start_ssl_client(): Setting up the SSL/TLS structure...
16:43:56.747 -> [V][ssl_client.cpp:112] start_ssl_client(): Loading CA cert
16:43:56.747 -> [V][ssl_client.cpp:177] start_ssl_client(): Setting hostname for TLS session...
16:43:56.782 -> [V][ssl_client.cpp:192] start_ssl_client(): Performing the SSL/TLS handshake...
[V][ssl_client.cpp:213] start_ssl_client(): Verifying peer X.509 certificate...
16:43:57.991 -> [V][ssl_client.cpp:222] start_ssl_client(): Certificate verified.
16:43:57.991 -> [V][ssl_client.cpp:237] start_ssl_client(): Free internal heap after TLS 236888
16:43:57.991 -> [V][ssl_client.cpp:276] send_ssl_data(): Writing HTTP request...
```

Comparing the listed ciphers from ssllab's server test and howsmyssl's client test the first common cipher with the problematic server config was
`TLS_DHE_RSA_WITH_AES_256_GCM_SHA384 (0x9f)   DH 4096 bits   FS`
I guess the problem is the big custom dh-key?

It's not a big problem as my newer config already preferred the curves, but still wanted to report it. Can we check if the DH is too big before choosing that cipher?

### Hardware:
Board:							NodeMCU ESP-32s
Core Installation version:			1.0.2
IDE name:						Arduino IDE

### Sketch: 
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiClientSecure/examples/WiFiClientSecure/WiFiClientSecure.ino
replacing server with a problematic one

chegewara, 2019-06-16:
If i am correct esp-idf works with 2048 bits certificates,  i think me and some other people got issues with 1024 or 4096 bits keys. But it was some time ago so i may be wrong.

closed 2019-08-29 by stale[bot]

bjdshp, 2022-03-04:
Getting same failure using this....

https://github.com/mobizt/Firebase-ESP-Client

Tried putting a 100 mF capacitor between ground and 3.3, but couldn't get it to write firmware.  Any suggestions?

salakorte, 2023-05-30:
sane error