espressif/arduino-esp32 #2897: SD begin() end() memory leak

cybercatnet, 2019-06-17:
### Hardware:
Board:							ESP32 Dev Module DOIT
Core Installation version:			1.0.1
IDE name:						Arduino IDE and Platform.io
Flash Frequency:					80Mhz?
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
In my program i have memory leak and i made a simple test.
SD.end() doesnt free all of his heap memory.
My program needs to check if the sd card is connected before touch files, and if is not i need to SD.begin() again in the case of the sd card is again connected. 

### Sketch:  
#include <SPI.h>
#include <SD.h>

void setup() {
  Serial.begin(9600);
}

void loop() {
  SD.begin(5);
  Serial.println((unsigned long)ESP.getFreeHeap());
  SD.end();
  delay(1000);
}
```

### Debug Messages:
```
ESP.getFreeHeap:
249644
249624
249604
249584
249564
249544
249524
249504
249484
249464
249444
249424
249404
249384
249364
249344
249324
249304
249284
249264
249244
249224

```

cybercatnet, 2019-08-31:
still

atanisoft, 2019-09-19:
@cybercatnet can you try updating [this](https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/src/sd_diskio.cpp#L654-L656) block to:
```
    if (card->base_path) {
        err = esp_vfs_fat_unregister_path(card->base_path);
        free(card->base_path);
    }
```
adding the free line.

sla256, 2019-11-29:
Seeing the same issue. Similar use case i.e. need to support removing / inserting SD card run time. Observing 20 bytes leaking as well.

sla256, 2019-11-30:
> @cybercatnet can you try updating [this](https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/src/sd_diskio.cpp#L654-L656) block to:
> 
> ```
>     if (card->base_path) {
>         err = esp_vfs_fat_unregister_path(card->base_path);
>         free(card->base_path);
>     }
> ```
> 
> adding the free line.
> 
> @me-no-dev fyi, the card->base_path is being leaked...

by the way, yes - this change fixes the issue

closed 2020-02-12 by stale[bot]

Anx2k, 2021-07-19:
Just an FYI, this leak still happens 1.0.6 and this fix doesn't seem to be incorporated.  I manually applied the fix and it appears to resolve the issue for me as well.

VojtechBartoska, 2021-09-22:
@SuGlider can we close this properly?

SuGlider, 2021-09-22:
Yeah, we can close it, as it was fixed by #5419
Thanks!

PR #5419 (2021-07-19): Fixes SD begin() end() memory leak #2897
Fixes: #2897