espressif/arduino-esp32 #2898: ESP32 with PSRAM

perf3ct1on666, 2019-06-17:
i purchased an esp32-wrover-b. i wanna to know how should i apply the PSRAM? it seen like very much different with assign xTask. So far what i know is just this few line code function:
 heap_caps_malloc(size, MALLOC_CAP_SPIRAM);   // allocate memory with size of PSRAM into application
 malloc(xxxx) // put the function into it will assign the function (am i right?????)
 free(xxxx)     // release the memory and remove the task... (use after malloc() to release memory)

so far i cant really find the example sketch from internet to apply and use PSRAM into my study...
so i hope some of you help please... this study quite important to me...

lbernstone, 2019-06-20:
Make sure to select a wrover type board in the IDE.  It will automatically set initialize your psram then.
There is an arduino function [ps_malloc](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-psram.h#L25) which will work like malloc.

perf3ct1on666, 2019-06-21:
@lbernstone 
i use esp32 dev module with enable PSRAM. btw i tried with wrover also seem it same result.
so, isit the enable PSRAM and malloc a size of psram, esp32 processor will auto arranging some memory to the PSRAM?

lbernstone, 2019-06-21:
There is currently nothing automatic about it.  If you want to store data in psram, you must manually allocate and free it.

closed 2019-09-03 by stale[bot]