espressif/arduino-esp32 #2900: [Feature Request] It would be useful to know if a received AsyncUDPPacket is part of (and which part of) a fragmented packet

ssilverman, 2019-06-18:
When I receive large packets, they are often fragmented. It would be useful if `AsyncUDPPacket`, via `AsyncUDP`'s `onPacket` call, contained information about if and how the packet is fragmented, for example, the _fragment offset_ and _total length_ information.

r1dd1ck, 2019-06-18:
@ssilverman This "fragmentation" you are experiencing is obviously caused by the application that sends it - most probably because it uses a 1024-byte payload limit/buffer where payloads larger than that get automatically split into multiple packets. I suppose you have to handle this type of "fragmentation" on your own @ sender/receiver ends.

ssilverman, 2019-06-18:
There should be enough information in the packet to be able to pass along fragmentation information. The fact that it's fragmented should be detectable. It's impossible, currently, to handle the fragmentation on the receiver with the current API.

Fragmentation or the fact that it's happening isn't relevant to this issue, as you're suggesting. I'm only suggesting to enhance the API to provide information that already should be in the fragmented packets, specifically, the _fragmentation offset_ and _total length_ information. See the Wikipedia link in my first post in this issue.

Unless, of course, the ESP32 low-level APIs don't provide access to this information?

ssilverman, 2019-06-18:
I just went through the `AsyncUDP.cpp` code and having access to `pbuf.tot_len`, in addition to the count inside `_recv`, are things I'm looking for. Let me see about a pull request...

ssilverman, 2019-06-18:
See the new pull request: #2904.

r1dd1ck, 2019-06-18:
So you are just looking to extend the functionality of the `AsyncUDP` arduino library - OK.

But have you checked that `pbuf.tot_len` contains the **total payload length** before the data was split at sender side?

ssilverman, 2019-06-19:
It doesn't matter what my network does; many networks are different. All I'm saying is it's useful to be able to access `pbuf.tot_len`. Are you suggesting this field is never set to something other than the same value as `pbuf.len`?

lbernstone, 2019-06-20:
UDP does not guarantee anything.  If you want to maintain a sequence in your data, you need to put an index number on each packet.  If you want to ensure that it passes through a network without getting fragmented, you need to set your packet size below a lowest common denominator MTU (typically 512 bytes).  If you want some guarantees on your data flow, there is a protocol for that called TCP.  This is not an issue with arduino, esp32 or MCUs in general.  If you want to discuss this issue, the IEEE has RFCs...

ssilverman, 2019-06-20:
These are all good points. In your opinion, would it or would it not be useful to include access to `pbuf.total_len`? It’s turning out that it’s not useful on my network...

r1dd1ck, 2019-06-21:
@ssilverman 
As I already wrote in my first comment, the "fragmentation" you are experiencing has nothing to do with packet fragmentation caused by the network. Your payload gets **split** into multiple single packets **by the sender**.

Packet fragmentation due to network constraints is handled by the **IP layer** and has nothing to do with the chosen transfer protocol (UDP, TCP, etc.) - eg. the fragmentation/reassembly process is courtesy of the IP stack (lwIP in this case).

What you are asking for would essentially be an **extension** of the UDP protocol, and as already pointed out by @lbernstone above - this is not the right place to discuss such things.

I hope I was clear enough and you understand the issue now :slightly_smiling_face:

ssilverman, 2019-06-24:
Thanks for taking the time to explain. It clarifies.

closed 2019-06-24 by ssilverman