espressif/arduino-esp32 #2905: serveStatic does not serve folders

vshymanskyy, 2019-06-19:
I.e, we need to to specify each file separately:
```
server.serveStatic("/img/favicon.png", SPIFFS, "/img/favicon.png");
server.serveStatic("/img/logo.png", SPIFFS, "/img/logo.png");
...
```

Instead of just:
```
server.serveStatic("/img/");
```

This works on ESP8266, but not on ESP32 currently (breaks compatibility).

atanisoft, 2019-08-02:
```
server.serveStatic("/img", SPIFFS, "/img");
```

This appears to work for both ESP8266 and ESP32...

vshymanskyy, 2019-08-20:
@atanisoft thanks, but unfortunately not working for me.
`index.html` is loaded without issues from the same SPIFFS.
For images I get:
```
Not found: /img/logo.png
```
When I specify each file separately, it works.

I'm using the latest v1.0.2 from Arduino Board Manager.

atanisoft, 2019-08-20:
This has nothing to do with the Arduino esp32 version but more likely an issue in your code. Can you share the rest of your code and we can look for the real issue in it

vshymanskyy, 2019-08-21:
@atanisoft here is a simple sketch that displays my problem:
[SPIFFS_test.zip](https://github.com/espressif/arduino-esp32/files/3523178/SPIFFS_test.zip)

atanisoft, 2019-08-21:
Try:
```
server.serveStatic("/", SPIFFS, "/");
```
Also turn up the core debug level to VERBOSE (in Arduino IDE set the board type to esp32 dev and there is a debug log level option in the menu). This debug logging should tell what is going wrong.

Also you can rename index.html to index.htm and it should automatically serve this when navigating to /, there is unfortunately no way to specify the default html file to serve when navigating to a directory when serving static content. There is with https://github.com/me-no-dev/ESPAsyncWebServer which provides all of the same features as the WebServer library you are using now and as added bonuses is faster and doesn't require you to add anything in the loop() method!

closed 2019-11-03 by stale[bot]