espressif/arduino-esp32 #2907: LAN8720 with ESP32 : Timed out waiting for PHY

esp13, 2019-06-19:
Hi,

I'm trying to connect my ESP32 (the "yellow" development board) to my Ethernet network (without Wifi) through a LAN8720 board. But I'm stuck at the classic error :
 `emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0xffff
emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0xffff`

What I have done:
I modified my LAN8720 Board ($2 board from aliexpress) using this: [https://sautter.com/blog/ethernet-on-esp32-using-lan8720/](https://sautter.com/blog/ethernet-on-esp32-using-lan8720/)
The disposition on my LAN8720 is a little different so I adapted:
[image]
[image]

I wired it again like written on [https://sautter.com/blog/ethernet-on-esp32-using-lan8720/](https://sautter.com/blog/ethernet-on-esp32-using-lan8720/):
[image]

And finally with the arduino IDE 1.8.6 I uploaded the "Exemple->Wifi->ETH_LAN8720_internal_clock" sketch after modifying
ETH_ADDR        0 to 1
define ETH_MDC_PIN     15 to 23
define ETH_MDIO_PIN    2 to 18

The full sketch:
```
/*
    This sketch shows how to configure different external or internal clock sources for the Ethernet PHY
*/

#include <ETH.h>

/* 
   * ETH_CLOCK_GPIO0_IN   - default: external clock from crystal oscillator
   * ETH_CLOCK_GPIO0_OUT  - 50MHz clock from internal APLL output on GPIO0 - possibly an inverter is needed for LAN8720
   * ETH_CLOCK_GPIO16_OUT - 50MHz clock from internal APLL output on GPIO16 - possibly an inverter is needed for LAN8720
   * ETH_CLOCK_GPIO17_OUT - 50MHz clock from internal APLL inverted output on GPIO17 - tested with LAN8720
*/
#define ETH_CLK_MODE    ETH_CLOCK_GPIO17_OUT

// Pin# of the enable signal for the external crystal oscillator (-1 to disable for internal APLL source)
#define ETH_POWER_PIN   -1

// Type of the Ethernet PHY (LAN8720 or TLK110)
#define ETH_TYPE        ETH_PHY_LAN8720

// I²C-address of Ethernet PHY (0 or 1 for LAN8720, 31 for TLK110)
#define ETH_ADDR        1

// Pin# of the I²C clock signal for the Ethernet PHY
#define ETH_MDC_PIN     23

// Pin# of the I²C IO signal for the Ethernet PHY
#define ETH_MDIO_PIN    18

static bool eth_connected = false;

void WiFiEvent(WiFiEvent_t event) {
  switch (event) {
    case SYSTEM_EVENT_ETH_START:
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname("esp32-ethernet");
      break;
    case SYSTEM_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      break;
    case SYSTEM_EVENT_ETH_GOT_IP:
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
      }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps");
      eth_connected = true;
      break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case SYSTEM_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default:
      break;
  }
}

void testClient(const char * host, uint16_t port) {
  Serial.print("\nconnecting to ");
  Serial.println(host);

  WiFiClient client;
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    return;
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    Serial.write(client.read());
  }

  Serial.println("closing connection\n");
  client.stop();
}

void setup() {
  Serial.begin(115200);
  WiFi.onEvent(WiFiEvent);
  ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE);
}

void loop() {
  if (eth_connected) {
    testClient("google.com", 80);
  }
  delay(10000);
}
```

I know this is probably not an issue, but something I missed or am doing wrong.
So if I ask on the wrong place, please tell me where I could ask, so I can move it.

Sorry for my terrible English.

closed 2019-09-02 by stale[bot]

pxdelta, 2019-10-22:
Hello @esp13, I have the same problem with same connections, did you succeed?

pxdelta, 2019-10-22:
My issue is fixed after removing VT2 (GPIO0 transistor) from "DOIT ESP32 Devkit V1" board

esp13, 2019-12-24:
> My issue is fixed after removing VT2 (GPIO0 transistor) from "DOIT ESP32 Devkit V1" board

Hi, thank you for your answer :)

Today i'm giving a new try after updating my esp32 firmware to esp32-idf3-20191224-v1.12-5-g42e45bd69.bin (support LAN and PPP but not bluetooth)
I'm getting the same errors as in my initial post.

So, how did you do to success?
- Did you modify your LAN8720 as I do? Or did you let it as it comes?
- Did you remove the crystal from your LAN8720?
- Witch sketch did you use and with witch configuration?
- Witch esp32 firmware did you use?
- Is my ESP32 board (yellow pin 19x2) similar to yours?
- Why did you remove the VT2 transistor, how do you find this?
- What are sides effects from removing this VT2 transistor?

This is a lot of questions ^^ Sorry !

DarlanJurak, 2020-03-25:
> Hi,
> 
> I'm trying to connect my ESP32 (the "yellow" development board) to my Ethernet network (without Wifi) through a LAN8720 board. But I'm stuck at the classic error :
> `emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0xffff emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0xffff`
> 
> What I have done:
> I modified my LAN8720 Board ($2 board from aliexpress) using this: https://sautter.com/blog/ethernet-on-esp32-using-lan8720/
> The disposition on my LAN8720 is a little different so I adapted:
> [image]
> [image]
> 
> EDIT: After reading that the lenght of wires can be an issue at 50MHz, I wired it with 10cm wires (instead of 20cm on the photo), still the same.
> 
> I wired it again like written on https://sautter.com/blog/ethernet-on-esp32-using-lan8720/:
> [image]
> 
> And finally with the arduino IDE 1.8.6 I uploaded the "Exemple->Wifi->ETH_LAN8720_internal_clock" sketch after modifying
> ETH_ADDR 0 to 1
> define ETH_MDC_PIN 15 to 23
> define ETH_MDIO_PIN 2 to 18
> 
> The full sketch:
> 
> ```
> /*
>     This sketch shows how to configure different external or internal clock sources for the Ethernet PHY
> */
> 
> #include <ETH.h>
> 
> /* 
>    * ETH_CLOCK_GPIO0_IN   - default: external clock from crystal oscillator
>    * ETH_CLOCK_GPIO0_OUT  - 50MHz clock from internal APLL output on GPIO0 - possibly an inverter is needed for LAN8720
>    * ETH_CLOCK_GPIO16_OUT - 50MHz clock from internal APLL output on GPIO16 - possibly an inverter is needed for LAN8720
>    * ETH_CLOCK_GPIO17_OUT - 50MHz clock from internal APLL inverted output on GPIO17 - tested with LAN8720
> */
> #define ETH_CLK_MODE    ETH_CLOCK_GPIO17_OUT
> 
> // Pin# of the enable signal for the external crystal oscillator (-1 to disable for internal APLL source)
> #define ETH_POWER_PIN   -1
> 
> // Type of the Ethernet PHY (LAN8720 or TLK110)
> #define ETH_TYPE        ETH_PHY_LAN8720
> 
> // I²C-address of Ethernet PHY (0 or 1 for LAN8720, 31 for TLK110)
> #define ETH_ADDR        1
> 
> // Pin# of the I²C clock signal for the Ethernet PHY
> #define ETH_MDC_PIN     23
> 
> // Pin# of the I²C IO signal for the Ethernet PHY
> #define ETH_MDIO_PIN    18
> 
> 
> static bool eth_connected = false;
> 
> void WiFiEvent(WiFiEvent_t event) {
>   switch (event) {
>     case SYSTEM_EVENT_ETH_START:
>       Serial.println("ETH Started");
>       //set eth hostname here
>       ETH.setHostname("esp32-ethernet");
>       break;
>     case SYSTEM_EVENT_ETH_CONNECTED:
>       Serial.println("ETH Connected");
>       break;
>     case SYSTEM_EVENT_ETH_GOT_IP:
>       Serial.print("ETH MAC: ");
>       Serial.print(ETH.macAddress());
>       Serial.print(", IPv4: ");
[... 155 lines cut ...]
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    Serial.write(client.read());
  }

  Serial.println("closing connection\n");
  client.stop();
}

void setup() {
  Serial.begin(115200);
  WiFi.onEvent(WiFiEvent);
  ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE);
}

void loop() {
  if (eth_connected) {
    testClient("google.com", 80);
  }
  delay(10000);
}

```

sauttefk, 2020-03-25:
I still have to emphasize 50MHz signals will not work reliable with such long cables flying in the breeze.

workpage2, 2020-05-13:
Hi guys.
There are two questions.
1. Will the link LED light when I connect power to the lan8720 without initialization?
2. I repeated the connection above. The link LED does not work. Error output:

        E (1034) emac: Timed out waiting for PHY register 0x2 to have value 0x0007(mask 0xffff). Current value 0xffff
        E (2034) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0(mask 0xfff0). Current value 0xffff
        E (2034) emac: Initialise PHY device Timeout

workpage2, 2020-05-13:
I take a timeout. PCB lan8720 is defective. When 3.3 volts is turned on, the green LED should light up.

workpage2, 2020-05-17:
This web page really helped me.
http://arduino.ru/forum/apparatnye-voprosy/podklyuchenie-ethernet-lan8720-i-esp32-devkit-c-esp32-devkit-v1

AH-007, 2022-01-23:
Can anyone tell me where i am going wrong please:

[image]

trungkien20188, 2022-08-27:
With the above connection diagram, do we have Modbus TCP communication sample code. Thank you.

ThomasPerl, 2023-03-18:
I have the same issue as @AH-007. But i'm having a different Board and don't know what Pin is GPIO00 and what is GPIO 17?
I have this Board here. I think GPIO17 is TXD2? But still I don't find GPIO00

ThomasPerl, 2023-03-18:
I think i solved something. My Serial Monitor is now printing ETH Started and ETH Connected without any Errormessage of power failure or something.
But it doesn't get an IP Address. How can this happen?

Takashi426, 2023-10-06:
i too fasing same issue not able to print ip adress