espressif/arduino-esp32 #2908: mDNS-Library does not work for certain devices

MorpheusDe97, 2019-06-19:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.0
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                      no
Upload Speed:					921600
Computer OS:                                          Windows 7

### Description:
I am currently experimenting with the mDNS-Library, especially with the browseService-example. https://github.com/espressif/arduino-esp32/blob/master/libraries/ESPmDNS/examples/mDNS-SD_Extended/mDNS-SD_Extended.ino#L40

I have several mDNS-capable devices in my appartment, like apple-TVs, printers etc. But as well the shelly-switches, that actually work with an ESP8266. According to the documentation, the shelly-devices publish an http-tcp-mDNS-service. But I was kind of surprised that the lib-example didn't find any shelly-devices. https://shelly-api-docs.shelly.cloud/#mdns-discovery

After spending a couple of hours experimenting, I found out that the shelly mDNS service actually works, but there seems to be a kind of bug in the ESP-mDNS-library, as the error seems to be on the ESP-side. I attached the wireshark-file which proves, that the shelly-device actually seems to be working. 
[shelly_esp.zip](https://github.com/espressif/arduino-esp32/files/3308067/shelly_esp.zip)

But as well, there seem to be some differents between the mDNS-responses as well: 
In contrast to the printer, the shelly device sends all the records in one answer block. Furthermore, the first answer-block of the shelly device is mDNS/UDP, not a http-tcp record. 

Maybe anybody has got an idea on how to solve this issue! 

### Sketch:  (leave the backquotes for [code formatting]
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESPmDNS/examples/mDNS-SD_Extended/mDNS-SD_Extended.ino

### Debug Messages:
```
Browsing for service_http._tcp.local. ... [W][ESPmDNS.cpp:214] queryService(): No results found!
no services found
```

MorpheusDe97, 2019-06-22:
I can reproduce this issue using Arduino-ESP32 V1.0.2. Should I post this issue in the ESP-IDF thread?

atanisoft, 2019-08-02:
The ESPmDNS library is a very thin wrapper around the ESP-IDF mdns code. I'd suggest reroute this to the ESP-IDF repo with the information you provided here so they can address it in the mdns code directly.

closed 2019-10-15 by stale[bot]

jrickard, 2020-01-06:
I have this exact same issue with a Shelly EM device.  It shows up in the program DISCOVERY and I can http to shellyem-B9E7BC.local in Safari.  But ESP32 mDNS cannot detect it at all.

There seems to be zero interest in fixing this lame mDNS implementation.

atanisoft, 2020-01-06:
> There seems to be zero interest in fixing this lame mDNS implementation.

@jrickard Please open an issue against https://github.com/espressif/esp-idf and provide as much detail as you can since the arudino-esp32 mDNS library is just a thin wrapper around the mDNS code from esp-idf. If the issue was reported to the correct place then there would likely be some interest in sorting out why it is not working correctly.

me-no-dev, 2020-01-20:
Using the word "lame" also does not help much :D