espressif/arduino-esp32 #2909: How to use string as key in preferences

usrdes, 2019-06-20:
Hi,
I am trying to use preferences with strings as keys.

### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			24/Nov/2018
IDE name:							Arduino IDE 
Flash Frequency:					80Mhz
PSRAM enabled:                                          yes
Upload Speed:						921600
Computer OS:                                            Windows 7

### Description:
In my code (and compile messages ) below  I want to use keys as "item_no_0", "item_no_1".
However the key may also be assembled from different strings like "item_1" or "Store_2" etc.

The compile fails because the key parameter needs to be passed in as const char * and
not a string. 

How do I make the strings as keys in preferences work?

I tried to copy strings into a char *s : 
String key = "item_no_"
int k = 0;
String myKey = key+String(k);
char * s = myKey; <-- this is incorrect and also fails compile.
Tried also  to copy as:
myKey(s, sizeof(myKey));  <-- again fails.

### Sketch: 
```

#include <Preferences.h>

Preferences preferences;

void setup() {
  Serial.begin(115200);
  Serial.println();

  preferences.begin("my-app", false);

  String key ="";
  String items = "Item_no_";
  int count =0;
  unsigned int value =0;

  for(count=0; count < 5; count++) {
    key = items+String(count);
    preferences.putUInt(key, count);
  }

  for(count=0;count < 5; count++) {
    key = items+String(count);
    Serial.print("key : "+key);
    value = preferences.getUInt(key, 0);
    Serial.printf(" value: %u\n", value);
    Serial.println();
  }
  // Close the Preferences
  preferences.end();

  // Wait 10 seconds
  delay(10000);
}

void loop() {}
```

### Debug Messages:
```
..\StartCounter.ino: In function 'void setup()':

StartCounter:33:35: error: no matching function for call to 'Preferences::putUInt(String&, int&)'

     preferences.putUInt(key, count);

                                   ^

In file included from ..\StartCounter.ino:12:0:

C:..esp32\libraries\Preferences\src/Preferences.h:39:16: note: candidate: size_t Preferences::putUInt(const char*, uint32_t)

         size_t putUInt(const char* key, uint32_t value);

                ^

C:..esp32\libraries\Preferences\src/Preferences.h:39:16: note:   no known conversion for argument 1 from 'String' to 'const char*'

StartCounter:39:39: error: no matching function for call to 'Preferences::getUInt(String&, int)'

     value = preferences.getUInt(key, 0);

                                       ^

In file included from ..\StartCounter.ino:12:0:

C:..esp32\libraries\Preferences\src/Preferences.h:56:18: note: candidate: uint32_t Preferences::getUInt(const char*, uint32_t)

         uint32_t getUInt(const char* key, uint32_t defaultValue = 0);

                  ^

C:..esp32\libraries\Preferences\src/Preferences.h:56:18: note:   no known conversion for argument 1 from 'String' to 'const char*'

exit status 1
no matching function for call to 'Preferences::putUInt(String&, int&)'

```

atanisoft, 2019-06-20:
use key.c_str() when calling the Preferences methods for get/put.

usrdes, 2019-06-20:
@atanisoft  That worked! Thank you very much.

closed 2019-06-20 by usrdes