espressif/arduino-esp32 #2910: Windows 10 PC can't access AP Mode.

GeorgeFlorian, 2019-06-20:
Board: ESP32 DevKit V4 with Wrover-B
OS: Windows 10
platform = https://github.com/platformio/platform-espressif32.git

I'm having a peculiar issue here. I can't access the WebServer when I am connected to the ESP32 AP Mode using Windows 10. But I can access the WebServer when I am connected to the AP using my Galaxy Note 9.

#and

After the PC fails to access the WebServer , when trying to connect with the phone to the AP it doesn't work. It seems like I am limited to one connection to the AP. It just doesn't work. No debug message, no nothing. Not even the phone returns an error. Which is rather strange I'd say...

This is how I set up AP Mode.
```
//------------------------- logOutput(String)
void logOutput(String string1) {
	circle.push(string1);	
	Serial.println(string1);	
}

logOutput((String)"Starting AP ... ");
  logOutput(WiFi.softAP(ssidAP, passwordAP) ? "Ready" : "WiFi.softAP failed ! (Password must be at least 8 characters long )");
  delay(500);
  logOutput((String)"Setting AP configuration ... ");
  logOutput(WiFi.softAPConfig(local_IP_AP, local_IP_AP, subnetAP) ? "Ready" : "Failed!");
  delay(500);
  logOutput((String)"Soft-AP IP address: ");
  logOutput(WiFi.softAPIP().toString());
```

Nothing strange until now.
This is the `-DCORE_DEBUG_LEVEL=5` aka `Verbose` :
```
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 15 - AP_STACONNECTED
dhcps: send_offer>>udp_sendto result 0
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
```

See the `dhcps: send_offer>>udp_sendto result 0` ?
I don't get that when I access the AP using the phone.

And that is all. No other verbose message.

DarvosBR, 2019-06-21:
Same here, Ap mode seems to be broken!!

`dhcps:` send_nak>>udp_sendto result 0`

Ap mode seems to broken, I already tried normal webserver and AsyncWebServer, both are not working properly, they just stop working from about 15 seconds after you connect with AP.

esp-idf seems to have this fixed already, but in arduino-esp32 is still broken

GeorgeFlorian, 2019-06-21:
That's just great !
Is there any other way to start/use AP ?

atanisoft, 2019-06-21:
Could this be related to #2899?

DarvosBR, 2019-06-21:
@atanisoft , well I did the test with heap memory and I can confirm the free heap memory size drops when the problem starts like mentioned on the issue you commented!

So Yes I do believe its related...

atanisoft, 2019-06-21:
@DarvosBR yes, that should work to test. I suspect there may be a few other LwIP settings which would be useful to adjust slightly to increase stability in the arduino-esp32 builds.

GeorgeFlorian, 2019-06-21:
> Could this be related to #2899?

Probably.

So you are saying that this should be a work-around:

> Actually, you don't need to re-compile the `arduino-esp32` core to change the buffer values.
> 
> All you need to do is:
> 
> * Edit `\tools\sdk\include\config\sdkconfig.h` and adjust the values to match `sdkconfig.default` from the `iperf` example folder.
> * Force a **full rebuild** when compiling your project (not sure how to do this directly from the UI, but changing the flash frequency or upload speed is usually enough to force it).
> 
> This applies to the **arduino** environment. I have no experience with PiO, so can't help you there.
> 
> You should see a **13-14kB drop** in free heap (as compared to arduino-esp32 defaults). If you don't see this drop, then the changes are not active, and you need to check what you did wrong 
> 
> **On a side note:** The buffer values from the `iperf` config are rather extreme, so later you should probably have a go at it and adjust them to best match your specific application needs.

Also, I've seen you mentioning a possible solution to the ESPAsyncWebServer's freeze. That would be something extraordinary indeed.

atanisoft, 2019-06-21:
@GeorgeFlorian I haven't tested the proposed rebuild but I don't think the steps outlined above will work since this is all pre-compiled into libesp32.a. The only solution I know of is to setup an esp-idf build (cmake or similar) with the arduino-esp32 sdkconfig, build and copy the libs over to the tools/sdk tree where they are today.

I'm not certain it will fix the async code but there may be a small chance that they are related.

GeorgeFlorian, 2019-06-21:
> @GeorgeFlorian I haven't tested the proposed rebuild but I don't think the steps outlined above will work since this is all pre-compiled into libesp32.a. The only solution I know of is to setup an [...]

That sounds like anything I have done. I've never used esp-idf and it sounds tricky to use.

As I am now, my best bet is to wait for a stable arduino-core 1.03 in which hopefully somebody will fix have fixed this issue.

DarvosBR, 2019-06-21:
@GeorgeFlorin Don't worry, I'm gonna try both solutions and have a return for you, independent of which solution works (if any do), I will tell you how and event provide file if needed!

I got your back George!

r1dd1ck, 2019-06-21:
@atanisoft
>  I haven't tested the proposed rebuild but I don't think the steps outlined above will work

so you have nothing to back your opinion up with, but you "think" it's garbage.. OK :grin:

I can only say that it worked for me - fixing the stack freeze issue #2899 (with **empirical evidence** to back that up). 

Those buffer parameters are not a fixed part of the pre-compiled core. You can easily check it by increasing the static buffer values and watch your free heap drop as a result.

And to also say something on-topic of #2910 - I don't see how it could be directly related, but who knows. 

Easy enough to try, right? :stuck_out_tongue_winking_eye:

atanisoft, 2019-06-21:
> so you have nothing to back your opinion up with, but you "think" it's garbage.. OK

Just experience from how the build process works in the environments I've used. Anything that is part of IDF is *precompiled* before any of the arduino-esp32 stack is compiled. Changes to sdkconfig (or sdkconfig.h) will usually require a recompilation of the IDF pieces for the changes to take effect. This is not an issue if you are using arduino as an IDF component though.

> Those buffer parameters are not a fixed part of the pre-compiled core. You can easily check it by increasing the static buffer values and watch your free heap drop as a result.

How is this possible, these are used only on the IDF side and IDF code is not recompiled inside Arduino IDE or in PIO when using the arduino framework option.

GeorgeFlorian, 2019-06-21:
> @GeorgeFlorin Don't worry, I'm gonna try both solutions and have a return for you, independent of which solution works (if any do), I will tell you how and event provide file if needed!
> 
> I got your back George!

That is great to hear, man ! Thank you @DarvosBR !

But I don't think that  https://github.com/espressif/arduino-esp32/issues/2899 is related. You and me both experience issues when trying to connect to the ESP while it's in AP Mode and that is all. We aren't sending large enough UDP Packets, if we send any at all, so that the Network Stack should crash.

Also, my ESP can stay in AP Mode indefinitely if I made a stable connection or no connection at all. I can connect and disconnect using my phone all day long and it will still work.

r1dd1ck, 2019-06-21:
@atanisoft 
> How is this possible, these are used only on the IDF side and IDF code is not recompiled inside Arduino IDE or in PIO when using the arduino framework option.

:roll_eyes:
Too many words, too little effort to find out yourself..

DarvosBR, 2019-06-21:
@r1dd1ck unfortunately the sdkconfig.h solution did not work. I'm gonna try the esp-idf, but I do agree this method should work.

@GeorgeFlorian The whole code works just fine when I access the board by the STA IP.
But on AP the code just freezes after a while (the board keeps working, the STA keeps working, but AP just frezzes.

atanisoft, 2019-06-21:
> Too many words, too little effort to find out yourself..

There is no need to be rude about this. In all other cases of IDF sdkconfig it will require a recompilation of the IDF code for the config change to take effect.

This however is one case where a recompilation is not necessary *ONLY* because the WiFi lib is calling esp_wifi_init with a locally created wifi_init_config_t (https://github.com/lbernstone/arduino-esp32/blob/master/libraries/WiFi/src/WiFiGeneric.cpp#L114) which the WIFI_INIT_CONFIG_DEFAULT expands to use the sdkconfig.h entries (https://github.com/espressif/esp-idf/blob/e7f85f1987aa9479c2dbab638ca83bcaef99be00/components/esp_wifi/include/esp_wifi.h#L193-L213). This is not typically the case.

@GeorgeFlorian give it a shot and see if it fixes it, if it does then https://github.com/espressif/arduino-esp32/pull/2912 should fix your issues when it has been merged and included in the next release.

DarvosBR, 2019-06-21:
well at least I have something new to share with you guys:

New error msg:

`dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
`

This send_offer msg had never appear before, at least not that I had noticed.

r1dd1ck, 2019-06-21:
@atanisoft 
>> @r1dd1ck Too many words, too little effort to find out yourself..
> There is no need to be rude about this.

That's not being rude, that's just stating a fact.

I'm sorry, but I go through the effort to provide a working solution, and then someone persistently keeps writing that it's garbage without even bothering to actually try once.. :neutral_face:
> This however is one case where a recompilation is not necessary

Glad you have finally checked, and acknowledge now  :+1: 

Have a nice sunny day :slightly_smiling_face:

DarvosBR, 2019-06-21:
Oh guys this is exciting, we are getting somewhere here!!

CORRUPT HEAP: Bad head at 0x3ffcf4a4. Expected 0xabba1234 got 0x3ffcf574
assertion "head != NULL" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x400dd8a7 on core 0

Backtrace: 0x4008cbac:0x3ffb3c60 0x4008cddd:0x3ffb3c80 0x400dd8a7:0x3ffb3ca0 0x4008c821:0x3ffb3cd0 0x40084f16:0x3ffb3cf0 0x400864bd:0x3ffb3d10 0x4000bec7:0x3ffb3d30 0x400f61c3:0x3ffb3d50 0x400f6223:0x3ffb3d70 0x400f626b:0x3ffb3d90 0x400f895a:0x3ffb3db0 0x400f89c2:0x3ffb3dd0 0x400d6ba1:0x3ffb3df0 0x400f5724:0x3ffb3e10 0x40088b49:0x3ffb3e40

Rebooting...

The new error issue I got!

so just a few extra informations, I just did a Fresh 1.0.2 instalation of the arduino-esp32 using the board manager from arduino IDE and the only thing I modify was the this:

```
WIFI_STATIC_RX_BUFFER_NUM = 8 → 10
WIFI_DYNAMIC_RX_BUFFER_NUM = 10 → 32
WIFI_RX_BA_WIN = 16 → 6
```

r1dd1ck, 2019-06-21:
> @DarvosBR 
> Oh guys this is exciting, we are getting somewhere here!!

Well, so far it only got you to a point where you will need to do a **full manual clean-up** & **re-install** :yum: because that heap corruption is very likely caused by some old files not being properly removed/replaced in the process :hankey: 

Or if not, then maybe forcing a full rebuild will help - the caching algorithm sometimes acts silly :roll_eyes:

atanisoft, 2019-06-21:
> CORRUPT HEAP: Bad head at 0x3ffcf4a4. Expected 0xabba1234 got 0x3ffcf574
> assertion "head != NULL" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
> abort() was called at PC 0x400dd8a7 on core 0

I agree with @r1dd1ck on the cleanup/rerun. But, it would be good to decode the stacktrace before cleanup/rerun. If this is from AsyncUDP there was this merge on master that is not on 1.0.2: https://github.com/espressif/arduino-esp32/commit/e9389e3122fc47227f961eaaa6541380ef4f2b0c

GeorgeFlorian, 2019-06-26:
Coming back with some updates of my own:

Without changing the code I know get `dhcps: send_offer>>udp_sendto result 0` even when I am connecting with my phone. It can connect though.

Here are the `serial debug verbose` when I hit Connect (to ESP's AP) on my phone:
```
Soft-AP IP address: 
109.108.112.114
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 15 - AP_STACONNECTED
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
```

As you can see:
`[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 17 - AP_STAIPASSIGNED` and `dhcps: send_offer>>udp_sendto result 0` show up **3 times** just for one connection. 

What is happening ?

I am using Platformio IDE and here is how the `platformio.ini` looks like:
```
[env:esp32doit-devkit-v1]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32dev
framework = arduino
board_build.flash_mode = qio
upload_port = /dev/ttyUSB0
monitor_port = /dev/ttyUSB0
monitor_speed = 115200

build_flags = -DCORE_DEBUG_LEVEL=5

#libraries
lib_deps =  https://github.com/me-no-dev/ESPAsyncWebServer.git
            https://github.com/me-no-dev/AsyncTCP.git
            https://github.com/marcmerlin/SmartMatrix_GFX.git
            https://github.com/marcmerlin/Framebuffer_GFX.git
            https://github.com/adafruit/Adafruit-GFX-Library.git
            https://github.com/FastLED/FastLED.git
```

Nathan-ma, 2019-06-27:
@GeorgeFlorian  yep same issue here, let me ask you something about this George (and I want to to know currently working to solve this problem my own, as soon as I get success I will let you know).

My question is: when you do connect with your phone, you have a stable connection even when that thing appears on your serial monitor (the DHCPS thing)? for me, no matter what I do, after 30 seconds connection freezes and I can refresh the page for example, until I quit the Wifi connection and reconnect.

Can you please try this for me ? connect it to your phone, refresh like 3 times with a 20 seconds interval and see if you can successfully do a fourth refresh!

Thank you very much

GeorgeFlorian, 2019-06-27:
> refresh like 3 times with a 20 seconds interval and see if you can successfully do a fourth refresh!

Refresh what ?

If I connect to the ESP's AP using the phone, the connection is stable for an undefined period of time.

Nathan-ma, 2019-06-27:
Refresh I meant if you are using a web server, but just now it hit me, maybe that's not your case!

GeorgeFlorian, 2019-06-27:
> Refresh I meant if you are using a web server, but just now it hit me, maybe that's not your case!

Actually it is. I am using  ESPAsyncWebServer. And it does not crash in AP, even after 4 refreshes.

Nathan-ma, 2019-06-27:
than I must recheck my hardware, i'm using ESPAsyncwebserver myself to read the value of a simple sensor and after a while AP freezes. Thank you George, this piece of information is valuable

GeorgeFlorian, 2019-06-28:
> than I must recheck my hardware, i'm using ESPAsyncwebserver myself to read the value of a simple sensor and after a while AP freezes. Thank you George, this piece of information is valuable

I haven't changed the code. Here is a little update:

I am now able to connect to the ESP32's AP using an Asus G771JM Laptop, that has Intel Dual Band Wireless-AC 7260. It seems pretty stable.
What I don't like is the fact that it takes more than 30 seconds to connect to it. Using a phone takes under 5 seconds.

Also, regarding the `ESPAsyncWebServer` in AP, I still get some hangings and "infinite loading" after some time. The first 2-3 refreshes work well. 
While it does NOT crash it sends me warnings:
```
[W][AsyncTCP.cpp:903] _poll(): rx timeout 4
[W][AsyncTCP.cpp:903] _poll(): ack timeout 4
```

Closing the browers and trying again seems to fix it. 

While in STATION mode, it works fairly fine for an undefined period of time, even after multiple refreshes. The only thing that I do not like in this situation is the fact that it's slow and sometimes it sends this warning:
```
[W][AsyncTCP.cpp:883] _poll(): pcb is NULL
```

GeorgeFlorian, 2019-07-09:
Any luck with this ?

I still can't connect all the time using a laptop or a PC.
So the IP gets assigned but I can't connect.
```
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
```
but there is no 
```
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 15 - AP_STACONNECTED
```

MkLHX, 2019-07-24:
there is something not work actually on AsyncTCP i have same kind of troubles.

Server not accessible in AP mode, tcp communication crash, werbserver crash sometimes...

https://github.com/espressif/arduino-esp32/issues/2995
https://github.com/espressif/arduino-esp32/issues/2979

GeorgeFlorian, 2019-07-29:
> there is something not work actually on AsyncTCP i have same kind of troubles.
> 
> Server not accessible in AP mode, tcp communication crash, werbserver crash sometimes...
> 
> #2995
> #2979

I don't think that the `AsyncTCP` or the `ESPAsyncWebServer` are at fault here. I think that it has something to do with `WiFi` .
I can't connect using my Home PC to the ESP32's AP, but I can use my office Laptop to connect to it and the web-server works perfectly fine. They both run under Windows 10.

MkLHX, 2019-07-30:
> 
> 
> > there is something not work actually on AsyncTCP i have same kind of troubles.
> > Server not accessible in AP mode, tcp communication crash, werbserver crash sometimes...
> > #2995
> > #2979
> 
> I don't think that the `AsyncTCP` or the `ESPAsyncWebServer` are at fault here. I think that it has something to do with `WiFi` .
> I can't connect using my Home PC to the ESP32's AP, but I can use my office Laptop to connect to it and the web-server works perfectly fine. They both run under Windows 10.
> 
> Also, check this: #3037

Thanks, made the same, i cross my fingers it's seems to work...

closed 2019-11-25 by stale[bot]

melvinpmathew, 2020-05-01:
@GeorgeFlorian did you get this issue resolved? Win10 + AP mode?

johnnytolengo, 2023-09-13:
has anyone fix that issue?