espressif/arduino-esp32 #2911: Watchdog is not enabled by default

toomasz, 2019-06-20:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.2
IDE name:						Visual Micro
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
Watchdog is not triggerred when i create endless while loop inside loop() method

### Sketch: 
```cpp

void setup()
{
	Serial.begin(115200);
	Serial.println("setup");
}

void loop()
{
	while (true)
	{
	}
}
```

atanisoft, 2019-06-20:
It is intentionally disabled by default, you will need to call enableLoopWDT() in setup to enable it.

toomasz, 2019-06-20:
@atanisoft do you know the reason why it's disabled by default? in arduino esp8266 its enabled

atanisoft, 2019-06-20:
@toomasz you would need to ask @me-no-dev that one.

GeorgeFlorian, 2019-06-24:
> @atanisoft do you know the reason why it's disabled by default? in arduino esp8266 its enabled

> so WDT is disabled in order to "help" run some code that is never meant to run on RTOS chip.

atanisoft, 2019-06-24:
@GeorgeFlorian that is a quote taken out of context and is not why the WDT is not enabled by default.

GeorgeFlorian, 2019-06-24:
> @GeorgeFlorian that is a quote taken out of context and is not why the WDT is not enabled by default.

I thought I could help.
Here's the context: https://github.com/me-no-dev/ESPAsyncWebServer/issues/524

closed 2019-09-06 by stale[bot]