espressif/arduino-esp32 #2918: Can not create int array in setup()

tonys-t, 2019-06-22:
### Hardware:
Board:							WeMos ESP32 DEVKIT WIFI BLE
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git?
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					921600
Computer OS:                                        Windows 8

### Description:
I found this problem when testing the speed of different methods of copying arrays.
The sketch was written for Arduino UNO and worked fine on Arduino MEGA 2560.
But not on ESP32.

When I declare the 
  int src[len], dst[len];
inside the setup() the ESP32 is rebooting 

when I declare 
  int src[len], dst[len];
just before the setup() everything seems to work fine.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#define len 1000
void copyB(int* src0, int* dst0, int len0) {
    for (int i = 0; i < len0; i++) {
        *dst0++ = *src0++;
    }
}

void setup() {
  int src[len], dst[len];
  
  Serial.begin(115200);
  delay(200);
  Serial.println("Starting test");
  
  for (int i = 0; i < len; i++) {
      src[i] = i;
  }
  
  memcpy(dst, src, sizeof(src[0])*len);
  
  copyB(src, dst, len);
  
  for (int i = 0; i < len; i++) {
      dst[i] = src[i];
  }
}

void loop() {
}
```

### Debug Messages:
```
entry 0x4008069c
***ERROR*** A stack overflow in task loopTask has been detected.
abort() was called at PC 0x40089008 on core 1

Backtrace: 0x40088dc4:0x3ffafec0 0x40088ff1:0x3ffafee0 0x40089008:0x3ffaff00 0x40085dd4:0x3ffaff20 0x40087700:0x3ffaff40 0x400876b6:0x3ffaff60 0x400d1949:0x000000e3

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8424
ho 0 tail 12 room 4
load:0x40080400,len:5868
entry 0x4008069c
***ERROR*** A stack overflow in task loopTask has been detected.
abort() was called at PC 0x40089008 on core 1

Decoding stack results
0x40088dc4: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x40088ff1: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x40089008: vApplicationStackOverflowHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 121
0x40085dd4: vTaskSwitchContext at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/tasks.c line 2799
0x400d1949: initApbChangeCallback at C:\Users\Tony\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32\esp32-hal-cpu.c line 44

```
In the installation of EspExceptionDecoder there is advice:
Change addr2line with GDB

but there is no explanation HOW :-)

lbernstone, 2019-06-22:
Calling a variable declaration in a function will use stack memory, which is much more limited than the heap.  Either make it global, allocate it with new, or use std::vector.
`int *src = new int[len];`

tonys-t, 2019-06-22:
Thank you for your quick answer. You should know, that I really appreciate it.
But!
Does this mean, that "upgrading" from slow and toy-like Arduino to ESP32 means, that I will have to start a private database with "good to know" pieces of advice to be able to write every simple project?
Look for example at the problems people have with the most basic use of UARTs in ESP32. What is going on here? See ESP32 devkit esp-idf UART1 example (of echo test) not working...
But again - sorry, this belongs to the forum, not here, I am just frustrated.
Thank you.

stickbreaker, 2019-06-23:
@tonys-t You need to review the hardware/software configuration.  This MCU only has 520KB of total Ram, It is a Dual processor cpu with 4MB flash, that is used to store code.  The platform is bases on the FreeRTOS operating system. Many, Many assumptions.  Your distaste with learning a new micro controller is at odds with your expressed interest?

Chuck.

tonys-t, 2019-06-26:
Hi Chuck,
you must be joking, right? Look at the sketch, that is taken from Arduino, that has much less RAM. I solved this "problem" with the hint from Ibernstone, even if I do not really understand why. I guess it has something to do with the C versus C++ mythology, creating and destructing objects etc.
One of the ideas of Arduino platform was portability between different processor clones. Need for rewriting a standard code, even for small changes is a disaster. From the very beginning of Arduino people were told, that digitalWrite() was actually unnecessary heavy and could be made faster by writing directly to the registers, but that would jeopardize portability of the code, so they should keep writing digitalWrite(). In the documentation for ESP32 there is not much written about the stack and what to do with it. To me it sounds like really very limited, if it can not survive a declaration of two integer arrays of length 1000. 
The last sentence of yours is to complicated for me to understand (my poor English). But I honor a simplicity principal: If you allow making simple things in a complicated way, you are digging a grave for you, when the more complicated things come.
With other words: If you allow yourself to be caught by a development system for declaration of two integer arrays of length 1000, how will your code look like controlling a coordinated swarm of drones?
I am almost 70 years old, so I am basically not here to learn new micro controllers. I want to solve my problems simply easy and fast enough to see them work :-)
But again, this discussion does not belong here. Have a nice time...

atanisoft, 2019-06-26:
@tonys-t there is a big difference in the AVR vs esp32. The first is the datatype, int on AVR is 16bit (likely) and on the esp32 it is 32bit. So your 1000 elements for each array is 40kb *each* which is higher than the 10kb heap that is allocated for setup() and loop(). The AVR does not have multi-tasking or per-task heap, it is all globally allocated. By shifting the declaration outside of setup() it would work on either AVR or esp32.

It has nothing to do with C vs C++. It has everything to do with the architecture difference between the platforms and the esp32 having a default stack size for the setup() and loop() methods set at 10kb.

tonys-t, 2019-06-26:
Hi @atanisft,
where are you smart guys, when I need you? I am not sarcastic here...

But to me it sounds more like 1000 elements of 32bits is 4kb not 40kb, so there is something else wrong with the stack/heap.

Is there a good book on the task concept I can read to understand? I am running into "watchdog" abort etc.

atanisoft, 2019-06-26:
A lot of us are available on gitter (see link in repo readme).

The WDT error you are encountering is likely due to a loop that doesn't have any call to delay in it. The delay call is necessary to allow the FreeRTOS system to run any other tasks which are ready, including IDLE1 or IDLE0. Adding delay(1) will likely fix the issue you are facing.

eos1d3, 2019-06-27:
> @tonys-t there is a big difference in the AVR vs esp32. The first is the datatype, int on AVR is 16bit (likely) and on the esp32 it is 32bit. 

AVR is 8 bit only.

atanisoft, 2019-06-27:
@eos1d3 AVR is definitely 8bit but there are 16bit integer types on that arch.

lbernstone, 2019-06-28:
It is a bit dense, but the [esp-idf documentation](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/freertos.html) explains the functions quite well, if not always good at explaining the big picture.

lbernstone, 2019-06-28:
[NKolban's book](https://leanpub.com/kolban-ESP32) may be helpful for you, as well.

closed 2019-09-10 by stale[bot]