espressif/arduino-esp32 #2919: eeprom2 failed to initialise when running "eeprom_class" example

sohaib9020, 2019-06-22:
Today i updated the esp32 arduino core to latest build.
I am getting the following message on executing the code:

> Failed to initialise AGE
> Restarting...

I am using the compatible partition table as required by the example.

  Generated partition that would work perfectly with this example
  #Name,   Type, SubType, Offset,   Size,    Flags
  nvs,     data, nvs,     0x9000,   0x5000,
  otadata, data, ota,     0xe000,   0x2000,
  app0,    app,  ota_0,   0x10000,  0x140000,
  app1,    app,  ota_1,   0x150000, 0x140000,
  eeprom0, data, 0x99,    0x290000, 0x1000,
  eeprom1, data, 0x9a,    0x291000, 0x500,
  eeprom2, data, 0x9b,    0x292000, 0x100,
  spiffs,  data, spiffs,  0x293000, 0x16d000,

Example code is given below:
`#include "EEPROM.h"

// Instantiate eeprom objects with parameter/argument names and size same as in the partition table
EEPROMClass  NAMES("eeprom0", 0x500);
EEPROMClass  HEIGHT("eeprom1", 0x200);
EEPROMClass  AGE("eeprom2", 0x100);

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("Testing EEPROMClass\n");
  if (!NAMES.begin(NAMES.length())) {
    Serial.println("Failed to initialise NAMES");
    Serial.println("Restarting...");
    delay(1000);
    ESP.restart();
  }
  if (!HEIGHT.begin(HEIGHT.length())) {
    Serial.println("Failed to initialise HEIGHT");
    Serial.println("Restarting...");
    delay(1000);
    ESP.restart();
  }
  if (!AGE.begin(AGE.length())) {
    Serial.println("Failed to initialise AGE");
    Serial.println("Restarting...");
    delay(1000);
    ESP.restart();
  }
}`

This example worked fine with the previous version of esp32 arduino code from 2018.

lbernstone, 2019-06-22:
In v1.0.3, EEPROM has moved from separate partitions into the nvs, so the partitioning scheme will not matter.
Can you turn on verbose logging to see if you get an error about why AGE fails?

stickbreaker, 2019-06-22:
The current EEPROM class does not use it's own partition, It uses a block inside the NVS partition, recreate your partition table without the EEPROM partitions, increase the size of NVS partition.

Chuck.

lbernstone, 2019-06-22:
jinx

stickbreaker, 2019-06-22:
@lbernstone The examples INO's need to be changed, the EEPROM class example still shows a partition table with EEPROM sections.  I think you should add a note about increasing NVS partition and removing EEPROM partitions.

Chuck.

lbernstone, 2019-06-23:
done :+1:

closed 2019-09-05 by stale[bot]