espressif/arduino-esp32 #2922: Experiencing issues with enableCore1WDT() and WDT

GeorgeFlorian, 2019-06-24:
Board: ESP32 DevKitV4 Wrover-B
OS: Linux Mint 19.1

Hello !

I am experiencing issues with `enableCore1WDT();`. I am trying to set this WDT to keep the ESP32 from becoming unresponsive whenever ESPAsyncWebServer decides to hangs or freeze. I was advised to do so by me-no-dev. The problem is that this, for me, is uncharted territory.

I dropped `enableCore1WDT();` at the beginning of `void setup()`:
```
void setup() {
  Serial.begin(115200);
  delay(1000);
  enableCore1WDT();
//...
```

And it returns an error:
```
0x400eb1eb: _svfiprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 930
0x400817e9: spi_flash_mmap_pages at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/soc/include/soc/soc_memory_layout.h line 173
0x400e9842: _vfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1736
0x4008a6fd: multi_heap_free_impl at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap.c line 375
0x400887fd: vTaskPrioritySet at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/include/freertos/portable.h line 208
```

```
E (18005) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (18005) task_wdt:  - IDLE1 (CPU 1)
E (18005) task_wdt: Tasks currently running:
E (18005) task_wdt: CPU 0: IDLE0
E (18005) task_wdt: CPU 1: loopTask
E (18005) task_wdt: Aborting.
abort() was called at PC 0x400eb1eb on core 0
```

My code became unusable after I've added `enableCore1WDT();`.

I really have no idea how this all works, as it is the first time I am trying something like this. Also there isn't much information online about this sort of stuff.

Also, here is the loop. Something in this seems to cause it.
On first boot, and in this specific case, the ESP32 is in AP_MODE and the following flags are set like this:
 * `restartFlag` is false
 * `AP_flag` is false

```
unsigned int free_spaces = 0;
String arrow = "";
String URL = "INVALID_URL";
String Brightness = "5";
int oldBright = 0;
String urlRefreshRate = "3";
bool AP_flag = false;
bool restartFlag = false;

int oldSignPlace = 0;
int httpFailCounter = 0;
int urlFailCounter = 0;

void loop() {
  if (restartFlag) {
    restart_sequence(5);
  }
  if(!AP_flag) { // Check if we are in AP_MODE
    int signPlace = 0;
    
    if (WiFi.status() == WL_CONNECTED && SPIFFS.exists("/configFile.txt")) {  //Check the current connection status
      logOutput((String)"IP ADDRESS: " + IP_copy.toString());
      if(URL.length() > 15) {
        HTTPClient http;
        Serial.println((String)"DEBUG: URL = " + URL);
        logOutput(http.begin(URL) ? "DNS Resolved" : "DNS Failed");
        int httpCode = http.GET(); //Make the request          
        if (httpCode > 0) { //Check for the returning code          
          logOutput((String)"[HTTP] GET... code: " + httpCode);
          if(httpCode == HTTP_CODE_OK) {
                String payload = http.getString();
                logOutput(payload);
                signPlace = getPlaces(payload);
                oldSignPlace = signPlace;
                httpFailCounter = 0;
                if(arrow == "") {
                  urlFailCounter ++;
                } else {
                  urlFailCounter = 0 ;
                }
          }
        } else {
            logOutput((String)"[HTTP] GET... failed, error: " + http.errorToString(httpCode).c_str());
            httpFailCounter++;
          }  
        http.end(); //Free the resources
        } else {
            logOutput("URL Link is invalid ! Please enter another URL");
        }
      if(urlFailCounter > 1) {
        SPIFFS.remove("/configFile.txt");
        scroll_text("URL Body format incorrect ! Couldn't find parking direction-sign !",LED_RED_HIGH,20);
        delay(100);
        restart_sequence(5);
      }
      if(httpFailCounter > 4) {
        delay(100);
        ESP.restart();
      }
      if(Brightness != NULL && Brightness.length() > 0 && Brightness.toInt() != oldBright) {
        oldBright = Brightness.toInt();
        matrixLayer.setBrightness((oldBright*255)/100);
      } 
      if(urlRefreshRate.toInt()*1000 >= 1000) {
        delay(urlRefreshRate.toInt()*1000);
      } else {
        delay(3000);
      }
    } // if (WiFi.status() == WL_CONNECTED)

    if(signPlace == 0) {
      signPlace = oldSignPlace;
    }
    if(signPlace == 2) {  // #>
          if (arrow == "<") {
            go_left(free_spaces, arrow);
          } else {
            go_right(free_spaces, arrow);
          }    
    } else if (signPlace == 3) { // <#
            if(arrow == ">") {
              go_right(free_spaces, arrow);
            } else {
              go_left(free_spaces, arrow);
            }
    }
  } // if(!APflag)
}
```

I've tried adding a `vTaskDelay(10)` but the WDT was triggered in this case as well.

```
E (18027) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (18027) task_wdt:  - IDLE1 (CPU 1)
E (18027) task_wdt: Tasks currently running:
E (18027) task_wdt: CPU 0: IDLE0
E (18027) task_wdt: CPU 1: SmartMatrixCalc
E (18027) task_wdt: Aborting.
abort() was called at PC 0x400eb1f3 on core 0
```
```
x400eb1f3: _svfiprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 936
0x400817e9: spi_flash_mmap_pages at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/soc/include/soc/soc_memory_layout.h line 173
0x400e984a: _vfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1736
0x4008a6fd: multi_heap_free_impl at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap.c line 375
0x400887fd: vTaskPrioritySet at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/include/freertos/portable.h line 208
```

GeorgeFlorian, 2019-06-24:
The answer is that you can add a `delay()` inside a `while` loop and that will let other tasks run.

closed 2019-06-24 by GeorgeFlorian

pfeerick, 2019-06-25:
It doesn't need to be much of a delay either... a `delay(1)` should be enough to let the the underlying RTOS do it's thing. That, or is `yield()` a thing on the ESP32 also? Basically, anything that prevents the watchdog from being fed for any length of time (can't remember... is it 500ms?) ... is likely to end in grief and exceptions.

pfeerick, 2019-06-25:
Oh... a matrix will colours... I'm running a couple of red led matrix panels... 7219 based... with ESP8266s... what ones are they, and what library do you use with them?

GeorgeFlorian, 2019-06-25:
> It doesn't need to be much of a delay either... a `delay(1)` should be enough to let the the underlying RTOS do it's thing. That, or is `yield()` a thing on the ESP32 also? Basically, anything that [...]

`yield()` did not work.

I am using an ESP32 DevKitV4 Wrover-B that drives a SMD 2727 RGB LED Panels using https://github.com/pixelmatix/SmartMatrix/tree/teensylc
On top of that I also have https://github.com/marcmerlin/SmartMatrix_GFX for added functionality.

gearsincorg, 2020-11-19:
I agree.  yield() does not work.

It would be nice if you could call delay(0) and JUST have it to a task yield without incurring any extra un-needed delay.
Currently delay(0) just returns and does not feed the WDT.

me-no-dev, 2020-11-19:
it would, but since the task that you are trying to yield to (and feed WDT) is lower priority than the one you are in, you need to delay for more than one RTOS tick in order to make sure that the scheduler will switch to it. `delay(2)` will do exactly that (Arduino uses 1kHz RTOS tick)

gearsincorg, 2020-11-19:
Ah...that helps to explain some other interesting timing I was seeing.   
Question: I know where the ESP-IDF reference is, but is there somewhere that covers the Arduino implementation (and explains things like the clock and the default cores and priorities)?