espressif/arduino-esp32 #2925: EPS32 lost WiFi.isHidden() ?

GeorgeFlorian, 2019-06-24:
Board: ESP32 DevKitV4 Wrover-B
OS: Linux Mint 19.1

Hello !

I am trying to scan for networks using ESPAsyncWebServer example:
```
  server.on("/scan", HTTP_GET, [](AsyncWebServerRequest *request){
  String json = "[";
  int n = WiFi.scanComplete();
  if(n == -2){
    WiFi.scanNetworks(true);
  } else if(n){
    for (int i = 0; i < n; ++i){
      if(i) json += ",";
      json += "{";
      json += "\"rssi\":"+String(WiFi.RSSI(i));
      json += ",\"ssid\":\""+WiFi.SSID(i)+"\"";
      json += ",\"bssid\":\""+WiFi.BSSIDstr(i)+"\"";
      json += ",\"channel\":"+String(WiFi.channel(i));
      json += ",\"secure\":"+String(WiFi.encryptionType(i));
      json += ",\"hidden\":"+String(WiFi.isHidden(i)?"true":"false");
      json += "}";
    }
    WiFi.scanDelete();
    if(WiFi.scanComplete() == -2){
      WiFi.scanNetworks(true);
    }
  }
  json += "]";
  request->send(200, "application/json", json);
  json = String();
  });
```

But `isHidden` is not a function in WiFi.
`error: 'class WiFiClass' has no member named 'isHidden'`

Is this normal ? From what I've found, ESP8266 had `WiFi.isHidden()`.

How can I work around this ?

atanisoft, 2019-06-24:
Doesn't look like it ever had that method on the esp32 side. You should be able to simply comment that line out and it should work.

GeorgeFlorian, 2019-06-24:
> Doesn't look like it ever had that method on the esp32 side. You should be able to simply comment that line out and it should work.

That's what I did, yes.
Is there any way of getting the same functionality as `WiFi.isHidden()` on the ESP32 ?

atanisoft, 2019-06-24:
> Is there any way of getting the same functionality as WiFi.isHidden() on the ESP32 ?

The esp-idf API only exposes an option to include hidden networks in the scan. It doesn't unfortunately have any flag included in scan results for if the network is hidden or not. So it is not likely to be added at this time other than to return false always.

If you want to list hidden networks though you can by calling WiFi.scanNetworks(true, true).

GeorgeFlorian, 2019-06-24:
> The esp-idf API only exposes an option to include hidden networks in the scan. It doesn't unfortunately have any flag included in scan results for if the network is hidden or not. So it is not [...]

Ok. Great ! Thank you for your time !

closed 2019-06-24 by GeorgeFlorian