espressif/arduino-esp32 #2928: Shouldn't WiFi.encryptionType return an enum not a simple int ?

GeorgeFlorian, 2019-06-25:
Hello !

I am trying to make a Network Scanner. All is good and dandy, but I have a small issue with `network += String(WiFi.encryptionType(i));`
Which is this:
```
/**
 * Return the encryption type of the networks discovered during the scanNetworks
 * @param i specify from which network item want to get the information
 * @return  encryption type (enum wl_enc_type) of the specified item on the networks scanned list
 */
wifi_auth_mode_t WiFiScanClass::encryptionType(uint8_t i)
{
    wifi_ap_record_t* it = reinterpret_cast<wifi_ap_record_t*>(_getScanInfoByIndex(i));
    if(!it) {
        return WIFI_AUTH_OPEN;
    }
    return it->authmode;
}
```
Shouldn't it return one of the following:
```
typedef enum {
    WIFI_AUTH_OPEN = 0,         /**< authenticate mode : open */
    WIFI_AUTH_WEP,              /**< authenticate mode : WEP */
    WIFI_AUTH_WPA_PSK,          /**< authenticate mode : WPA_PSK */
    WIFI_AUTH_WPA2_PSK,         /**< authenticate mode : WPA2_PSK */
    WIFI_AUTH_WPA_WPA2_PSK,     /**< authenticate mode : WPA_WPA2_PSK */
    WIFI_AUTH_WPA2_ENTERPRISE,  /**< authenticate mode : WPA2_ENTERPRISE */
    WIFI_AUTH_MAX
} wifi_auth_mode_t;
```

**???**

It doesn't. It returns a simple int: 3,4, etc.
How can I make it return `WIFI_AUTH_WPA2_PSK` for example ?

atanisoft, 2019-06-25:
It is returning the enum value but when you pass it into the String constructor it is treated as an integer, which is the base type for enums. If you want a string form of the authentication mode you would need to use a switch statement or similar to convert it before passing it to the String object.

GeorgeFlorian, 2019-06-25:
> It is returning the enum value but when you pass it into the String constructor it is treated as an integer, which is the base type for enums. If you want a string form of the authentication mode [...]

I thought that the `switch` was somehow magically done and that it should've returned in String form.

This should suffice, right ?
```
switch ( WiFi.encryptionType(i) ) {

  case 0 : 
   network += "WIFI_AUTH_OPEN";
   break;
  case 1 : 
   network += "WIFI_AUTH_WEP";
   break;
  case 2 : 
   network += "WIFI_AUTH_WPA_PSK";
   break;
  case 3 : 
   network += "WIFI_AUTH_WPA2_PSK";
   break;
  case 4 : 
   network += "WIFI_AUTH_WPA_WPA2_PSK";
   break;
  case 5 : 
   network += "WIFI_AUTH_WPA2_ENTERPRISE";
   break;
  case 6 : 
   network += "WIFI_AUTH_MAX";
   break;
}
```

atanisoft, 2019-06-25:
Yes, your code should work but instead of using 0-6 use the enum names.

GeorgeFlorian, 2019-06-25:
> Yes, your code should work but instead of using 0-6 use the enum names.

  case WIFI_AUTH_OPEN : 
   network += "WIFI_AUTH_OPEN";
   break;
  case WIFI_AUTH_WEP : 
   network += "WIFI_AUTH_WEP";
   break;
  case WIFI_AUTH_WPA_PSK : 
   network += "WIFI_AUTH_WPA_PSK";
   break;
  case WIFI_AUTH_WPA2_PSK : 
   network += "WIFI_AUTH_WPA2_PSK";
   break;
  case WIFI_AUTH_WPA_WPA2_PSK : 
   network += "WIFI_AUTH_WPA_WPA2_PSK";
   break;
  case WIFI_AUTH_WPA2_ENTERPRISE : 
   network += "WIFI_AUTH_WPA2_ENTERPRISE";
   break;
  case WIFI_AUTH_MAX : 
   network += "WIFI_AUTH_MAX";
   break;
}

That broke me. If `WiFi.encryptionType(i)` returns in `int`, how will the case work with something that isn't `int` ?

atanisoft, 2019-06-25:
It isn't returning an int, it returns the enum type. The String() constructor converts it to the base type of int.

GeorgeFlorian, 2019-06-25:
> It isn't returning an int, it returns the enum type. The String() constructor converts it to the base type of int.

So it's `String network;` fault. Ok, got it. Thanks !

closed 2019-06-25 by GeorgeFlorian