espressif/arduino-esp32 #2931: Better support for your official boards

Triangulix, 2019-06-25:
Hello 

I'm trying to use the Arduino framework on an ESP32-WROOM-32D with 8MB of flash.

I'm wondering why this official board is not supported by this official Espressif Arduino core.
You have support for all kinds of weird boards which most users probably will never see in their life but your official boards, in my case the ESP32-WROOM-32D, are not even listed as supported.

Users like me try to use "ESP32 Dev Module", which seems to be a sloppy way for writing "ESP32 DevKitC V4", as workaround but this board definition doesn't have a variant with 8MB of flash.

What do you recommend as a workaround for the time being until you have added full support for your own boards?

Best Regards
Triangulix

atanisoft, 2019-06-25:
The ESP32-WROOM-32D only has 4MB flash per https://www.espressif.com/en/products/hardware/modules If you have a module that has 8MB then it is likely not from Espressif themselves and would fall under one of the handful of boards which offer 8MB flash (heltec_wifi_lora_32_V2, heltec_wireless_stick, wipy3, gpy)

Triangulix, 2019-06-25:
The module is this one:
https://www.digikey.de/product-detail/en/espressif-systems/ESP32-WROOM-32D-8MB/1904-1024-1-ND/9381733?cur=EUR&lang=en

The variant with 8MB of flash is mentioned here:
https://docs.espressif.com/projects/esp-idf/en/latest/hw-reference/modules-and-boards.html#esp32-wroom-32d-esp32-wroom-32u

Best Regards
Triangulix

atanisoft, 2019-06-25:
That isn't a board but instead a chip that defaults to 4MB but has ordering options for 8MB or 16MB. Since it is not a board it shouldn't have a specific boards entry.

Now, if you have a specific board/module that *USES* one of these chips it can be registered as a board if it doesn't already exist.

Triangulix, 2019-06-25:
You call that PCB with lots of components on it a chip?

The Espressif page states:
Both modules have ESP32-D0WD chip on board of a smaller footprint than ESP32-D0WDQ6 installed in ESP32-WROOM-32. By default the module has 4MB flash and may be also ordered with custom flash size of 8 or 16MB, see Espressif Products Ordering Information. 

So I don't see why only the variants with 4 and 16MB are supported but not the variant with 8MB.

atanisoft, 2019-06-25:
The item you linked to is a chip, it is usually part of a board/module (https://www.espressif.com/en/products/hardware/esp32-devkitc/overview)

There are a handful of variants that support the 8MB flash size but they are for specific boards.

Triangulix, 2019-06-25:
Okay, then I need to adapt my question to your teminology:

Why is there no support for ESP32 chips on custom boards as it exists for ESP8266 and any other microcontroller supported by Arduino?
Has the support for Espressif products not included in 3rd party products been dropped in general?

atanisoft, 2019-06-25:
Most likely there hasn't been a request to support 8MB chips until now. Submitted a PR to add 8MB as an option in the menu for flash size. Default will remain as 4MB though.

Triangulix, 2019-06-25:
Okay I understand. Thanks for the information.

Triangulix, 2019-06-25:
I just saw the boards.txt. I'm going to try it. Thanks again for this.

Triangulix, 2019-06-25:
Are you sure that this change is enough to support the 8MB variant?
There is no partition scheme for 8MB. The default scheme only partitions 4MB of the flash memory. Or do I understand something wrong?

Triangulix, 2019-06-25:
I now added the following lines to my boards.txt:

esp32.menu.PartitionScheme.default=Default 8MB with spiffs
esp32.menu.PartitionScheme.default.build.partitions=default_8MB

esp32.menu.FlashSize.8M=8MB (64Mb)
esp32.menu.FlashSize.8M.build.flash_size=8MB
esp32.menu.FlashSize.8M.build.partitions=default_8MB

Also I noticed a bug in another board definition and replaced the line...
heltec_wifi_lora_32_V2.menu.FlashSize.2M.build.partitions=default_8MB

with the line
heltec_wifi_lora_32_V2.menu.FlashSize.8M.build.partitions=default_8MB

atanisoft, 2019-06-26:
Updated PR with the addition of default_8MB, the default partition for the esp32 is still 4MB so you will need to pick the schema and flash size accordingly.

closed 2019-09-08 by stale[bot]