espressif/arduino-esp32 #2941: Dual external pin interrupt firing

ddieffen, 2019-06-27:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.2
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:
I'm building a simple triac based dimmer using an AC Light Dimmer from RoboDyn module and an ESP32 Wroom dev board. It worked on an ATEGA328 but fails with the ESP32.

GPIO4 is connected to the Zero Crossing output of the AC module
GPIO5 is connected to the Gate input of the AC module
the AC module is powered by the 3V3 regulator of the ESP32 dev board

As shown on the oscilloscope screenshot below, why is the interrupt triggered twice ? This behavior zwas not happening with a Arduino Nano.

[image]<br /><a target='_blank' href='https://fr.imgbb.com/'>image hosting</a><br />

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

const byte phaseCtrl = 5;
const byte interruptPin = 4;
const byte ledPin = 2;

int dimming = 10;

void setup() {
  Serial.begin(115200);
  pinMode(phaseCtrl, OUTPUT);
  pinMode(ledPin, OUTPUT);
  pinMode(interruptPin, INPUT);
  attachInterrupt(digitalPinToInterrupt(interruptPin), zero_crosss_int, RISING);
}

void loop() {

}

void zero_crosss_int()  // function to be fired at the zero crossing to dim the light
{
  int dimtime = (75 * dimming);  // For 75 uS for 50Hz signal (100Hz zc) in 128 steps of 75 uS each
  delayMicroseconds(dimtime);    // Off cycle
  digitalWrite(phaseCtrl, HIGH);   // triac firing
  delayMicroseconds(10);         // triac On propagation delay (for 60Hz use 8.33)
  digitalWrite(phaseCtrl, LOW);    // triac Off
  delayMicroseconds(10);         // triac On propagation delay (for 60Hz use 8.33)
  digitalWrite(phaseCtrl, HIGH);   // triac firing
  delayMicroseconds(10);         // triac On propagation delay (for 60Hz use 8.33)
  digitalWrite(phaseCtrl, LOW);    // triac Off
  delayMicroseconds(10);         // triac On propagation delay (for 60Hz use 8.33)
  digitalWrite(phaseCtrl, HIGH);   // triac firing
  delayMicroseconds(10);         // triac On propagation delay (for 60Hz use 8.33)
  digitalWrite(phaseCtrl, LOW);    // triac Off
  delayMicroseconds(10);         // triac On propagation delay (for 60Hz use 8.33)
  digitalWrite(phaseCtrl, HIGH);   // triac firing
  delayMicroseconds(10);         // triac On propagation delay (for 60Hz use 8.33)
  digitalWrite(phaseCtrl, LOW);    // triac Off
  delayMicroseconds(10);         // triac On propagation delay (for 60Hz use 8.33)
  digitalWrite(phaseCtrl, HIGH);   // triac firing
  delayMicroseconds(10);         // triac On propagation delay (for 60Hz use 8.33)
  digitalWrite(phaseCtrl, LOW);    // triac Off
  delayMicroseconds(10);         // triac On propagation delay (for 60Hz use 8.33)
}
```

bertmelis, 2019-06-27:
Can't really read your screenshot. But maybe your signal is too slow it the ESP too fast. So during transition, the rising edge is detected twice.

lbernstone, 2019-06-28:
This can be fixed by setting a global lock variable at the top of your interrupt routine.  However, you are going to ultimately run into issues with your ISR taking too long and setting off the watchdog.  You should set the lock variable in the ISR, and then do the triac processing in your main loop (or use FreeRTOS semaphores if you know how to do that).

ddieffen, 2019-06-28:
Thanks, you were right, the issue was about a bouncy input signal. 
The locking mechanism worked !

closed 2019-06-28 by ddieffen